lumapu/ahoy #1320: Webseite / WebInstaller: Error 404

cptechnik, 2024-01-03:
File not found
Image nicht gefunden wenn man flashen will...

[image]

Sozpeed, 2024-01-04:
Geht bei mir seit drei Tagen auch nicht, selbe Fehlermeldung.

stefan123t, 2024-01-04:
Die ESP8266 Version 0.8.36 wurde m.W. vom git Action Build Prozess nicht erfolgreich gebaut.
Ein erneutes Bauen der alten Version ist so weit ich weiß auch nicht automatisch möglich.
Könnt Ihr nicht einfach eine neuere Version installieren ?

@lumapu ist die 0.8.36 die aktuelle stable / master Release Version ?

Sozpeed, 2024-01-04:
Hallo Stefan,

vielen Dank für deine Informationen, da wäre ich in 100 Jahren nicht drauf gekommen. 

Da ich wenig Ahnung habe, hat alles etwas länger gedauert. Aber es hat funktioniert.

Vielen herzlichen Dank.

Mit freundlichen Grüßen

Lars Morgenroth

lumapu, 2024-01-04:
habe es soeben gefixt, es hat ein `.bin` im `manifest.json` gefehlt.

cptechnik, 2024-01-05:
danke,danke,DANKE!

cptechnik, 2024-01-05:
...stimmt... also falls niemand was dagegen hat...

closed 2024-01-05 by cptechnik