lumapu/ahoy #1322: [Bug]no data when sending limit with MQTT in 0.8.36 or 0.8.42 (worked in last 0.7.36

tavoc, 2024-01-04:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

square dot

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.42

### Github Hash

0d10d19 :: 0.8.4

### Build & Flash Method

ESP Tools (flash)

### Setup

FHEM is sending with MQTT every 15 seconds a limit (at the moment all the time it is MQTT got topic: inverter/ctrl/limit/0 )

inverter communicates every 15 seconds to HM1500. Inverter Gap 500ms

### Debug Serial Log output

```shell
0.8.42 - 0d10d19
---------------
09:00:50.063 I: (#0) Radio infos: -6 -6 0 -6 -6 | t: 5, s: 3, f: 1, n: 1 | p: 1
09:00:50.066 I: (#0) TX 27 CH40 | 15 02 80
09:00:50.818 I: (#0) RX  78ms | 27 CH75 | 95 02
09:00:50.819 I: (#0) RX 150ms | 27 CH40 | 95 03
09:00:50.820 I: (#0) RX 207ms | 27 CH61 | 95 04
09:00:50.821 I: (#0) RX 252ms | 27 CH40 | 95 05
09:00:50.821 I: (#0) RX 353ms | 27 CH40 | 95 07
09:00:50.823 W: (#0) frame 1 missing: request retransmit (6 attempts left)
09:00:50.834 I: (#0) TX 11 CH40 | 15 e7 81
09:00:51.968 I: (#0) RX  37ms | 27 CH75 | 95 01
09:00:51.969 W: (#0) frame 6 missing: request retransmit (5 attempts left)
09:00:51.970 I: (#0) TX 11 CH40 | 15 e0 86
09:00:51.111 W: (#0) frame 6 missing: request retransmit (4 attempts left)
09:00:51.112 I: (#0) TX 11 CH40 | 15 e0 86
09:00:51.245 I: (#0) RX  44ms | 27 CH75 | 95 06
09:00:51.246 W: (#0) frame 8 missing: request retransmit (3 attempts left)
09:00:51.247 I: (#0) TX 11 CH40 | 15 ee 88
09:00:51.379 I: (#0) RX  39ms | 27 CH75 | 95 08
09:00:51.380 W: (#0) frame 9 missing: request retransmit (2 attempts left)
09:00:51.381 I: (#0) TX 11 CH40 | 15 ef 89
09:00:51.483 I: (#0) RX  34ms | 23 CH75 | 95 89
09:00:51.484 I: (#0) Payload (138)
09:00:51.485 -----
[... 446 lines cut ...]
09:02:41.069 I: MQTT got topic: inverter/ctrl/limit/0
09:02:42.069 I: (#0) Radio infos: -6 -6 0 -6 -6 | t: 13, s: 9, f: 2, n: 2 | p: 1
09:02:42.071 I: (#0) sendControlPacket cmd: 0b
09:02:42.072 I: (#0) TX 19 CH40 | 51 0b 81
09:02:42.574 I: (#0) request timeout: 500ms
09:02:42.574 -----
09:02:43.074 I: (#0) Radio infos: -6 -6 -2 -6 -6 | t: 14, s: 9, f: 2, n: 3 | p: 1
09:02:43.076 I: (#0) sendControlPacket cmd: 0b
09:02:43.077 I: (#0) TX 19 CH40 | 51 0b 81
09:02:43.580 I: (#0) request timeout: 500ms
09:02:43.580 -----
09:02:44.080 I: (#0) Radio infos: -6 -6 -4 -6 -6 | t: 15, s: 9, f: 2, n: 4 | p: 1
09:02:44.082 I: (#0) sendControlPacket cmd: 0b
09:02:44.082 I: (#0) TX 19 CH40 | 51 0b 81
09:02:44.585 I: (#0) request timeout: 500ms
09:02:44.585 -----
09:02:45.085 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 16, s: 9, f: 2, n: 5 | p: 1
09:02:45.086 I: (#0) sendControlPacket cmd: 0b
09:02:45.087 I: (#0) TX 19 CH61 | 51 0b 81
09:02:45.589 I: (#0) request timeout: 500ms
09:02:45.589 -----
09:02:46.089 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 17, s: 9, f: 2, n: 6 | p: 1
09:02:46.090 I: (#0) sendControlPacket cmd: 0b
09:02:46.091 I: (#0) TX 19 CH75 | 51 0b 81
09:02:46.593 I: (#0) RX  36ms | 17 CH23 | d1 81

tavoc, 2024-01-04:
[image]

added a picture of the radio statistics

reserve85, 2024-01-04:
I was able to observe that too. When zeroexport script is active the live date doesn´t update. 

We had no sun here in the last weeks, so ZeroExport did not set any limit to the inverter. I only noticed this in a short time slot when the sun was shining for a moment.

tavoc, 2024-01-04:
Als Idee, hat sich evtl die Art geändert wie man Limits übertragen muss? Das Interface sieht anders aus:
[image]

Im wesentlichen steuere ich wie folgt (fest immer alle 15 Sekunden)

([MQTT2_Tasmota_Zaehler:eHz_P_L1_L2_L3] < -100) (set MQTT2_FHEM_Server publish inverter/ctrl/limit/0 [limit:limit_hoy]W) DOELSE (set MQTT2_FHEM_Server publish inverter/ctrl/limit/0 1500W)

ab 100W zuviel Einspeisung wird limitiert. limit:limit_hoy ist ein integer Wert. Immer der nächst größere hunderter Wert. Ansonsten halt maximum

Im Log von ahoy steht jedoch immer nur MQTT got topic: inverter/ctrl/limit/0

müsste da nicht MQTT got topic: inverter/ctrl/limit/0 **1500W** stehen?

blueline13, 2024-01-04:
Schau dir bitte mal das an #https://github.com/lumapu/ahoy/issues/1313#issue-2062185191 Dort wirst du wertvolle Hilfen finden, was bei dir gerade schief läuft. 

Wenn du es probieren willst nimm mal eine alte Version 0.7.36 aber unbedingt die Hinweise im andere Issue beachten.

lumapu, 2024-01-05:
in den Logs fällt auf, dass das limit `0` ist. Meines wissens sollte man keine kleineren Werte  als `2` senden. Bei der Code-Analyse ist mir aufgefallen, dass schon seit der ersten `0.8.xx` Version das Power-Limit-Acknowledge nicht mehr MqTT übertragen wurde. Das habe ich der aktuellsten dev `0.8.43` behoben.

tavoc, 2024-01-05:
Moin, danke. sobald der Schnee weg ist probiere ich mal die 0.8.43

@lumapu : eigentlich setze ich immer das Powerlimit 1500W. In den Logs sieht es das so aus

`09:02:21.099 I: MQTT got topic: inverter/ctrl/limit/0
09:02:28.063 I: (#0) Radio infos: -6 -6 0 -6 -6 | t: 12, s: 8, f: 2, n: 2 | p: 1
09:02:28.065 I: (#0) sendControlPacket cmd: 0b
09:02:28.066 I: (#0) TX 19 CH40 | 51 0b 81
09:02:28.568 I: (#0) RX  35ms | 17 CH75 | d1 81
09:02:28.569 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0`

Das Topic lautet "0", aber mit dem Wert 1500W. Der MQTT Befehl hierzu lautet 
`set MQTT2_FHEM_Server publish inverter/ctrl/limit/0 1500W`

Evtl ließe sich direkt im Log die erste Zeile um den gewünschten Powerlimit Wert ergänzen, dann wir das sichtbarer. Aktuell kommt erst 7 Sekunder später das ACK (in der 0.8.42)

@blueline13 Danke für die Hinweise. Ich werde schauen inwieweit ich die umsetzen kann. Bis zur 0.7.36 war meine Einspeisungssteuerung jedoch ohne Probleme aktiv.

* ich setze immer ganze Werte (100er) in W
* immer alle 15 Sekunden (damit wurde implizit auch auf das ACK gewartet)
* Da meine Grundlast bei ca 200W ist wurde nie unter 200W geregelt

Ich werde mal sehen inwieweit ich die Regel umschreiben kann, dass der gleiche Wert nicht wiederholt wird. Also nur eine Änderung gesendet wird.
Bei den Versionen <0.8 hat es ja so funktioniert.

tavoc, 2024-01-05:
`0.8.43 - 21d77f5
---------------
10:27:08.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 513, s: 211, f: 5, n: 297 | p: 1
10:27:08.137 I: (#0) TX 27 CH23 | 15 0b 80
10:27:08.639 I: (#0) RX  80ms | 27 CH61 | 95 02
10:27:08.640 I: (#0) RX 141ms | 27 CH61 | 95 03
10:27:08.641 I: (#0) RX 182ms | 27 CH61 | 95 84
10:27:08.642 W: (#0) frame 1 missing: request retransmit (3 attempts left)
10:27:08.643 I: (#0) TX 11 CH23 | 15 e7 81
10:27:08.786 I: (#0) RX  31ms | 27 CH61 | 95 01
10:27:08.787 I: (#0) Payload (62)
10:27:08.789 -----
10:27:08.789 I: com loop duration: 656ms
10:27:08.790 -----
10:27:23.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 514, s: 212, f: 5, n: 297 | p: 1
10:27:23.137 I: (#0) TX 27 CH23 | 15 0b 80
10:27:23.639 I: (#0) RX 107ms | 27 CH75 | 95 02
10:27:23.640 I: (#0) RX 132ms | 27 CH61 | 95 03
10:27:23.641 I: (#0) RX 194ms | 27 CH61 | 95 84
10:27:23.642 W: (#0) frame 1 missing: request retransmit (3 attempts left)
10:27:23.643 I: (#0) TX 11 CH23 | 15 e7 81
10:27:23.786 I: (#0) RX  31ms | 27 CH61 | 95 01
10:27:23.787 I: (#0) Payload (62)
10:27:23.789 -----
10:27:23.789 I: com loop duration: 656ms
10:27:23.790 -----
10:27:38.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 515, s: 213, f: 5, n: 297 | p: 1
10:27:38.137 I: (#0) TX 27 CH23 | 15 0b 80
10:27:38.577 I: MQTT got topic: inverter/ctrl/limit/0
10:27:38.639 I: (#0) RX  66ms | 27 CH61 | 95 02
10:27:38.640 I: (#0) RX 130ms | 27 CH61 | 95 03
10:27:38.641 I: (#0) RX 175ms | 27 CH40 | 95 84
10:27:38.642 W: (#0) frame 1 missing: request retransmit (3 attempts left)
10:27:38.643 I: (#0) TX 11 CH23 | 15 e7 81
10:27:38.791 W: (#0) frame 1 missing: request retransmit (2 attempts left)
10:27:38.793 I: (#0) TX 11 CH23 | 15 e7 81
10:27:39.929 I: (#0) RX  41ms | 27 CH61 | 95 01
10:27:39.930 I: (#0) Payload (62)
10:27:39.932 -----
10:27:39.933 I: com loop duration: 799ms
10:27:39.933 -----
10:27:53.134 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 516, s: 214, f: 5, n: 297 | p: 1
10:27:53.136 I: (#0) sendControlPacket cmd: 0b
10:27:53.137 I: (#0) TX 19 CH23 | 51 0b 81
10:27:53.639 I: (#0) RX  35ms | 17 CH61 | d1 81
10:27:53.640 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
10:27:53.642 -----
10:27:53.642 I: com loop duration: 509ms
10:27:53.643 -----
10:27:58.440 I: MQTT got topic: inverter/ctrl/limit/0
10:28:08.134 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 517, s: 215, f: 5, n: 297 | p: 1
10:28:08.136 I: (#0) sendControlPacket cmd: 0b
10:28:08.137 I: (#0) TX 19 CH23 | 51 0b 81
10:28:08.639 I: (#0) RX  36ms | 17 CH61 | d1 81
10:28:08.640 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
10:28:08.641 -----
10:28:08.642 I: com loop duration: 508ms
10:28:08.643 -----
10:28:18.511 I: MQTT got topic: inverter/ctrl/limit/0
10:28:23.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 518, s: 216, f: 5, n: 297 | p: 1
10:28:23.137 I: (#0) sendControlPacket cmd: 0b
10:28:23.138 I: (#0) TX 19 CH23 | 51 0b 81
10:28:23.640 I: (#0) RX  35ms | 17 CH61 | d1 81
10:28:23.641 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
10:28:23.642 -----
10:28:23.643 I: com loop duration: 509ms
10:28:23.644 -----
10:28:38.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 519, s: 217, f: 5, n: 297 | p: 1
10:28:38.136 I: (#0) TX 27 CH23 | 15 05 80
10:28:38.482 I: MQTT got topic: inverter/ctrl/limit/0
[... 48 lines cut ...]
10:30:08.134 I: (#0) Radio infos: -6 0 -6 -6 -6 | t: 526, s: 221, f: 5, n: 300 | p: 1
10:30:08.136 I: (#0) TX 27 CH23 | 15 05 80
10:30:08.639 I: (#0) RX  39ms | 27 CH61 | 95 81
10:30:08.640 I: (#0) Payload (14)
10:30:08.642 -----
10:30:08.642 I: com loop duration: 508ms
10:30:08.642 -----
10:30:23.135 I: (#0) Radio infos: -6 2 -6 -6 -6 | t: 527, s: 222, f: 5, n: 300 | p: 1
10:30:23.136 I: (#0) TX 27 CH23 | 15 0b 80
10:30:23.639 I: (#0) RX  88ms | 27 CH61 | 95 02
10:30:23.640 I: (#0) RX 149ms | 27 CH61 | 95 03
10:30:23.641 I: (#0) RX 195ms | 27 CH40 | 95 84
10:30:23.642 W: (#0) frame 1 missing: request retransmit (3 attempts left)
10:30:23.643 I: (#0) TX 11 CH23 | 15 e7 81
10:30:23.785 W: (#0) frame 1 missing: request retransmit (2 attempts left)
10:30:23.785 I: (#0) TX 11 CH23 | 15 e7 81
10:30:24.917 I: (#0) RX  47ms | 27 CH61 | 95 01
10:30:24.919 I: (#0) Payload (62)
10:30:24.921 -----
10:30:24.921 I: com loop duration: 787ms
10:30:24.922 -----
`

Powerlimit wurde mit 1500W bis 10:29:30 geschickt. In der Zeit schickt der Inverter weiterhin keine Daten (mit der 0.8.43). Sobald die nicht mehr geschickt werden, geht es wieder.
Ich werde mal schauen wie es sich verhält wenn ich abwechselnd alle 15 Sekunden 1000W und dann 1500W sende.

tavoc, 2024-01-05:
Jetzt habe ich mal verschiedene Werte händisch per MQTT probiert und immer gewartet bis im Log das ACK kam. Keine Besserung, die Werte werden nicht aktualisiert. Erst ca 40-60 Sekunden nach dem letzten Powerlimit setzen kommen wieder Werte. 
Mit der alten Version wurden die scheinbar schneller abgearbeitet als jetzt. Das hieße aktuell, dass ich wohl nur alle 60 Sekunden einen Zielwert schreiben kann.
@blueline13 ich habe beim manuellen setzen die Kommentare berücksichtigt. Also immer andere Werte, immer gewartet bis im Log das "(#0) has accepted power limit set point 1000.00 with PowerLimitControl 0" kam. Der MQTT Befehl lautete dabei beispielsweise

`set MQTT2_FHEM_Server publish inverter/ctrl/limit/0 1000W`

`0.8.43 - 21d77f5
---------------
10:34:08.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 542, s: 236, f: 6, n: 300 | p: 1
10:34:08.137 I: (#0) TX 27 CH23 | 15 0b 80
10:34:08.639 I: (#0) RX  39ms | 27 CH61 | 95 01
10:34:08.640 I: (#0) RX 104ms | 27 CH61 | 95 02
10:34:08.641 I: (#0) RX 145ms | 27 CH61 | 95 03
10:34:08.642 I: (#0) RX 190ms | 27 CH40 | 95 84
10:34:08.643 I: (#0) Payload (62)
10:34:08.645 -----
10:34:08.645 I: com loop duration: 511ms
10:34:08.646 -----
10:34:16.421 I: MQTT got topic: inverter/ctrl/limit/0
10:34:23.134 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 543, s: 237, f: 6, n: 300 | p: 1
10:34:23.136 I: (#0) sendControlPacket cmd: 0b
10:34:23.137 I: (#0) TX 19 CH23 | 51 0b 81
10:34:23.639 I: (#0) request timeout: 500ms
10:34:23.639 -----
10:34:24.139 I: (#0) Radio infos: -6 3 -6 -6 -6 | t: 544, s: 237, f: 6, n: 301 | p: 1
10:34:24.141 I: (#0) sendControlPacket cmd: 0b
10:34:24.142 I: (#0) TX 19 CH23 | 51 0b 81
10:34:24.644 I: (#0) RX  37ms | 17 CH61 | d1 81
10:34:24.645 I: (#0) has accepted power limit set point 1000.00 with PowerLimitControl 0
10:34:24.646 -----
10:34:24.647 I: com loop duration: 1513ms
10:34:24.648 -----
10:34:38.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 545, s: 238, f: 6, n: 301 | p: 1
10:34:38.136 I: (#0) TX 27 CH23 | 15 05 80
10:34:38.639 I: (#0) RX  40ms | 27 CH61 | 95 81
10:34:38.640 I: (#0) Payload (14)
10:34:38.642 -----
10:34:38.642 I: com loop duration: 508ms
10:34:38.642 -----
10:34:53.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 546, s: 239, f: 6, n: 301 | p: 1
10:34:53.136 I: (#0) TX 27 CH23 | 15 0b 80
10:34:53.639 I: (#0) RX  70ms | 27 CH61 | 95 02
10:34:53.640 I: (#0) RX 133ms | 27 CH61 | 95 03
10:34:53.641 I: (#0) RX 174ms | 27 CH61 | 95 84
10:34:53.642 W: (#0) frame 1 missing: request retransmit (3 attempts left)
10:34:53.643 I: (#0) TX 11 CH23 | 15 e7 81
10:34:53.785 I: (#0) RX  30ms | 27 CH61 | 95 01
10:34:53.787 I: (#0) Payload (62)
10:34:53.789 -----
10:34:53.789 I: com loop duration: 655ms
10:34:53.790 -----
10:35:00.245 I: MQTT got topic: inverter/ctrl/limit/0
10:35:08.134 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 547, s: 240, f: 6, n: 301 | p: 1
10:35:08.136 I: (#0) sendControlPacket cmd: 0b
10:35:08.137 I: (#0) TX 19 CH23 | 51 0b 81
10:35:08.639 I: (#0) RX  36ms | 17 CH61 | d1 81
10:35:08.640 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
10:35:08.641 -----
10:35:08.642 I: com loop duration: 508ms
10:35:08.643 -----
10:35:13.862 I: MQTT got topic: inverter/ctrl/limit/0
10:35:23.134 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 548, s: 241, f: 6, n: 301 | p: 1
10:35:23.136 I: (#0) sendControlPacket cmd: 0b
10:35:23.137 I: (#0) TX 19 CH23 | 51 0b 81
10:35:23.639 I: (#0) RX  34ms | 17 CH61 | d1 81
10:35:23.640 I: (#0) has accepted power limit set point 1000.00 with PowerLimitControl 0
10:35:23.641 -----
10:35:23.642 I: com loop duration: 508ms
10:35:23.643 -----
10:35:29.223 I: MQTT got topic: inverter/ctrl/limit/0
[... 175 lines cut ...]
10:40:54.642 I: com loop duration: 508ms
10:40:54.643 -----
10:41:09.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 574, s: 262, f: 6, n: 306 | p: 1
10:41:09.136 I: (#0) TX 27 CH23 | 15 0b 80
10:41:09.639 I: (#0) RX  40ms | 27 CH61 | 95 01
10:41:09.640 I: (#0) RX 105ms | 27 CH61 | 95 02
10:41:09.641 I: (#0) RX 121ms | 27 CH61 | 95 03
10:41:09.642 I: (#0) RX 185ms | 27 CH40 | 95 84
10:41:09.643 I: (#0) Payload (62)
10:41:09.645 -----
10:41:09.645 I: com loop duration: 511ms
10:41:09.646 -----
10:41:24.135 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 575, s: 263, f: 6, n: 306 | p: 1
10:41:24.136 I: (#0) TX 27 CH23 | 15 0b 80
10:41:24.639 I: (#0) RX  66ms | 27 CH61 | 95 02
10:41:24.640 I: (#0) RX 130ms | 27 CH61 | 95 03
10:41:24.641 I: (#0) RX 171ms | 27 CH61 | 95 84
10:41:24.642 W: (#0) frame 1 missing: request retransmit (3 attempts left)
10:41:24.643 I: (#0) TX 11 CH23 | 15 e7 81
10:41:24.786 I: (#0) RX  32ms | 27 CH61 | 95 01
10:41:24.788 I: (#0) Payload (62)
10:41:24.791 -----
10:41:24.792 I: com loop duration: 658ms
10:41:24.792 -----
`

blueline13, 2024-01-05:
Moin, ich kann deinen Erfahrungen mit der 0.8.43 bestätigen. Sei zeigt keine wirklichen Verbsserungen zu 0.8.42.

scharrrfi, 2024-01-05:
Same problem here. I have downgraded Ahoy to 0.7.36 until this is fixed.

lumapu, 2024-01-05:
please test again once `0.8.44` is completly compiled

blueline13, 2024-01-05:
Leider nicht gelöst. Bei 4 Wechselrichtern siehr das so aus. Das "gelbe" ist der MQTT vom AC_Power Total und das Blaue ist die kumilierte Leistung von 4 Sonoffs. Auf der Status Seite der DTU wo man sehen kann ob die Wechselrichter Online sind passen die einzelnen Werte nur MQTT nicht.
[image]

blueline13, 2024-01-05:
0.8.44 -> Es wird schon besser. Was das Webserial angeht, kommen die Infos deutlich schneller. Auch das Ack vom MQTT/API wird etwas schneller, hat aber dafür nun einen riesigen Streuengsbereich von 2-25 Sekunde bekommen.

tavoc, 2024-01-06:
mit der 0.8.44 kann ich automatisiert alle 25 Sekunden zuverlässig das Limit senden (aktuell wieder immer 1500).
Bei alle 20 Sekunden aktualisiert sich das Webinterface ca. alle 1-2 Minuten.

es ist also schon wesentlich besser als die 0.8.43 oder davor, aber noch nicht auf dem Level der 0.7.36

das Log

`12:33:45.642 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
12:33:45.643 -----
12:33:45.644 I: com loop duration: 1514ms
12:33:45.645 -----
12:33:50.865 I: MQTT got topic: inverter/ctrl/limit/0
12:33:59.131 I: (#0) Radio infos: -6 -6 -6 -6 1 | t: 3610, s: 788, f: 132, n: 2690 | p: 1
12:33:59.133 I: (#0) sendControlPacket cmd: 0b
12:33:59.134 I: (#0) TX 19 CH75 | 51 0b 81
12:33:59.636 I: (#0) request timeout: 500ms
12:33:59.636 -----
12:34:00.137 I: (#0) Radio infos: -6 -6 -6 -6 -1 | t: 3611, s: 788, f: 132, n: 2691 | p: 1
12:34:00.139 I: (#0) sendControlPacket cmd: 0b
12:34:00.139 I: (#0) TX 19 CH75 | 51 0b 81
12:34:00.641 I: (#0) request timeout: 500ms
12:34:00.641 -----
12:34:01.141 I: (#0) Radio infos: -6 -6 -6 -6 -3 | t: 3612, s: 788, f: 132, n: 2692 | p: 1
12:34:01.142 I: (#0) sendControlPacket cmd: 0b
12:34:01.143 I: (#0) TX 19 CH75 | 51 0b 81
12:34:01.645 I: (#0) RX  36ms | 17 CH23 | d1 81
12:34:01.646 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
12:34:01.647 -----
12:34:01.648 I: com loop duration: 2517ms
12:34:01.649 -----
12:34:11.248 I: MQTT got topic: inverter/ctrl/limit/0
12:34:14.131 I: (#0) Radio infos: -6 -6 -6 -6 -1 | t: 3613, s: 789, f: 132, n: 2692 | p: 1
12:34:14.133 I: (#0) sendControlPacket cmd: 0b
12:34:14.134 I: (#0) TX 19 CH75 | 51 0b 81
12:34:14.636 I: (#0) request timeout: 500ms
12:34:14.636 -----
12:34:15.137 I: (#0) Radio infos: -6 -6 -6 -6 -3 | t: 3614, s: 789, f: 132, n: 2693 | p: 1
12:34:15.138 I: (#0) sendControlPacket cmd: 0b
12:34:15.139 I: (#0) TX 19 CH75 | 51 0b 81
12:34:15.641 I: (#0) request timeout: 500ms
12:34:15.641 -----
12:34:16.141 I: (#0) Radio infos: -6 -6 -6 -6 -5 | t: 3615, s: 789, f: 132, n: 2694 | p: 1
12:34:16.142 I: (#0) sendControlPacket cmd: 0b
12:34:16.143 I: (#0) TX 19 CH75 | 51 0b 81
12:34:16.645 I: (#0) request timeout: 500ms
12:34:16.645 -----
12:34:17.145 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3616, s: 789, f: 132, n: 2695 | p: 1
12:34:17.146 I: (#0) sendControlPacket cmd: 0b
12:34:17.147 I: (#0) TX 19 CH3 | 51 0b 81
12:34:17.649 I: (#0) request timeout: 500ms
12:34:17.650 -----
12:34:18.150 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3617, s: 789, f: 132, n: 2696 | p: 1
12:34:18.151 I: (#0) sendControlPacket cmd: 0b
12:34:18.152 I: (#0) TX 19 CH23 | 51 0b 81
12:34:18.654 I: (#0) request timeout: 500ms
12:34:18.654 -----
12:34:19.154 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3618, s: 789, f: 132, n: 2697 | p: 1
12:34:19.155 I: (#0) sendControlPacket cmd: 0b
12:34:19.156 I: (#0) TX 19 CH40 | 51 0b 81
12:34:19.658 I: (#0) request timeout: 500ms
12:34:19.658 -----
12:34:20.158 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3619, s: 789, f: 132, n: 2698 | p: 1
12:34:20.159 I: (#0) sendControlPacket cmd: 0b
12:34:20.160 I: (#0) TX 19 CH61 | 51 0b 81
12:34:20.662 I: (#0) RX  35ms | 17 CH03 | d1 81
12:34:20.663 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
12:34:20.664 -----
12:34:20.665 I: com loop duration: 6534ms
12:34:20.666 -----
12:34:29.131 I: (#0) Radio infos: -6 -6 -6 -4 -6 | t: 3620, s: 790, f: 132, n: 2698 | p: 1
[... 47 lines cut ...]
12:35:03.655 I: (#0) request timeout: 500ms
12:35:03.655 -----
12:35:04.155 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3627, s: 791, f: 133, n: 2703 | p: 1
12:35:04.156 I: (#0) sendControlPacket cmd: 0b
12:35:04.157 I: (#0) TX 19 CH61 | 51 0b 81
12:35:04.659 I: (#0) request timeout: 500ms
12:35:04.659 -----
12:35:05.159 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 3628, s: 791, f: 133, n: 2704 | p: 1
12:35:05.160 I: (#0) sendControlPacket cmd: 0b
12:35:05.161 I: (#0) TX 19 CH75 | 51 0b 81
12:35:05.663 I: (#0) RX  35ms | 17 CH23 | d1 81
12:35:05.664 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
12:35:05.665 -----
12:35:05.666 I: com loop duration: 6535ms
12:35:05.667 -----
12:35:11.945 I: MQTT got topic: inverter/ctrl/limit/0
12:35:14.131 I: (#0) Radio infos: -6 -6 -6 -6 -4 | t: 3629, s: 792, f: 133, n: 2704 | p: 1
12:35:14.133 I: (#0) sendControlPacket cmd: 0b
12:35:14.134 I: (#0) TX 19 CH75 | 51 0b 81
12:35:14.636 I: (#0) RX  35ms | 17 CH23 | d1 81
12:35:14.637 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
12:35:14.638 -----
12:35:14.639 I: com loop duration: 508ms
12:35:14.640 -----
`

lumapu, 2024-01-06:
es kommt sehr bald die `0.8.46`, da habe ich hoffentlich wieder eine kleine Verbesserung eingebaut =)

tavoc, 2024-01-09:
Hallo, ich hatte heute mal wieder ein wenig Zeit, es gab mit der 0.8.46 im Vergleich zur 0.8.45 keine wesentlichen Verbesserungen feststellen können. Aktuell liege ich im Bereich von 20-25 Sekunden für das regelmäßige Setzen von Limits.

lumapu, 2024-01-11:
bitte nochmal mit `0.8.51` bzw. `0.8.52` testen

tavoc, 2024-01-12:
Getestet mit der 08.52

Ich kann wieder auf 15 Sekunden gehen, es kommen ausreichend oft Daten im Liveview.
Das einzige was auffällt, ist, dass das Powerlimit meist auf n/a steht (das ist aber für mich nicht wichtig). Alle paar Minuten steht auch mal der richtige Wert drin von 100%

Log:
0.8.52 - 455d29a
---------------
08:23:26.131 I: (#0) Radio infos: -6 2 -6 -6 -6 | t: 68, s: 17, f: 1, n: 49 | p: 1
08:23:26.134 I: (#0) TX 27 CH23 | 15 0b 80
08:23:26.636 I: (#0) RX  40ms | 27 CH61 | 95 01
08:23:26.637 I: (#0) RX 105ms | 27 CH61 | 95 02
08:23:26.639 I: (#0) RX 123ms | 27 CH61 | 95 03
08:23:26.640 I: (#0) RX 184ms | 27 CH61 | 95 84
08:23:26.641 I: (#0) Payload (62)
08:23:26.644 -----
08:23:26.644 I: com loop duration: 514ms
08:23:26.645 -----
08:23:41.131 I: (#0) Radio infos: -6 2 -6 -6 -6 | t: 69, s: 18, f: 1, n: 49 | p: 1
08:23:41.134 I: (#0) TX 27 CH23 | 15 0b 80
08:23:41.636 I: (#0) RX  39ms | 27 CH61 | 95 01
08:23:41.637 I: (#0) RX 101ms | 27 CH75 | 95 02
08:23:41.639 I: (#0) RX 206ms | 27 CH61 | 95 84
08:23:41.640 W: (#0) frame 3 missing: request retransmit (8 attempts left)
08:23:41.641 I: (#0) TX 11 CH23 | 15 e5 83
08:23:41.786 I: (#0) RX  45ms | 27 CH61 | 95 03
08:23:41.787 I: (#0) Payload (62)
08:23:41.789 -----
08:23:41.790 I: com loop duration: 659ms
08:23:41.791 -----
08:23:41.089 I: MQTT got topic: inverter/ctrl/limit/0
08:23:42.290 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 70, s: 19, f: 1, n: 49 | p: 1
08:23:42.296 I: (#0) sendControlPacket cmd: 0b
08:23:42.297 I: (#0) TX 19 CH23 | 51 0b 81
08:23:42.801 I: (#0) RX  36ms | 17 CH61 | d1 81
08:23:42.803 I: (#0) has accepted power limit set point 1000.00 with PowerLimitControl 0
08:23:42.805 -----
08:23:42.806 I: com loop duration: 673ms
08:23:42.807 -----
08:23:56.131 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 71, s: 20, f: 1, n: 49 | p: 1
08:23:56.133 I: (#0) TX 27 CH23 | 15 05 80
08:23:56.636 I: (#0) RX  40ms | 27 CH61 | 95 81
08:23:56.637 I: (#0) Payload (14)
08:23:56.639 -----
08:23:56.639 I: com loop duration: 510ms
08:23:56.640 -----
08:24:01.062 I: MQTT got topic: inverter/ctrl/limit/0
08:24:02.136 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 72, s: 21, f: 1, n: 49 | p: 1
08:24:02.138 I: (#0) sendControlPacket cmd: 0b
08:24:02.139 I: (#0) TX 19 CH23 | 51 0b 81
08:24:02.641 I: (#0) RX  36ms | 17 CH61 | d1 81
08:24:02.642 I: (#0) has accepted power limit set point 1000.00 with PowerLimitControl 0
08:24:02.645 -----
08:24:02.646 I: com loop duration: 510ms
08:24:02.647 -----
08:24:11.131 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 73, s: 22, f: 1, n: 49 | p: 1
08:24:11.133 I: (#0) TX 27 CH23 | 15 0b 80
08:24:11.636 I: (#0) RX  39ms | 27 CH61 | 95 01
08:24:11.637 I: (#0) RX 128ms | 27 CH40 | 95 03
08:24:11.639 W: (#0) frame 2 missing: request retransmit (8 attempts left)
08:24:11.640 I: (#0) TX 11 CH23 | 15 e4 82
08:24:11.772 W: (#0) frame 2 missing: request retransmit (7 attempts left)
08:24:11.773 I: (#0) TX 11 CH23 | 15 e4 82
08:24:12.907 I: (#0) RX  36ms | 27 CH61 | 95 02
08:24:12.908 W: (#0) frame 4 missing: request retransmit (6 attempts left)
08:24:12.909 I: (#0) TX 11 CH23 | 15 e2 84
08:24:11.041 W: (#0) frame 4 missing: request retransmit (5 attempts left)
08:24:11.042 I: (#0) TX 11 CH23 | 15 e2 84
08:24:12.180 W: (#0) frame 4 missing: request retransmit (4 attempts left)
08:24:12.182 I: (#0) TX 11 CH23 | 15 e2 84
[... 144 lines cut ...]
08:26:33.646 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
08:26:33.649 -----
08:26:33.650 I: com loop duration: 1516ms
08:26:33.651 -----
08:26:41.131 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 91, s: 35, f: 1, n: 54 | p: 1
08:26:41.133 I: (#0) TX 27 CH23 | 15 0b 80
08:26:41.636 I: (#0) RX  39ms | 27 CH61 | 95 01
08:26:41.637 I: (#0) RX 103ms | 27 CH61 | 95 02
08:26:41.639 I: (#0) RX 121ms | 27 CH61 | 95 03
08:26:41.640 W: (#0) frame 4 missing: request retransmit (8 attempts left)
08:26:41.641 I: (#0) TX 11 CH23 | 15 e2 84
08:26:41.753 I: (#0) RX  35ms | 27 CH61 | 95 84
08:26:41.754 I: (#0) Payload (62)
08:26:41.756 -----
08:26:41.757 I: com loop duration: 627ms
08:26:41.758 -----
08:26:52.357 I: MQTT got topic: inverter/ctrl/limit/0
08:26:53.136 I: (#0) Radio infos: -6 4 -6 -6 -6 | t: 92, s: 36, f: 1, n: 54 | p: 1
08:26:53.139 I: (#0) sendControlPacket cmd: 0b
08:26:53.140 I: (#0) TX 19 CH23 | 51 0b 81
08:26:53.642 I: (#0) RX  34ms | 17 CH61 | d1 81
08:26:53.643 I: (#0) has accepted power limit set point 1500.00 with PowerLimitControl 0
08:26:53.646 -----
08:26:53.647 I: com loop duration: 511ms
08:26:53.648 -----

closed 2024-02-17 by lumapu