lumapu/ahoy #1323: Sunset&Sunrise als Schaltausgang am ESP32

Andy-Op, 2024-01-04:
erst mal vielen Dank, das ihr euch mit der 0.8.42 dem Thema angenommen habt. Ich bin neu bei ahoyDTU und mein Schwerpunkt ist im analogen Bereich. Daher bitte ich Rücksicht walten zu lassen, wenn mir die Dinge rund um den ESP32 nicht so vertraut sind. Bauen will ich bzgl. einer Nachteinspeisung am Balkonkraftwerk das hier:

[image]

Ich hatte hier noch einen DTU-Bausatz V3.0 aus folgender Quelle rumliegen:

https://wechselrichter-versand.de/Informationen/Anleitung-openDTU-HM/

Die Platine bietet Synergien zum Einbau eines Optokopplers. Herfür habe ich den 230 Volt Zweig umfunktioniert. Mit dem 1mm Bohrer 3 Löcher für den IC-Sockel gebohrt. Masse liegt gerade neben an. Das passt. Mit einem PC817 bin ich bei einem Primärstrom von 20mA, Das ergibt einen Vorwiderstand von (3,3V-1,3V)/0.02A = 100 Ohm . 20mA sind wohl an der Grenze dessen was der ESP32 dauerhaft an Output erlaubt. Low-Power Optokoppler wie TLP621−4 sind Pin-kompatibel. Der Vorwiderstand ist dabei auf 180 Ohm zu vergrößern. Hierbei habe ich unterstellt, dass ich Nachts ein high von 3,3 Volt bekomme.

[image]
[image]

Gibt es einen 3,3 Volt high? Wo soll dann denn der 100 Ohm hin?

mtet2, 2024-01-04:
z.B. GPIO 13 oder 14 siehe Bild. Und dann den Pin im Setup auch aktivieren.
Kleiner Tip noch: "Die Lötstellen der IC-Fassung sehen überwiegend wie kalte Lötstellen aus".
Die würde ich nachlöten.
[image]

Andy-Op, 2024-01-04:
Moin, den GPIO14 dann wohl weniger:
[image]
P.S.: Die Lötstellen sind verbleit und top. Da täuscht das Bild.

Andy-Op, 2024-01-04:
Hab die 0.8.42 drauf, Geo-Daten sind eingetragen. Wo genau erfolgt die GPIO Zuweisung/Aktivierung?

mtet2, 2024-01-04:
Hier ein Beispiel für einen ESP8266
[image]

Andy-Op, 2024-01-04:
Ich vermute dann mal System Config > Status Led > Night time

mtet2, 2024-01-05:
Na dann weiter suchen ;-)
[image]

knickohr, 2024-01-05:
Also sehe ich das richtig ?  Bei euch funktioniert die LED ?  Bei meinem Fusion-Board leider irgendwie nicht, obwohl wenn ich die LED an einen anderen Schaltausgang einer anderen Funktion lege tut es. Nur eben beim dem neuen Night Time nicht 😞 Scheint wohl ein Problem vom Fusion-Board zu sein.

mtet2, 2024-01-05:
Am ESP8266 habe ich den D2 GPIO4 genommen, weil er bei mir frei war und er auch für Relaisbeschaltung geeignet sein soll. Die "LED polarity" auf  high active gesetzt, damit der Optokoppler nachts einschaltet.  ;-) 
Das statische Schalten des Optokopplers funktioniert auch noch mit 10 mA IF Strom sicher, so das man auch mit einem 200 Ohm Vorwiderstand arbeiten kann.

knickohr, 2024-01-05:
Ja, es funktioniert 🫣

Es funktioniert nur genau anders herum wie erhofft. Also wenn die LED (aktiv high) an ist, dann ist es Nacht. Kein Problem, man muß es nur wissen 😂

Andy-Op, 2024-01-05:
Ich werde an meiner nachgeschalteten Treiberstufe C1 auf 470uF vergrößern. Dann kann ich auch einen Reboot überbrücken. Habt ihr sonst noch irgendwelche Watchdogs am laufen die rebooten?

Andy-Op, 2024-01-06:
Moin, noch ein Wunsch an Tag 2: Ich würde gerne den aktuellen Schaltstatus der LED auf der Web-Oberfläche sehen. Könnte auch so was wie Tag-/Nachtansicht sein. Oder dort wo man die LED aktiviert nebenan. Leer not defined, schwarz aus, grün an. Das wäre im Störungsfall hilfreich. Beim Aufbau der Website müsste ja genug Zeit da sein.

knickohr, 2024-01-06:
Da hast Du doch den Status Deines Pins :

[image]

Andy-Op, 2024-01-06:
Du schaltest ja schon die Hintergrundfarbe. Grün müsste high sein, gelb low und weiß nicht definiert. Ist das so?

Ich würde da auch einen andere Bezeichnung verwenden: Not defined/aktiv high/inaktiv low. Dazu gäbe es die Korrespondierende: Not defined/aktiv low/inaktiv high.

closed 2024-02-17 by lumapu