lumapu/ahoy #1324: [feature request] Show Time an Date of MaxPower Values.

Thoric67, 2024-01-04:
Hallo,

Ich würde gerne den Zeitpunkt der MaxPower Werte sehen, entweder per Mouseover oder wie bei den Alarms.

Mit freundlichen Grüßen

Thomas

## Hardware
  * [x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: HM700
Retailer URL: ______

### nRF24L01+ Module
* [x] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [x] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.34
Github Hash: 3178325

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2024-01-04:
SiehevWunschliste #1199

Gubi2023, 2024-01-04:
siehe https://github.com/lumapu/ahoy/issues/1123#issue-1867952141

zappoXX, 2024-01-06:
Moin,
hab gerade gesehen das Max AC Power im Menü durchschimmert... tritt seit Version 0.8.43 bei mir unter iOS mit allen Browsern auf (Safari, Chrome, Firefox), aber ist halt nur eine optische Geschichte...

[image]

Gubi2023, 2024-01-06:
Am Handy ist der Mouseover schlecht ausführbar, ziemliche Fummelei, bis das Datum angezeigt wird....

lumapu, 2024-01-06:
bei mir (Android) geht es sehr gut, ich muss einfach nur auf den MaxPwr klicken und sehe dann den Wert.
@zappoXX wird es bei dir dauerhaft angezeigt, oder nur wenn du es angezeigt hattest passiert der GUI Bug?

zappoXX, 2024-01-06:
Ja, sofort beim Klick auf‘s Menü… Dauerhaft!!

Gubi2023, 2024-01-06:
bei mir wird der Max-Wert nur angezeigt, wenn ich mit Mouseover das Datum aktiviert habe. und nach aktualisieren der Seite ist er weg.

zappoXX, 2024-01-06:
Ja… genau wie bei mir!!!

lumapu, 2024-01-07:
da bin ich leider kein Experte, evtl. findet sich ja hier ein HTML / CSS Experte 😇

gone-for-coding, 2024-01-09:
Kann es sein, dass der Zeitstempel umgerechnet wird? Eventuell Zeitzone/Sommerzeit oder ähnliches?

Bei mir steht im Tooltip 13:58 Uhr obwohl das Maximum um 11:58 Uhr war. Zu dem Zeitpunkt wo ich das gesehen habe, war es tatsächlich auch erst kurz nach 13 Uhr, lag also in der Zukunft.

lumapu, 2024-01-09:
@gone-for-coding guter Punkt, muss ich mir anschauen.

gone-for-coding, 2024-01-10:
> @gone-for-coding guter Punkt, muss ich mir anschauen.

Heute sieht es bisher allerdings richtig aus. Falls es noch Mal auftritt, melde ich mich.

gone-for-coding, 2024-01-10:
@lumapu 
Jetzt ist es wieder falsch. Pop-Up sagt 13:47 Uhr, aber da war er schon deutlich unter dem Maximum was um 12:21 Uhr war.

lumapu, 2024-01-10:
wenn kann es nur 1 Stunde offset sein, da wir aktuell Winterzeit haben und versehentlich die Zeitzone zweimal addiert wird, aber was "krummes" kann ich mir nicht vorstellen wie bei dir 1,5h

gone-for-coding, 2024-01-11:
> wenn kann es nur 1 Stunde offset sein, da wir aktuell Winterzeit haben und versehentlich die Zeitzone zweimal addiert wird, aber was "krummes" kann ich mir nicht vorstellen wie bei dir 1,5h

Zeitzone oder Sommer/Winterzeit kann es nicht sein. Es ist jetzt 13:02 Uhr und ich habe seit >30 Minuten rund 430W. Peak war 457W um 12:10 Uhr. Dennoch wird im Tooltip 13:01 Uhr angezeigt. Ich habe auch mehrfach komplett neu geladen.

Um 13:07 Uhr zeigt er noch immer 13:01. Wenn ich sehe, dass es sich ändert, melde ich mich noch Mal. Hat niemand sonst diese Auffälligkeit?

lumapu, 2024-01-11:
allerdings seltsam, dann stimmt die Logik dahiner womöglich nicht

lumapu, 2024-01-11:
hab den Fehler 😊. Der Timestamp wurde immer dann aktualisiert, wenn einer der DC-Kanäle ein neues Maximum erreicht hatte, nicht wie erwartet der AC-Channel.
fix kommt gleich ...

Gubi2023, 2024-01-11:
bleibt der mouseover oder wird der  geändert? wie gesagt, mit I-Phone ziemliche Fummelei

lumapu, 2024-01-11:
@Gubi2023 und wie 😇?

Gubi2023, 2024-01-11:
vielleicht wird der der Anzeigefehler damit auch gleich beseitigt..

https://github.com/lumapu/ahoy/issues/1324#issuecomment-1879510917

lumapu, 2024-01-11:
den verstehe ich nicht 😂. Habe aber glaube ich was gefunden, einfach morgen mal mit der `0.8.52` testen

closed 2024-02-17 by lumapu