lumapu/ahoy #1326: [Bug]

tester277, 2024-01-05:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [X] I will attach/upload an Image of my wiring

### Version

0.8.43

### Github Hash

Dev

### Build & Flash Method

was already installed

### Setup

Na

### Debug Serial Log output

```shell
n/a
```

### Error description

Nach Update 0.8.43 werden inkorrekte Werte über MQTT übermittelt.
Ich habe 2 Inverter in meinem Setup. Es wird mit der 0.8.43 nur die Gesamtleistung eines Inverters gesendet (abweichend vom Live View hier die Summe der Leitung beider Inverter). Mit 0.8.42 gibt es diesbezüglich keine Probleme.

RealNBB, 2024-01-05:
Bestätige. Selbes Problem hier auch.

rmayergfx, 2024-01-05:
Bitte Titel anpassen [Bug] ist etwas wenig aussagekräftig.

tester277, 2024-01-05:
> Bitte Titel anpassen [Bug] ist etwas wenig aussagekräftig.

Erledigt sorry

lumapu, 2024-01-05:
sehe es bei mir auch, bin schon dran es zu fixen

knickohr, 2024-01-05:
Leider mit der 44 noch nicht gefixed, das Total AC Power wackelt bei meinen 16 Invertern wie ein Kuhschwanz 😱

RealNBB, 2024-01-05:
> sehe es bei mir auch, bin schon dran es zu fixen

Leider besteht das Problem weiterhin mit 8.44

blueline13, 2024-01-05:
bei mir sieht das so aus. 
[image]

tester277, 2024-01-05:
Bei mir wird der Wert eines (beliebigen1/2) Inverters angezeigt, dabei frage ich das Topic total/P_AC ab.
Leider bei mir nicht gelöst.

lumapu, 2024-01-05:
hatte es zuerst falsch verstanden (heute Mittag). Jetzt sehe ich was ihr meint, wird dann in der `0.8.45` gefixt.

knickohr, 2024-01-06:
Funktioniert wieder mit der 45 👍

tester277, 2024-01-06:
Ja ich kann auch bestätigen, dass der Total Wert der Leistung wieder stimmt.
Danke für das schnelle Bug Fixing!

closed 2024-02-17 by lumapu