lumapu/ahoy #1327: Error: "Inverter serial does not match, got: 0x00000000, expected: 0x91960870"  HMS series

dave5655, 2024-01-05:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version:0.80.36
Github Hash: _______

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [x] USB Serial Log (attached)
* [x] Setup settings (use our templates ... to be added)

Hi there, 
I am trying to read the data from my inverter. I have a HMS 1600-4T and get this error. I typed the correrct serial number in the web interface, doublechecked that. CMT 2300a is "connected".  attached a picture, so you can confirm this.

Here the debugging code from the webserial:
0.8.36 - 6fb0535
---------------
11:31:09.092 I: (#0) Radio infos: t: 918, s: 0, f: 0, n: 918 | p: 20
11:31:09.093 I: (#0) TX 868.00Mhz | 15 01 80
11:31:09.597 I: (#0) RX  29ms | 0 -89dBm | 00 00
11:31:09.598 W: Inverter serial does not match, got: 0x00000000, expected: 0x91960870
11:31:09.598 W: (#0) frame 1 missing: request retransmit (3 attempts left)
11:31:09.609 I: (#0) TX 868.00Mhz | 15 18 81
11:31:09.712 W: (#0) frame 1 missing: request retransmit (2 attempts left)
11:31:09.712 I: (#0) TX 868.00Mhz | 15 18 81
11:31:09.815 W: (#0) frame 1 missing: request retransmit (1 attempts left)
11:31:09.815 I: (#0) TX 868.00Mhz | 15 18 81
11:31:10.918 W: (#0) frame 1 missing: request retransmit (0 attempts left)
11:31:10.918 -----
[image]
[image]
also tried to add a new inverter but didnt work
Thanks for you effort!

Linki84, 2024-01-05:
Hallo,
hast du schon versucht den Wechselrichter vom Stromnetz zu trennen, die DC Leitungen zu trennen, kurz warten und dann die DC Leitungen wieder anzuschließen?
Ich hatte zu beginn mit meinem HMS auch einige Probleme.
Das hat geholfen. Am nächsten Tag hat die Kommunikation mit dem Wechselrichter gleich beim Hochfahren am Morgen funktioniert

dave5655, 2024-01-05:
HI, ja habe ich schon probiert... alles abgesteckt, kurz gewartet, es wurde dann auch kein wechselrichter mehr erkannt, alles wieder angesteckt, trotzdem keine veränderung... Habe das aber jetzt schon mehrfach gehört, dass am nächsten tag alles ging, ich denke ich warte einfach mal bis morgen.

dave5655, 2024-01-05:
ahhhh jetzt hats "Klick" gemacht. ich habe eine ältere firmware aufgespielt, da hatte ich den gleichen fehler, dadurch bin ich jedoch hier aufmerksam geworden:  https://github.com/lumapu/ahoy/issues/1237
Sprich, es hat nur ein Draht gefehlt. Dadurch das beim CMT Modul "connected" angezeigt wurde, habe ich gar nicht mehr nach einem verdrahtungsfehler gesucht...

closed 2024-01-05 by dave5655