lumapu/ahoy #1331: MQTT :  Yield Day wird nur wenn alle Inverter online

knickohr, 2024-01-07:
Der Bug besteht schon sehr lange, kam mit irgendeiner 0.7.x rein.

Der Wert von Yield Day, Topic topic/total/yieldDay wird nur ausgegeben wenn alle Inverter im Status 2 sind. Ich vermute auch das es beim Yield Total, Topic topic/total/YieldTotal genauso ist.

In der WebGui und auf dem Display ist es OK.

[image]

[image]

lumapu, 2024-01-07:
hast sich das nicht geändert in den letzten Versionen? Ich finde es ist jetzt viel sauberer implementiert.
Aber es geht um die Totalwerte oder? War das nicht ein Feature, dass dieser Wert nur übertragen wird, wenn alle da sind?
Evtl. sollten wir es umbauen, sodass wir eine bestimmte Zeit warten und dann den Totalwert ausgeben, unabhängig davon ob alle online sind.

knickohr, 2024-01-07:
Oh, ich bin noch bei 460001 😅

tester277, 2024-01-07:
@knickohr -mich würde interessieren, was das für eine Visualisierung für die Netzfrequenz ist. In das mit Grafana erstell?
Wenn ja wie heißt das Panel? 
Vielen Dank

knickohr, 2024-01-07:
Node-Red, siehe #413

lumapu, 2024-01-07:
Die Totalwerte werden nur gesendet, wenn alle Inverterstati > `STARTING` sind
```cpp
enum class InverterStatus : uint8_t {
    OFF,
    STARTING,
    PRODUCING,
    WAS_PRODUCING,
    WAS_ON
};
```

knickohr, 2024-01-07:
Korrektur :
>= Starting 😉

knickohr, 2024-01-08:
Das Problem hierbei ist :
Wenn nich alle Inverter einmal gleichzeitig im Status 2 „Producing“ sind, wird kein Wert ausgegeben, bzw. Immer 0. Und momentan bei der Wetterlage mit auch noch Schnee auf den Modulen kann das sehr lange dauern.

Steht bei mir heute immer noch auf 0, obwohl im Display und der WebGUI bereits 112Wh rein kamen 😉

knickohr, 2024-01-10:
Funktioniert bei der 49 noch nicht 😉

lumapu, 2024-01-10:
der Vergleich ist aber jetzt `!= OFF`

knickohr, 2024-01-11:
Öhmmm *räusper*

[image]

Und das über 6 Stunden 😱 Sieht für mich sobald, als würde mindestens immer ein Inverter offline sein. Da ist irgendwo ein ganz dicker Hund drin.

Eben noch schnell die „goldene“ Version drauf gemacht und siehe da, selbst in dem Schummerlicht kommen tröpfelnd die einzeln YieldDay rein und werden summiert, obwohl der eine oder andere Inverter bereits sich schlafen legt.

lumapu, 2024-01-11:
dann habe ich es dich nicht falsch gesehen. Hast du zufällig auch in der WebUI geschaut?

knickohr, 2024-01-11:
Jupp, da ist es korrekt, sowie auf dem Display.

[image]

You69Man, 2024-01-11:
> Jupp, da ist es korrekt, sowie auf dem Display.

Heißt das, du hast die 51 drauf, und keine reboots oder "settings-kaputt" in Zusammenhang mit Display?

knickohr, 2024-01-12:
Nein, es ist soweit alles OK. Nur wie „drüben“ mal geschrieben :  Wenn ich eine neue Version drauf klatsche dann ist es kaputt. Und wenn ich Glück habe kann ich in den AP booten und mir die Settings neu importieren. Es ist definitiv was mit den Settings.

Da fällt mir gerade ein, vielleicht sollte ich mal kalt booten nach einem Update auf die „neuen“ Versionen. Vielleicht tritt dann das Problem auf, weil er nach einem Softboot die Settings nicht liest ?  Oder liest er die immer nach einem boot ?

lumapu, 2024-01-12:
Settings werden immer gelesen, aber wie immer: ein Powercylce hat schon viele Probleme behoben

knickohr, 2024-01-12:
Nun ja, dann würden ja nach dem er kaputt ist, er sich bei einem Neustart berappeln. Tut er aber dann nicht mehr. Egal, ist nicht Thema dieses Issues, das muß ich auch noch testen 😉

lumapu, 2024-01-12:
ein neuer Versuch, ich denke diesmal hab ich's erwischt 😉

Linki84, 2024-01-13:
Hallo @lumapu 
Hat wirklich geklappt mit der 0.8.53
[image]

Jedoch in der Übersicht wird der Wechselrichter erst bei 3,1W als verfügbar und produzierend angezeigt
[image]

[image]

knickohr, 2024-01-15:
Funktioniert 👍 Ab der ersten Wh 😎

commit 455d29a6 (2024-01-11): 0.8.52
* possible fix of 'division by zero' #1345
* fix lang #1348 #1346
* fix timestamp `max AC power` #1324
* fix stylesheet overlay `max AC power` #1324
* fix download link #1340
* fix history graph
* try to fix #1331
closed 2024-01-15 by knickohr