lumapu/ahoy #1332: Turn on/off über MQTT?

BerziOnline, 2024-01-07:
## Hardware
  * [ ] ESP8266
  * [x] ESP32
  * [ ] Raspberry Pi

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a Plus model capable of the required 256kBit/s mode
* [ ] square dot indicates original Nordic Semicon chip 
* [ ] round dot indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [x] circuit board
* [ ] external antenna (SMA)

### Version / Git SHA: 
Version: 0.8.36
Github Hash: 6fb05359398f24e14f4ddc3e7304b55b24fdd53c

### Build & Flash Method:
* [x] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Ich habe meinen Inverter bislang immer per MQTT über "ahoydtu.ctrl.power.0" mit 0 oder 1 an und ausschalten können. Aufgrund anderer Bugs verweilte ich lange auf der 0.7.26. Nach einem Upgrade auf die 0.8.36 nimmt er mir hier keine Befehle mehr an. Ich finde aber auch keine Alternative per MQTT dazu.

Hat sich hier das Topic geändert?

Vielen Dank!

knickohr, 2024-01-07:
Rest API
https://github.com/lumapu/ahoy/blob/main/User_Manual.md#control-via-rest-api

Aber es ist tatsächlich ein Bug und sollte gefixed werden. Nein, der Topic hat sich nicht geändert.

BerziOnline, 2024-01-07:
Danke für den Hinweis auf die RestAPI. Bei Alternative dachte ich zunächst im Bereich MQTT, da ich dachte, dass sich evtl. Pfade oder Topics geändert haben.

Mit der RestAPI war ich bislang auch nicht erfolgreich, über das Webfrontend klappt es aber, weshalb ich davon ausgehe, dass ich zu blöd bin.

Folgende Requests müssten doch korrekt sein, oder?

http://<IP>/api/ctrl?id=0&cmd=power&val=0
=> WR ausschalten (Turn Off)

http://<IP>/api/ctrl?id=0&cmd=power&val=1
=> WR anschalten (Turn On)

Irgendwie tut sich beim Turn On/Off nur über das Webfrontend was :-/

blueline13, 2024-01-07:
Ich habe es eben mit der 0.8.46 probiert und ich konnte meine Inverter aus- und wieder einschalten.

BerziOnline, 2024-01-07:
@blueline13 Danke für den Hinweis. Ich habe soeben die 0.8.47 installiert und damit geht es sofort wieder. Bug wurde dort entweder bewusst oder unbewusst behoben :-)

knickohr, 2024-01-08:
Der Inverter wird mit diesem Befehl nicht ausgeschaltet, sondern nur in den Standby gesetzt.

wenn jetzt über Nacht der Strom weg ist, dann wundert es mich nicht das er mit diesem Befehl nicht mehr starten, bzw. Komplett außer Tritt kommt.

Oder habe ich jetzt was falsch verstanden ?

BerziOnline, 2024-01-08:
@knickohr Es geht mir nur um dieses Standby. Da war ich in der Begrifflichkeit nicht ganz sauber, Sorry.

Da mein WR an einer Batterie hängt und diese ab einer Untergrenze nicht mehr einspeisen soll, bzw dann eben zu späterem Zeitpunkt wieder anfangen soll, ist dieser Turn Off/On recht relevant für meine Schaltung, da der WR sonst bei Einspeisestopp die sowieso schon recht leere Batterie weiterhin unnötig leersaugt. Bei Standby natürlich auch marginal, aber in diesem Zustand ist das dann zu vernachlässigen. Wenn man nur das Limit auf 0 setzt ohne ihn in standby zu versetzen, dann zieht er deutlich mehr.

Mein WR geht über Nacht nicht aus, da mit der Batterie immer eine Quelle anliegt, also nicht über die Uhrzeit meiner Nachrichten verwundert sein, dass ich dort weiterhin testen kann.

Vllt eine kurze andere Frage:
Gibt es denn einen MQTT Datenpunkt über welchen man zuverlässig herausfinden kann, ob der WR sich gerade im Standby befindet oder nicht? (gucke das fürs troubleshooting immer im Webfrontend nach) Der hier besagte ist ja nur zum schreiben - und letztendlich sagt dieser beim auslesen ja nur das letzte cmd aus, aber nicht, ob das auch geglückt ist.

blueline13, 2024-01-08:
@BerziOnline 
Ich habe auch ein Ticket #1313  offen wo die Antwortzeiten ein Problem mit den 0.8.xx Versionen sind und @lumapu hat ja auch geschrieben das er das als Bug einstuft. Ich würde sagen wir geben den Entwicklern Zeit. In einen der letzten Versionen ist ja schon ein klein wenig besser geworden. Manchmal brauchen gute Dinge ein wenig mehr Zeit :wink:

lumapu, 2024-01-08:
du könntest doch die aktuelle Leistung des Wechselrichters nehmen, die ist dann `0`

knickohr, 2024-01-08:
Was passiert denn wenn ihr auf das Ack wartet ?

Ahoy-DTU/Invertername/ack_pwr_Limit mit Payload true

Achtung, es kommt immer ein true, man muß auf das Eintreffen der Nachricht warten.

blueline13, 2024-01-08:
Ich kann nur für mich sagen das ich immer auf DTU/invertername/ch0/active_PowerLimit warte bis dort der geschriebene Wert zurück gegeben wird. Weil es vor der 0.8.xx eben extrem schnell da war und dann wusste ich auch ob er den richtigen Wert geschrieben hatte. 

Stimmt der ack_pwr_limit steht immer auf true und habe es deswegen mehr oder weniger ignoriert 😇 Dann werde ich es mal mit dem geänderten Timestamp probieren.

BerziOnline, 2024-01-09:
> du könntest doch die aktuelle Leistung des Wechselrichters nehmen, die ist dann `0`

Danke für den Hinweis. Nach genauerem Betrachten hilft das Objekt "ahoydtu/HM1500/available", was im Standby '3' ist und im laufenden Betrieb '2'. Mehr Statuscodes konnte ich nicht herausfinden, aber sie geben das zurück was ich gesucht hatte. Deinen Vorschlag mit der Einspeisung auf 0 oder nicht, habe ich als zusätzliche Prüfung mit reingenommen, danke dir!

> Was passiert denn wenn ihr auf das Ack wartet ?
> 
> Ahoy-DTU/Invertername/ack_pwr_Limit mit Payload true
> 
> Achtung, es kommt immer ein true, man muß auf das Eintreffen der Nachricht warten.

Das Ack kommt nur bei einem gesetzten Power-Limit, nicht bei dem Power 0/1. Aber war ein valider Gedanke, danke! 

> @BerziOnline Ich habe auch ein Ticket #1313 offen wo die Antwortzeiten ein Problem mit den 0.8.xx Versionen sind und @lumapu hat ja auch geschrieben das er das als Bug einstuft. Ich würde sagen wir [...]

Ja absolut, keine meiner issues ist irgendwie auf Ungeduld aus. Nachdem ich meine Scripte nun nochmals sauberer überarbeitet habe (Inverter Intervall maximal alle 5s, kein weiteres Cmd ohne vorheriges Ack, keinesfalls ein Cmd doppelt schicken, ...) wird die ursprüngliche Vermutung weiterhin deutlicher, dass der WR in den 0.8.x Versionen deutlich träger reagiert. Warum das so ist, weiß ich nicht. Aber ein 1-zu-1 Tausch gegen die 0.7.26 führt zu unmittelbaren Reaktionen und Annahmen von Limits/Befehlen und bei den neueren Versionen ist jedes Mal ein starkes Delay enthalten, wie auch von dir bereits beschrieben.

blueline13, 2024-01-09:
Ich habe mir das jetzt mit den Zeiten angeschaut und es ist deutlich besser nur auf das Flag DTU/inverter/ack_pwr_limit zu warten.  Sicher noch nicht das Verhalten wie bei den 0.7.xx Versionen, aber deutlich schneller als das bestätigte "active_PowerLimit".

Dann werde ich meinen Code ein wenig anpassen und darauf verzichten auf das DTU/inverter/chX/active_PowerLimit zu vergleichen und nur beim nächsten schreiben zu Prüfen welchen Wert ich beim letzten mal geschrieben hatte um gleiche Wert zu vermeiden.

Ich habe noch eine Frage, wertet ihr das Flag DTU/inverter/ack_pwr_limit irgendwie in der DTU aus? Meine Idee ist es vor dem Schreiben das Flag im MQTT auf FALSE zu setzen und dann nur auf das TRUE als Bestätigung zu warten. Dann kann ich mir den Umweg über die Zeitstempel der Änderung ersparen.

@lumapu Danke für den Schubs in die bessere Richtung 😃

lumapu, 2024-01-09:
klingt aber noch nicht nach zufrieden bei euch. Da müssen wir nochmal ran, die aktuelle Version soll in allen Belangen die beste Version sein 😄

BerziOnline, 2024-01-09:
> Ich habe mir das jetzt mit den Zeiten angeschaut und es ist deutlich besser nur auf das Flag DTU/inverter/ack_pwr_limit zu warten. Sicher noch nicht das Verhalten wie bei den 0.7.xx Versionen, aber deutlich schneller als das bestätigte "active_PowerLimit".

Ich bin in unseren Gesprächen schon immer nur von dem ack_pwr_limit ausgegangen und betrachte das andere zur Limitsetzung nicht. Der Versatz ist dennoch deutlich höher als bei der 0.7.26 :/

> Ich habe noch eine Frage, wertet ihr das Flag DTU/inverter/ack_pwr_limit irgendwie in der DTU aus? Meine Idee ist es vor dem Schreiben das Flag im MQTT auf FALSE zu setzen und dann nur auf das TRUE als Bestätigung zu warten. Dann kann ich mir den Umweg über die Zeitstempel der Änderung ersparen.

Ein Acknowledge ist einfach nur ein flag was zurück kommt, weshalb es auch kein false gibt. Es sollte im Sinne des Erfinders sein einfach nur nach dem letzten Zeitstempel zu schauen, wann das letzte ack (hier halt dann immer als true dargestellt) eingegangen ist. Ich verstehe es du vorhast, aber es hat nicht wirklich einen Mehrwert und wäre eher eine Code-Kosmetik :-)

lumapu, 2024-01-09:
achso du meinst, das Topic `ack` soll nicht `true` sein, sondern den aktuellen Zeitstempel enthalten, zB. `123123123`?

BerziOnline, 2024-01-09:
@lumapu nene, ich finde dass das sehr getreu dargestellt ist und der Zeitstempel ja als Zeitstempel sowieso vorhanden ist :)  Wollte @blueline13 nur auf seine Frage hin meine Meinung mitgeben, dass das künstliche Zurückändern auf ein false eher Kosmetik wäre :)

lumapu, 2024-01-09:
auf false setzen habe ich sogar versucht. es geht garnicht, da es bei mir wohl zeitgleich beim Broker ankam

blueline13, 2024-01-10:
@BerziOnline  Ja genau nur Kosmetik um den Code später leichter lesen zu können. Bei meinem MQTT konnte ich den Topic ack_pwr_limit ändern ohne das er sofort neu gesetzt wurde. Er wurde erst wieder neu gesetzt als die DTU nach einem neuen Power Limit das ACK gesendet hatte. 

@lumapu Dein Hinweis, das es bei dir nicht klappt, hat mich dazu bewogen einen anderen Weg zu gehen und den Topc nicht zu zu beschreiben das es bei mir scheinbar ein Zufall war das es geklappt hatte. Manchmal hilft einfach etwas Schlaf. 

Danke euch beiden.

knickohr, 2024-01-10:
> auf false setzen habe ich sogar versucht. es geht garnicht, da es bei mir wohl zeitgleich beim Broker ankam

Ich glaube er meint, ob msn es nach einem true beispielsweise nach 1s auf false setzt. Hmmm, das wäre aber dann auch Käse, denn wenn der Client es nicht mitbekommen hat, dann geht’s unter.

Die beste Lösung ist wohl das der Client, der es auswertet, selbst danach auf false setzt.

BerziOnline, 2024-01-10:
@lumapu Ich habe soeben mal nochmal die Logs durchgesehen und es ist definitiv so, dass mit den Versionen oberhalb der 0.7.26 ein ultra krasses Delay der Befehlsannahme eingewandert ist. Ich weiß das so genau, weil ich eine Wasserpumpe habe, die für 10 Sekunden anspringt und dabei direkt 1kw zieht, den Verbrauch nach oben springen lässt, und dann wieder 1kw nach unten fallen lässt. Deshalb befasse ich mich schon recht lange damit wie schnell der Wechselrichter reagiert. Über eine Tasmota Steckdose wird das nach wenigen ms getriggert, dass eine Erhöhung an den WR geschickt wird, aber die Erhöhung wird halt erst durch das Ack im Schnitt 10-20 Sekunden später angenommen. (und es ist auch nicht so, dass er es direkt annimmt, aber erst beim nächsten Intervall das Ack kommt, er aber schon vorher angepasst hat - nein, das sieht man am Stromzähler, dass er auch wirklich erst nach dieser Zeit überhaupt anpasst.)
Bis dahin ist die Pumpe schon wieder aus und ich speise über 1kw für die Zeit ein, weil es weitere 10-20s zur erneuten Verringerung braucht. Diese Delays gab es zuvor so nicht, ich kann aber echt nicht erkennen woher das in den neuen Versionen kommt.

Während dieser Zeit wird der Inverter mit keinem anderen Befehl penetriert. Irgendwas läuft hier anders zwischen DTU und Inverter seither - ich kann aber nicht überblicken was.

Mit der 0.7.26 hat das innerhalb 1-5s geklappt, eher deutlich unter 5s.

Das ist ein echtes Problem in den neuen Versionen, worauf ich hier nur nochmal hinweisen wollte - weil das etwas in diese false/true Kosmetik hier abgedriftet ist :-D 

=> Soll ich hierfür ein neues issue eröffnen?

lumapu, 2024-01-10:
nein passt, bin wieder beim Thema 😅 wobei der Titel des Issues überhaupt nicht mit dem aktuellen Thema zusammenpasst:
entweder umbenennen oder doch einen neuen Issue aufmachen

closed 2024-02-17 by lumapu