lumapu/ahoy #1337: [Request] Add Grid Profile 0x0D00

QuickStopGroceries, 2024-01-08:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.36

### Github Hash

6fb0535

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

...

### Error description

Please add Grid Profile for inverter HM-800 (French Model)
Unknown Profile

`0D 00 20 00 00 08 08 FC 07 30 00 01 0A 55 00 01 09 E2 10 00 13 88 12 8E 00 01 14 1E 00 01 20 00 00 01 30 07 01 2C 0A 55 07 30 14 1E 12 8E 00 32 00 1E 40 00 07 D0 00 10 50 00 00 01 13 9C 01 90 00 10 70 00 00 01 `

lumapu, 2024-01-08:
please check our latest development release, there the profile is already included 😊
https://fw.ahoydtu.de/fw/dev/ahoy_v0.8.48/

QuickStopGroceries, 2024-01-09:
Ja, funktioniert - super! 👍 Auch vielen Dank für die deutsche Übersetzung.

Die Kirsche auf der Torte wäre jetzt noch, wenn man sein Grid Profile lokal sichern und ein neues Grid Profile auf den WR hochladen könnte. Dann könnte ich meinen französischen WR auch konform in Deutschland betreiben. @lumapu 

Dieser Issue kann dann geschlossen werden.

QuickStopGroceries, 2024-01-10:
@stefan123t Das ist mir schon klar. 😄 Ich kann's auch durch ein Python-Script jagen und daraus ein Binary erzeugen. Aber dann ist halt Ende momentan. Zumindest für mich als Nicht-Entwickler.

Mir ist schon klar, dass das Thema keine Prio hat, aber die Funktion "Update Grid Profile" wäre schon nett. ☺️

closed 2024-02-17 by lumapu