lumapu/ahoy #1338: [Bug] Lots of "Time calibration" messages

exo1987, 2024-01-08:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.48

### Github Hash

 3c5be9a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MQTT intervall = 0
inverter intervall = 5 sec
reset at midnight
HM1200 (3 of 4 panels connected)
no power limit
One inverter

### Debug Serial Log output

```shell
0.8.48 - 3c5be9a
---------------
16:21:40.997 I: (#0) RX  43ms | 27 CH23 | 95 01
16:21:40.999 I: (#0) RX 157ms | 27 CH23 | 95 03
16:21:39.001 I: (#0) RX 199ms | 27 CH23 | 95 84
16:21:40.261 I: (#0) RX  32ms | 27 CH23 | 95 02
16:21:40.263 I: (#0) Payload (62)
16:21:40.264 I: alarm ID incremented to 11672
16:21:40.266 -----
16:21:41.277 I: (#0) RX  17ms | 27 CH23 | 95 01
16:21:41.281 I: (#0) RX  81ms | 27 CH23 | 95 02
16:21:41.283 I: (#0) RX 185ms | 27 CH23 | 95 84
16:21:41.397 I: (#0) RX  15ms | 27 CH23 | 95 03
16:21:41.404 I: (#0) Payload (62)
16:21:41.406 I: alarm ID incremented to 11673
16:21:41.408 -----
16:21:43.670 -----
16:21:44.919 I: (#0) RX  21ms | 27 CH23 | 95 01
16:21:44.922 I: (#0) RX 127ms | 27 CH23 | 95 03
16:21:44.924 I: (#0) RX 189ms | 27 CH23 | 95 04
16:21:43.030 I: (#0) RX  16ms | 27 CH23 | 95 02
16:21:44.155 I: (#0) RX  23ms | 27 CH23 | 95 05
16:21:44.260 I: (#0) RX  14ms | 27 CH23 | 95 06
16:21:44.378 I: (#0) RX  24ms | 27 CH23 | 95 07
16:21:44.491 I: (#0) RX  22ms | 27 CH23 | 95 08
16:21:44.608 I: (#0) RX  27ms | 27 CH23 | 95 09
16:21:44.712 I: (#0) RX  24ms | 15 CH23 | 95 8a
16:21:44.714 I: (#0) Payload (146)
16:21:44.716 -----
16:21:46.968 I: (#0) RX  25ms | 15 CH23 | 95 81
16:21:46.970 I: (#0) Payload (2)
16:21:46.973 -----
16:21:47.973 -----
16:21:48.224 -----
16:21:49.143 W: send queue almost full, consider to increase interval, 95 of 100 entries used
16:21:49.475 I: (#0) RX  17ms | 27 CH23 | 95 81
16:21:49.477 I: (#0) Payload (14)
16:21:49.479 -----
16:21:51.730 I: (#0) RX  23ms | 27 CH23 | 95 01
16:21:51.732 I: (#0) RX  90ms | 27 CH23 | 95 02
16:21:51.835 I: (#0) RX  22ms | 19 CH23 | 95 83
16:21:51.837 I: (#0) Payload (38)
16:21:51.839 -----
16:21:51.090 I: (#0) RX  29ms | 27 CH23 | 95 01
16:21:51.092 I: (#0) RX 135ms | 27 CH23 | 95 03
16:21:51.094 I: (#0) RX 197ms | 27 CH23 | 95 84
16:21:52.216 I: (#0) RX  27ms | 27 CH23 | 95 02
16:21:52.219 I: (#0) Payload (62)
16:21:52.221 -----
16:21:53.472 -----
16:21:54.252 W: send queue almost full, consider to increase interval, 98 of 100 entries used
16:21:54.723 I: (#0) RX  24ms | 27 CH23 | 95 01
16:21:54.725 I: (#0) RX 131ms | 27 CH23 | 95 03
16:21:54.726 I: (#0) RX 192ms | 27 CH23 | 95 04
16:21:54.728 I: (#0) RX 274ms | 19 CH23 | 95 86
16:21:55.851 I: (#0) RX  33ms | 27 CH23 | 95 02
16:21:55.954 I: (#0) RX  24ms | 27 CH23 | 95 84
16:21:55.956 W: (#0) CRC Error -> complete retransmit
16:21:55.957 -----
16:21:56.958 I: (#0) RX  29ms | 27 CH23 | 95 01
16:21:56.211 I: (#0) RX  30ms | 27 CH23 | 95 02
16:21:56.328 I: (#0) RX  23ms | 27 CH23 | 95 03
16:21:56.432 I: (#0) RX  23ms | 27 CH23 | 95 84
16:21:56.434 I: (#0) Payload (62)
16:21:56.435 I: alarm ID incremented to 11682
16:21:56.436 -----
16:21:57.443 -----
16:21:57.448 I: com loop duration: 188353ms
16:21:57.448 -----
16:21:59.847 I: (#0) RX  27ms | 27 CH23 | 95 01
16:21:59.849 I: (#0) RX  91ms | 27 CH23 | 95 02
16:21:59.851 I: (#0) RX 194ms | 27 CH23 | 95 04
16:21:59.852 I: (#0) RX 235ms | 27 CH23 | 95 05
16:21:59.854 I: (#0) RX 339ms | 27 CH23 | 95 07
16:21:59.855 I: (#0) RX 380ms | 27 CH23 | 95 08
16:22:00.974 I: (#0) RX  25ms | 27 CH23 | 95 03
16:21:59.086 I: (#0) RX  22ms | 27 CH23 | 95 06
16:22:00.190 I: (#0) RX  23ms | 19 CH23 | 95 89
16:22:00.193 I: (#0) Payload (134)
16:22:00.196 -----
16:22:00.197 I: com loop duration: 1103ms
16:22:00.198 -----
16:22:04.598 I: (#0) RX  18ms | 27 CH23 | 95 01
16:22:04.600 I: (#0) RX 124ms | 27 CH23 | 95 03
16:22:04.602 I: (#0) RX 186ms | 27 CH23 | 95 84
16:22:04.707 I: (#0) RX  14ms | 27 CH23 | 95 02
16:22:04.710 I: (#0) Payload (62)
16:22:04.712 I: alarm ID incremented to 11686
16:22:04.713 -----
16:22:04.714 I: com loop duration: 619ms
16:22:04.714 -----
16:22:09.848 -----
16:22:09.849 I: com loop duration: 753ms
16:22:09.850 -----
16:22:14.847 I: (#0) RX  60ms | 27 CH23 | 95 01
16:22:14.849 I: (#0) RX 150ms | 27 CH03 | 95 03
16:22:14.851 I: (#0) RX 191ms | 27 CH03 | 95 04
16:22:15.958 I: (#0) RX  41ms | 27 CH23 | 95 02
16:22:14.061 I: (#0) RX  53ms | 23 CH23 | 95 85
16:22:14.063 I: (#0) Payload (74)
16:22:14.065 -----
16:22:14.065 I: com loop duration: 971ms
16:22:14.066 -----
16:22:19.597 I: (#0) RX  68ms | 27 CH23 | 95 03
16:22:19.599 I: (#0) RX  96ms | 27 CH03 | 95 02
16:22:19.601 I: (#0) RX 200ms | 27 CH03 | 95 84
16:22:19.712 I: (#0) RX  45ms | 27 CH23 | 95 01
16:22:19.714 I: (#0) Payload (62)
16:22:19.715 I: alarm ID incremented to 11690
16:22:19.717 -----
16:22:19.717 I: com loop duration: 623ms
16:22:19.718 -----
16:22:24.847 I: (#0) RX  28ms | 27 CH23 | 95 01
16:22:24.850 I: (#0) RX  92ms | 27 CH23 | 95 02
16:22:24.852 I: (#0) RX 195ms | 27 CH23 | 95 04
16:22:24.853 I: (#0) RX 236ms | 23 CH23 | 95 85
16:22:25.971 I: (#0) RX  27ms | 27 CH23 | 95 03
16:22:25.973 I: (#0) Payload (74)
16:22:25.975 -----
16:22:25.976 I: com loop duration: 881ms
16:22:25.976 -----
16:22:29.641 I: (#0) RX  34ms | 27 CH23 | 95 01
16:22:29.644 I: (#0) RX  98ms | 27 CH23 | 95 02
16:22:29.645 I: (#0) RX 202ms | 27 CH23 | 95 84
16:22:29.761 I: (#0) RX  25ms | 27 CH23 | 95 03
16:22:29.764 I: (#0) Payload (62)
16:22:29.765 I: alarm ID incremented to 11694
16:22:29.767 -----
16:22:29.767 I: com loop duration: 629ms
16:22:29.768 -----
16:22:34.847 I: (#0) RX  38ms | 27 CH23 | 95 01
16:22:34.849 I: (#0) RX 102ms | 27 CH23 | 95 02
16:22:34.851 I: (#0) RX 206ms | 27 CH23 | 95 04
16:22:34.852 I: (#0) RX 247ms | 23 CH23 | 95 85
16:22:35.955 I: (#0) RX  14ms | 15 CH23 | 95 81
16:22:35.957 I: (#0) Payload (2)
16:22:35.959 -----
16:22:35.959 I: com loop duration: 865ms
16:22:35.960 -----
16:22:39.597 I: (#0) RX  33ms | 27 CH23 | 95 01
16:22:39.599 I: (#0) RX  97ms | 27 CH23 | 95 02
16:22:39.601 I: (#0) RX 201ms | 27 CH23 | 95 84
16:22:39.717 I: (#0) RX  26ms | 27 CH23 | 95 03
16:22:39.719 I: (#0) Payload (62)
16:22:39.720 I: alarm ID incremented to 11698
16:22:39.722 -----
16:22:39.723 I: com loop duration: 628ms
16:22:39.723 -----
```

### Error description

Hallo zusammen,

die Firmware 0.7.36 lief bei mir bisher immer am stabilsten. Alle 0.8.*** führten bisher immer zu sehr schlechten Funkstatistiken, bei denen die failed- und lost Raten höher waren, als die Rate für den erfolgreichen Empfang. Die aktuelle 0.8.48 funktioniert jedoch mal wieder einigermaßen stabil.

[image]

Bei einem 5 Sekunden Intervall für die Abfrage, erhalte ich ca. alle 10-50 Sekunden Antworten zurück (für mich vollkommen ausreichend). Jedoch schießen die Meldungen enorm in die Höhe. Seit heute morgen habe ich ca. 17.300 mal die Meldung "Time calibration" erhalten.

[image]

Bei der Firmware 0.7.36 hatte ich am Ende des Tages überhaupt keine Meldung. Liegt das daran, dass nur 3 von 4 Panels am Inverter angeschlossen sind?

Ich hoffe ihr könnt mir hier weiterhelfen... 

Viele Grüße
Exo1987

Dirk-Ahoy, 2024-01-08:
Hast du mehrere DTUs zum auslesen?

exo1987, 2024-01-08:
Guter Punkt... ich habe noch eine Hoymiles DTU im Einsatz. Das war aber bei 0.7.36 gar kein Thema. Da liefen beide parallel ohne Meldungen oder verlorene Pakete.

Dirk-Ahoy, 2024-01-08:
Das ist aber das Problem die Fehler 2 sind Zeitfehler - dh Pakete mit falschem Timestamp daher kommt das- in den älteren Versionen war das noch nicht drin bzw nicht ausgewertet.  Immer nur eine DTU laufen lassen oder mit den Fehlern leben. Habe den auch wenn ich mal wieder was teste.

exo1987, 2024-01-08:
Ok, danke für den wertvollen Tipp. Dann schalte ich mal die HM DTU ab...

stefan123t, 2024-01-09:
The description "Time calibration" for Event ID 2 was found in the Hoymiles S-Miles Installer application.
We have therefor renamed Event 2 from "DTU command failed" to "Time calibration" here https://github.com/lumapu/ahoy/commit/49ef94cb7a1435821816807697addb0108e1d00e in `hmInverter.h` as described in https://github.com/tbnobody/OpenDTU/issues/121#issuecomment-1752124208.

We still do not know for sure when / why this event occurs, but it most likely has something to do with the timing of request / responses between the AhoyDTU and your inverter (think of a heartbeat request). 
Some of the requests we send do have a timestamp included and bad reception between DTU and inverter can cause retransmission of the request or part of the message. Maybe in such cases the timestamp has to be updated / validated ?

If you are using two DTUs the collision of such timed messages are to be expected. 
Hence you likely get a lot of Error / Events in such cases.

lumapu, 2024-01-09:
hat es Besserung gezeigt?

exo1987, 2024-01-11:
Ja, es sind fast keine Meldungen mehr vorhanden... trotzdem aktualisiert er bei einem 5 Sekundenintervall weitaus weniger oft als bei der 0.7.36. Bei der 0.8.51 sind es ca. 45 Sekunden.

lumapu, 2024-01-11:
hast du hierzu nochmal ein Log, um es zu nachzuvollziehen?

Krankenkasse, 2024-01-28:
Habe das selbe problem. Habe auch eine original HM DTU noch im einsatz..

0.8.36 - 6fb0535
---------------
10:18:59.128 I: (#0) RX  31ms | 27 CH03 | 95 01
10:18:59.262 I: (#0) RX  35ms | 27 CH03 | 95 04
10:18:59.526 I: (#0) RX  31ms | 27 CH03 | 95 05
10:18:59.660 I: (#0) RX  32ms | 27 CH03 | 95 06
10:18:59.662 -----
10:18:59.663 I: com loop duration: 1551ms
10:18:59.663 -----
10:19:13.863 I: (#0) RX  88ms | 27 CH03 | 95 02
10:19:13.865 I: (#0) RX 149ms | 27 CH03 | 95 03
10:19:13.866 I: (#0) RX 194ms | 27 CH75 | 95 04
10:19:13.867 I: (#0) RX 256ms | 27 CH75 | 95 05
10:19:13.868 I: (#0) RX 297ms | 27 CH75 | 95 06
10:19:13.869 I: (#0) RX 340ms | 27 CH61 | 95 07
10:19:13.871 I: (#0) RX 400ms | 13 CH61 | 95 88
10:19:13.007 I: (#0) RX  31ms | 27 CH03 | 95 01
10:19:13.009 I: (#0) Payload (112)
10:19:13.010 -----
10:19:13.010 I: com loop duration: 899ms
10:19:13.010 -----
10:19:28.613 -----
10:19:28.614 I: com loop duration: 503ms
10:19:28.614 -----
10:19:43.613 I: (#0) RX  76ms | 27 CH03 | 95 02
10:19:43.615 I: (#0) RX 141ms | 23 CH03 | 95 83
10:19:43.750 I: (#0) RX  37ms | 27 CH03 | 95 01
10:19:43.751 I: (#0) Payload (42)
10:19:43.752 I: alarm ID incremented to 1953
10:19:43.753 -----
10:19:43.754 I: com loop duration: 643ms
10:19:43.754 -----
10:19:58.863 I: (#0) RX 111ms | 27 CH03 | 95 02
10:19:58.865 I: (#0) RX 152ms | 27 CH03 | 95 03
10:19:58.866 I: (#0) RX 259ms | 27 CH75 | 95 05
10:19:58.867 I: (#0) RX 300ms | 27 CH75 | 95 06
10:19:58.868 I: (#0) RX 362ms | 27 CH75 | 95 07
10:19:58.869 I: (#0) RX 405ms | 27 CH61 | 95 08
10:19:58.871 I: (#0) RX 445ms | 27 CH61 | 95 09
10:19:58.874 I: (#0) RX 507ms | 27 CH61 | 95 0a
10:19:58.877 I: (#0) RX 550ms | 27 CH40 | 95 0b
10:19:58.881 I: (#0) RX 611ms | 19 CH40 | 95 8c
10:19:59.412 I: (#0) RX  31ms | 27 CH03 | 95 01
10:19:59.679 I: (#0) RX  37ms | 27 CH03 | 95 04
10:19:59.681 I: (#0) Payload (182)
10:19:59.683 -----
10:19:59.683 I: com loop duration: 1572ms
10:19:59.684 -----
10:20:13.613 I: (#0) RX  38ms | 27 CH03 | 95 01
10:20:13.615 I: (#0) RX 103ms | 27 CH03 | 95 02
10:20:14.979 I: (#0) RX  31ms | 23 CH03 | 95 83
10:20:14.981 I: (#0) Payload (42)
10:20:14.982 I: alarm ID incremented to 1958
10:20:14.983 -----
10:20:14.983 I: com loop duration: 872ms
10:20:14.984 -----
10:20:28.863 I: (#0) RX  79ms | 27 CH03 | 95 02
10:20:29.996 I: (#0) RX  31ms | 27 CH03 | 95 01
10:20:29.133 I: (#0) RX  32ms | 27 CH03 | 95 03
10:20:29.268 I: (#0) RX  32ms | 27 CH03 | 95 04
10:20:29.502 I: (#0) RX  34ms | 23 CH03 | 95 85
10:20:29.504 I: (#0) Payload (74)
10:20:29.505 -----
10:20:29.505 I: com loop duration: 1395ms
10:20:29.506 -----
10:20:43.613 I: (#0) RX  73ms | 27 CH03 | 95 02
10:20:43.615 I: (#0) RX 138ms | 23 CH03 | 95 83
10:20:43.751 I: (#0) RX  33ms | 27 CH03 | 95 01
[... 142 lines cut ...]
10:24:43.863 I: (#0) RX  72ms | 27 CH03 | 95 02
10:24:43.865 I: (#0) RX 136ms | 27 CH03 | 95 03
10:24:43.866 I: (#0) RX 181ms | 27 CH75 | 95 04
10:24:43.867 I: (#0) RX 222ms | 27 CH75 | 95 05
10:24:43.868 I: (#0) RX 285ms | 27 CH61 | 95 06
10:24:43.869 I: (#0) RX 326ms | 27 CH61 | 95 07
10:24:43.871 I: (#0) RX 388ms | 27 CH61 | 95 08
10:24:43.874 I: (#0) RX 431ms | 27 CH40 | 95 09
10:24:43.878 I: (#0) RX 470ms | 15 CH40 | 95 8a
10:24:43.014 I: (#0) RX  32ms | 27 CH03 | 95 01
10:24:43.016 I: (#0) Payload (146)
10:24:43.017 -----
10:24:43.018 I: com loop duration: 907ms
10:24:43.018 -----
10:24:58.613 I: (#0) RX  66ms | 27 CH03 | 95 02
10:24:58.615 I: (#0) RX 131ms | 23 CH03 | 95 83
10:24:58.748 I: (#0) RX  32ms | 27 CH03 | 95 01
10:24:58.750 I: (#0) Payload (42)
10:24:58.751 I: alarm ID incremented to 1991
10:24:58.752 -----
10:24:58.752 I: com loop duration: 642ms
10:24:58.753 -----
10:25:13.863 -----
10:25:13.864 I: com loop duration: 753ms
10:25:13.864 -----

Dirk-Ahoy, 2024-01-28:
Keine andere DTU dabei - ist dann normal das die Fehler kommen !!!

closed 2024-02-17 by lumapu