lumapu/ahoy #1340: Download-Link zu Dev-Release geht seit 0.8.46 nicht mehr

BerziOnline, 2024-01-09:
Die letzte Version bei welcher der Link noch funktionierte war die 0.8.46.

Das Issue hier soll sich nicht damit befassen, damit ich weiß wo ich sie sonst herbekäme - es soll nur aufzeigen, dass seither der Link nicht mehr geht. 

[image]

Was mir aufgefallen ist, ist dass sich die Dateinamen seither in den Releases geändert haben:
https://fw.ahoydtu.de/fw/dev/ahoy_v0.8.46/ESP32/
=>
https://fw.ahoydtu.de/fw/dev/ahoy_v0.8.49/ESP32/

Eventuell haut das irgendeinen Ablauf kaputt, der dann die Zip-Datei für den Download-Link auf der Startseite erstellen müsste.

huaak, 2024-01-11:
..eigentlich fehlt im Link (ver 0.8.51) nur das /fw vor dem /dev ..also statt aktuell "https://fw.ahoydtu.de/dev/"
müsste es "https://fw.ahoydtu.de/fw/dev/" heißen..

Gubi2023, 2024-01-11:
war ich zu schnell, oder hat sich noch ein Fehler eingeschlichen??

[image]

lumapu, 2024-01-11:
bei mir geht's

Gubi2023, 2024-01-11:
Mit dem Handy lande ich hier.

Gubi2023, 2024-01-11:
war doch zu schnell, jetzt gehts. Danke!

closed 2024-02-17 by lumapu