lumapu/ahoy #1342: Nach Settings-Änderung DTU nicht mehr erreichbar... V0.8.49

Gubi2023, 2024-01-09:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.49

### Github Hash

???

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Setup-Änderung

### Error description

1.: in der deutschen Übersetzung wurde bei den Settings aus "Reset Max at midnight" plötzlich "lösche Max-Werte bei Sonnenuntergang" 
2. als ich die Settings änderte, erschien nach Safe eine neue Maske, die es bisher nicht gab.
3. nach Bestätigung versuchte das Programm plötzlich auf eine andere IP-Adresse  sich zu verbinden, die es nicht gab. Seither ist meine DTU nicht mehr erreichbar....
[image]

soelli22, 2024-01-09:
Hast du ein Display an deiner DTU? Damit habe ich habe bemerkt, das mit der 0.8.49 und .48 die DTU ständig neu bootet. (DTU war über web Zugriff kaum erreichbar.)
Mit der 0.8.45 läuft alles normal. Ich verwende ein ESP32. Wenn ich meine Inverter lösche, läuft es normal hoch. Bin zurück auf die .45, dann passt es.

Gubi2023, 2024-01-09:
nein, kein Display. DTU ist "tot",  blinkt auch nicht mehr....

lille2000, 2024-01-09:
Ich kenne die selbe Fehlermeldung von Firmware-Updates. An und zu brechen die FW Updates mit “Netzwerkzeitüberschreitung” ab. Da hilft auch kein Reboot der DTU, es bricht wieder nach ein paar Sekunden später ab. Irgendwann geht es dann und danach ist die DTU tot, da hilft nur noch neu aufsetzen per USB.  Danach geht wieder alles. 
ingesamt hatte ich jetzt 4x den Fall.

Gubi2023, 2024-01-10:
2. als ich die Settings änderte,(im Inverter-Menue) erschien nach Safe eine neue Maske, die es bisher nicht gab, sieht so aus.
[image]
Nach Drücken der Next-Taste kam obige Fehlermaske. Meine alte IP-Adresse war http://192.168.178.60

Gubi2023, 2024-01-10:
du meinst im WLAN-Namen? hat bisher noch nie gestört: 
[image]

Gubi2023, 2024-01-10:
[image]

knickohr, 2024-01-10:
Das stört vielleicht die Fritte nicht, aber die DTU

Gubi2023, 2024-01-10:
die DTU läuft seit 1 Jahr mit diesem WLAN. Jetzt stört es plötzlich??

lumapu, 2024-01-10:
nein, die SSID ist nicht das Problem. Die obige Seite (Setup Wizard) wird aufgerufen, wenn man hinter der IP irgendwas eingibt, das der ESP nicht zuordnen kann, zB. http://192.168.178.60/Gubi2023
Der Next Button ist fest auf die IP `192.168.4.1` kodiert, da diese Seite nur für die Ersteinrichtung gedacht ist.

Gubi2023, 2024-01-12:
habe jetzt die 8.52 installiert. Läuft anfangs problemlos, aber sobald ich irgendwelche Settings ändere, ist die DTU wieder scheintot 
siehe https://github.com/lumapu/ahoy/issues/1345#issuecomment-1887229972    
ich habe Minimalkonfig, also ESP8266 ohne Display

Gubi2023, 2024-01-12:
zurück auf 8.44: Settingsänderungen problemlos möglich.

lumapu, 2024-01-18:
wie sieht es mit neueren Versionen aus, im besten Fall mit der aktuell neuesten?

Gubi2023, 2024-01-18:
meine DTU ist gestern abgeraucht, kann erst morgen wieder testen....

lumapu, 2024-01-18:
oh was heißt das? FW oder HW?

Gubi2023, 2024-01-18:
mein NRF-Modul hatte den Geist aufgegeben, ich hoffe das neue tutet.

Gubi2023, 2024-01-19:
8.59: Änderungen der Settings jetzt problemlos möglich

closed 2024-01-19 by Gubi2023