lumapu/ahoy #1343: Fehlermeldungen werden nicht übersetzt(?)

Gubi2023, 2024-01-09:
Fehlermeldungen werden nicht übersetzt(?)
[image]

_Originally posted by @Gubi2023 in https://github.com/lumapu/ahoy/issues/1335#issuecomment-1883500761_

Gubi2023, 2024-01-09:
Version 0.8.49

lumapu, 2024-01-09:
ja genau, bisher wurden nur die statischen HTML Seiten übersetzt, die anderen Themen folgen bald, erfordern aber noch eine Struktur die heute noch nicht da ist.
Der issue ist gut für alle, denen das bis dahin auch auffällt. Ich denke die Deutsche Version ist für viele schon viel einfacher zu verstehen als bisher.

Gubi2023, 2024-01-10:
8.50.: Sieht bisher gut aus, ich lass ihn mal noch offen, falls noch Fehler auftauchen...

Gubi2023, 2024-01-11:
[image]

Werden Ereignismeldungen auch noch übersetzt?

lumapu, 2024-01-11:
Freiwillige vor 😇

closed 2024-02-17 by lumapu