lumapu/ahoy #1345: Ständiger Neustart mit 0.8.49

soelli22, 2024-01-09:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.49

### Github Hash

3aa4751

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Intervall 15sec

### Error description

Hallo zusammen,

ich habe ein Problem sowohl mit der 0.8.49 als auch .48
Sobald ich diese Versionen installiere, startet die DTU immer neu nach etwa 3-5 Sekunden.
Getestet habe ich sowohl Original, als auch de und prometheus.
Lösche ich die WR alle (2x HM 800) oder schalte das nRF24 ab, dann läuft alles, es kommen nur verständlicher Weise keine Daten.
Ein Rückfall auf 0.8.45 lässt alles wie gewohnt laufen.

dtuuser, 2024-01-09:
Bei mir mit .48 und ESP32 seit der Installation einen Tag und 22 Stunden stabil ohne Neustart

knickohr, 2024-01-09:
Ein Serial Debus wäre jetzt hilfreich.

lumapu, 2024-01-09:
kann mir vorstellen, dass z.B. eine der LEDs auf einem nicht PWM Pin konfiguriert wurde - ich weiß nicht was dann passiert.

lumapu, 2024-01-09:
hier gibt es ähnliche Probleme #1342

knickohr, 2024-01-10:
> kann mir vorstellen, dass z.B. eine der LEDs auf einem nicht PWM Pin konfiguriert wurde - ich weiß nicht was dann passiert.

Das ist sehr gut möglich 😱

soelli22, 2024-01-10:
Ich habe nun mal fix die 0.8.50 versucht, das endet im gleichen Problem.
leider habe ich 2 linke Hände mit je 5 Daumen. (Zumindest bei solchen Themen. 
Deshalb: Wie erstelle ich ein serial debug ???
Als Anhang mal 3 Bilder. Wenn das nRF aktiviert ist, Läuft die Webkonsole genau so weit, wie auf Bild 1 ersichtlich, dann kommt immer der Reboot.
[image]
[image]
[image]

soelli22, 2024-01-10:
Aus der Web Konsole kommt maximal so ein Block, dann erfolgt immer der Reboot.

>

--------------------------------
Welcome to AHOY!

point your browser to http://192.168.178.25 (Station)
to configure your device
--------------------------------

Wifi event: 11
Wifi event: 10
Wifi event: 11
[WiFi] AP disabled
[WiFi] mDNS established: AHOY-DTU.local
I: [NTP]: getNtpTime failed
08:49:13.620 I: [NTP]: 2024-01-10 07:49:13 UTC
08:49:13.622 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
08:49:13.623 I: (#0) TX 27 CH23 | 15  *  *  *  * 85 08 79 52 80 0b 00 65 9e 4b f9 00 00 00 00 00 00 00 00 c9 f9 d6 
08:49:14.138 I: (#0) request timeout: 500ms
08:49:14.138 -----
08:49:14.638 I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
08:49:14.639 I: (#1) TX 27 CH23 | 15  *  *  *  * 85 08 79 52 80 0b 00 65 9e 4b fa 00 00 00 00 00 00 00 00 39 ed 8e 
08:49:15.143 I: (#1) RX  70ms | 27 CH61 | 95  *  *  *  *  *  *  *  * 02 a7 36 00 05 eb 1a 00 01 00 02 09 39 13 86 00 6c 38 
08:49:15.146 I: (#1) RX 135ms | 23 CH61 | 95  *  *  *  *  *  *  *  * 83 00 00 00 05 03 e8 00 60 00 01 1f 1c 9a 
08:49:15.158 W: (#1) frame 1 missing: request retransmit (8 attempts left)
08:49:15.159 I: (#1) TX 11 CH23 | 15  *  *  *  * 85 08 79 52 81 1a 
08:49:15.291 I: (#1) RX  31ms | 27 CH61 | 95  *  *  *  *  *  *  *  * 01 00 01 01 56 00 0d 00 2c 01 5a 00 14 00 45 00 02 eb 
08:49:15.294 I: (#1) Payload (42): 00 01 01 56 00 0d 00 2c 01 5a 00 14 00 45 00 02 a7 36 00 05 eb 1a 00 01 00 02 09 39 13 86 00 6c 00 00 00 05 03 e8 00 60 00 01 
08:49:15.309 I: alarm ID incremented to 1
08:49:15.310 -----
08:49:15.810 I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 0
08:49:15.811 I: (#0) TX 27 CH40 | 15  *  *  *  * 85 08 79 52 80 0b 00 65 9e 4b fb 00 00 00 00 00 00 00 00 a9 e0 ad

jimknopf63, 2024-01-10:
Bei mir läuft die 0.8.49 seit 24 Std ohne Neustart... 2 WR der HM Serie und ESP32 mit 1,3Zoll Display

lumapu, 2024-01-10:
dann bräuchten wir jetzt ein USB Log, soweit sieht alles richtig aus.

knickohr, 2024-01-10:
Nach dem Umstellen der Settings und Save war die DTU tot 😢 Sie machte zwar noch einen AP auf, aber man kam nicht mehr rein 😩 Das Ding hängt sich irgendwo beim Lesen der Settings auf. Also habe ich die letzten 2 Stunden damit verbracht ganz von Vorne anzufangen. Erase, Flash über die IDF, Neueinrichtung … Und ja, ich mußte zurück bis zur 47 um meine Settings wieder zu speichern.

Die beiden Issues sind gerechtfertigt, er zerschießt offenbar wieder die JSON-Settings oder liest sie nicht mehr korrekt ein. Denke aber ersteres ist der Fall, ansonsten wäre die DTU ja nicht beim Speichern abgekackt.

You69Man, 2024-01-10:
Können bitte alle, bei denen so ein zyklischer Neustart auftritt bitte dazu angeben, ob sie ein Display aktiviert haben oder nicht. 
Um ausschließen (oder bestätigen) zu können, dass es aus dieser Ecken kommen kann.
Danke!

knickohr, 2024-01-10:
Bei mir h#Not ein ePaper dran, wie schon seit jeher 😉

soelli22, 2024-01-10:
Ich habe ein 1.3Zoll SSH1106.

dtuuser, 2024-01-10:
Keine Reboots : SSD1306 Display

You69Man, 2024-01-10:
> Bei mir hängt ein ePaper dran, wie schon seit jeher 😉

Hast du auch die reboots?

knickohr, 2024-01-10:
Nein, erst nach dem die Settings in die Hose gingen. Dann rebootete die DTU und kam nicht mal mehr in den AP.

You69Man, 2024-01-10:
Hab mal was versucht, falls es mit dem neuen display graph zu tun haben sollte:  #1347

soelli22, 2024-01-11:
Leider hilft die 0.8.51 auch nicht. Nach dem Aufspielen hatte ich den Eindruck, es läuft. Einmal in den Einstellungen was verändert und wieder in der Reboot Schleife. Display aus, und alles läuft wie gewohnt.

HorstyS, 2024-01-11:
also, ich habe diese probleme nicht. ich habe zwei AhoyDTU gleichzeitig die einen HM-700 abfragen. beide ESP32, einer mit einem SH1106 OLED, die andere mit  einem ePaper. beide habe ich die letzten tage jeweils seit der .46 bis heute zur 0.8.51 hochgezogen (EN version). der OLED liefert auch noch MQTT daten. bis dato alles stabil.

You69Man, 2024-01-11:
> Leider hilft die 0.8.51 auch nicht. Nach dem Aufspielen hatte ich den Eindruck, es läuft. Einmal in den Einstellungen was verändert und wieder in der Reboot Schleife. Display aus, und alles läuft wie gewohnt. 

Kannst bitte Mal versuchen das Display einzuschalten, aber die Graph Ratio auf 0 zu setzen?

BerziOnline, 2024-01-11:
Ich bin soeben von der 0.8.50 auf die 0.8.51 und meine ahoyDTU startet seither im Sekundentakt neu. Ich komme immer ganz kurz aufs webfrontend, aber dann startet sie auch schon wieder neu. Mist - wie bekomme ich sie möglichst simpel wieder auf die 0.8.50?

You69Man, 2024-01-11:
> Ich bin soeben von der 0.8.50 auf die 0.8.51 und meine ahoyDTU startet seither im Sekundentakt neu. Ich komme immer ganz kurz aufs webfrontend, aber dann startet sie auch schon wieder neu. Mist - wie bekomme ich sie möglichst simpel wieder auf die 0.8.50?

Woran erkennst du, dass die DTU im Sekundentakt neu startet? Hast du ein Display? Oder über USB Serielle? Webserial?

BerziOnline, 2024-01-11:
> Woran erkennst du, dass die DTU im Sekundentakt neu startet? Hast du ein Display? Oder über USB Serielle? Webserial?

Ja, ich habe eine fertige AhoyDTU mit Case und Display gekauft gehabt. Dort sehe ich, dass sie neustartet und auch über das Verhalten des Webfrontends - immer kurz da, dann nicht mehr - ebenfalls habe ich im Smarthome einen Verfügbarkeitscheck per Ping. Die DTU geht im Sekundentakt hoch und runter. Es sind aber keine Verbindungsprobleme (deshalb vmtl die Frage nach dem Display)

Ich musste sie nun komplett vom Strom nehmen. Bekomme ich die wieder ans Leben? Das Webfrontend ist immer für ca. 5 Sekunden da und reagiert auch. Nur für ein Downgrade reicht die Zeit nicht.

You69Man, 2024-01-11:
K

> Ja, ich habe eine fertige AhoyDTU mit Case und Display gekauft gehabt. Dort sehe ich, dass sie neustartet und auch über das Verhalten des Webfrontends - immer kurz da, dann nicht mehr - ebenfalls [...]

Der Webinstaller unter  https://ahoydtu.de/getting_started/ "Firmware flashen" müsste die DTU eigentlich wieder einfangen können.

BerziOnline, 2024-01-11:
@You69Man Danke dir für den Hinweis! Ist sie dadurch komplett resettet, sprich ich muss sie wieder komplett neu einrichten oder sollte sich das Flashen so verhalten, wie wenn ich die .bin über das Update-Feld eingespielt habe?

HorstyS, 2024-01-11:
tatsächlich hatte ich mal mal das verhalten mit dem ständigen reboot, fällt mir gerade ein. aber das war noch vor einer 0.8.4x soweit ich mich erinnere.
ich habe dann auch wie von @You69Man beschrieben über den web-installer alles gelöscht und installiert. danach die aktuellste DEV version installiert und alles manuell neu konfiguriert. dauert ja nicht lange und ist sicherer als ein vielleicht altes backup einzuspielen das nicht mehr ganz passt.

EinLaie, 2024-01-11:
Ich habe das gleiche Problem. Ich habe es geschaft in der kurzen Online-Zeit, das Display SH1106 auf "none" zu setzen. Danach meldet sich die DTU normal und du kannst einen Downgrade machen. Der Fehler hat definitiv mit dem Display zutun.

BerziOnline, 2024-01-11:
@EinLaie Ohhhh yeah! Das hat geholfen - du bist mein Held! Danke dir!

EinLaie, 2024-01-11:
Beim Flashen hatte ich vor Tagen ein anderes Problem. Ich konnte die Firmware nicht aufspielen. Fehler 4, zu weinig Speicherplatz. Musste nach dem Flashen erst eine andere .bin einspielen, damit ich die Ahoy wieder drauf bekam. Das Installations-Wlan hat auch dann eine andere SSID. Ansonsten geht's dann weiter wie immer.

Gubi2023, 2024-01-11:
https://github.com/lumapu/ahoy/issues/1345#issuecomment-1886800006

bei mir ebenso. 8.51 war ca.10 Min erreichbar, jetzt "scheintot"

Gubi2023, 2024-01-11:
[image]

jimknopf63, 2024-01-11:
Nachtrag, die 0.8.50 läuft einwandfrei auf ESP32 mit 1,3 Zoll Display. Die 08.51 ließ sich installieren bekam aber keine gültige Zeit und somit keine Verbindung zum WR. Zurück auf die 0.8.50 ging leider nicht mehr, nach reboot war die Ahoy nicht mehr per IP zu erreichen. Auch nicht über AP WLAN ahoydtu. Habe es per USB Kabel dann über ahoydtu.de neu geflasht mit löschen, Update auf 0.8.50, config wieder eingespielt und läuft seitdem wieder.

You69Man, 2024-01-11:
> 
> 
> > Kannst bitte Mal versuchen das Display einzuschalten, aber die Graph Ratio auf 0 zu setzen?
> 
> Hilft leider nicht, bringt mich wieder in die Reboot Schleife.

Du hast gemeint, ohne Display läuft es bei dir, richtig? Kannst du dann bitte Mal versuchen, die Konfiguration auf Werkseinstellungen zu setzen, und nach Reboot noch einmal alles probieren?

soelli22, 2024-01-11:
> Du hast gemeint, ohne Display läuft es bei dir, richtig? Kannst du dann bitte Mal versuchen, die Konfiguration auf Werkseinstellungen zu setzen, und nach Reboot noch einmal alles probieren?

Richtig. Sobald Display auf "aus" steht, läuft es normal. Weder Reset AhoyDTU, noch Reset auf Werkseinstellungen (nicht WIFI) hilft. Alle Daten wurden nochmals händisch eingetragen (WLAN Daten, Wechselrichter ect.)

HorstyS, 2024-01-11:
Und was passiert wenn du das Display nun wieder einschaltest? Ich würde vermuten, dass es dann wieder normal mit der DTU läuft. Mehr Gefühl sagt mir es liegt irgendwie an den Einstellungen die nach dem Update korrupt sind. Nach einem erase, Neuinstallation und manuellem eintragen der settings läuft es bei mir wieder super mit den Displays.

soelli22, 2024-01-11:
> Und was passiert wenn du das Display nun wieder einschaltest? Ich würde vermuten, dass es dann wieder normal mit der DTU läuft.

Leider nicht! Versucht habe ich jetzt noch WebInstaller mit Erase Device auf 0.8.36. Dann auf 0.8.51 gleiches Problem.
Das Ganze noch einmal mit Update auf 0.8.46, läuft einwandfrei. Alles irgendwie komisch...

soelli22, 2024-01-11:
>Wifi event: 1
BSSID 0: fc ec da 8a 9b 90
BSSID 1: 78 8a 20 b4 f0 87
try to connect to AP with BSSID: fc ec da 8a 9b 90
Wifi event: 4
I: connectionEvent

[WiFi] Connected
Wifi event: 7
I: connectionEvent

--------------------------------
Welcome to AHOY!

point your browser to http://192.168.178.25 (Station)
to configure your device
--------------------------------

Wifi event: 11
Wifi event: 10
[WiFi] AP disabled
Wifi event: 11
[WiFi] mDNS established: AHOY-DTU.local
I: [NTP]: 2024-01-11 18:40:14 UTC
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 0, s: 0, f: 0, n: 0 | p: 0
I: (#0) TX 27 CH23 | 15  *  *  *  * 85 08 79 52 80 0B 00 65 A0 36 0F 00 00 00 00 00 00 00 00 54 8E 89 
I: (#0) request timeout: 500ms
-----
I: com loop duration: 506ms
-----
I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 0
I: (#0) TX 27 CH40 | 15  *  *  *  * 85 08 79 52 80 0B 00 65 A0 36 10 00 00 00 00 00 00 00 00 64 02 2A 
I: (#0) RX  71ms | 27 CH75 | 95  *  *  *  *  *  *  *  * 02 3D 85 00 04 C0 AE 00 4A 00 47 09 52 13 8A 09 7D FE 
I: (#0) RX 135ms | 23 CH75 | 95  *  *  *  *  *  *  *  * 83 00 02 00 66 03 E8 00 C4 00 02 ED 22 90 
W: (#0) frame 1 missing: request retransmit (8 attempts left)
I: (#0) TX 11 CH40 | 15  *  *  *  * 85 08 79 52 81 A5 
I: (#0) RX  32ms | 27 CH75 | 95  *  *  *  *  *  *  *  * 01 00 01 01 4E 01 87 05 19 01 4E 01 75 04 DF 00 06 A6 
I: (#0) Payload (42): 00 01 01 4E 01 87 05 19 01 4E 01 75 04 DF 00 06 3D 85 00 04 C0 AE 00 4A 00 47 09 52 13 8A 09 7D 00 02 00 66 03 E8 00 C4 00 02 
I: alarm ID incremented to 2
-----
I: com loop duration: 678ms
-----
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d5f4a  PS      : 0x00060e30  A0      : 0x800d6702  A1      : 0x3ffb20f0  
A2      : 0x3ffdd3a4  A3      : 0x4372e666  A4      : 0x3ffb20fc  A5      : 0x3ffaff00  
A6      : 0x00000001  A7      : 0x3ffb0320  A8      : 0x9a5fc9ef  A9      : 0x000000ff  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000400  A13     : 0x3ffc9e54  
A14     : 0x00000003  A15     : 0x00000040  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x40084bc5  LEND    : 0x40084bcd  LCOUNT  : 0x00000027  

Backtrace: 0x400d5f47:0x3ffb20f0 0x400d66ff:0x3ffb2120 0x400ddbce:0x3ffb2170 0x400ddc78:0x3ffb21c0 0x40196719:0x3ffb21e0 0x400eb235:0x3ffb2200 0x400ee782:0x3ffb2270 0x40107f69:0x3ffb2290

ELF file SHA256: d9022f872ada7514

E (16149) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13132
load:0x40080400,len:3036
entry 0x400805e4
E (518) esp_core_dump_flash: No core dum���ѥѥ���found!
E (518) esp_core_dump_flash: No core dump partition found!
I: Initializing FS ..
I:  .. done
I: Config Version: 8
I: Settings valid: true
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1000 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry attempts made for
    last transmission	= 15
Multicast		= Allowed
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Enabled
Primary Mode		= RX
TX address		= 0x5443018101
pipe 0 (closed) bound	= 0x5443018101
pipe 1 ( open ) bound	= 0x5279088501
pipe 2 (closed) bound	= 0xc3
pipe 3 (closed) bound	= 0xc4
pipe 4 (closed) bound	= 0xc5
pipe 5 (closed) bound	= 0xc6
I: DTU_SN: 85087952

---------
AhoyDTU Info:
Version: 0.8.51
Github Hash: 9f39e5c

---------
AP MODE
SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

Wifi event: 0
Wifi event: 2
Wifi event: 10
Wifi event: 11
Wifi event: 10
connect to network 'xxxxxxxxxxxxxx' ...
I: added inverter HM xxxxxxxxxxxxxx
scanning APs with SSID xxxxxxxxxxx
Wifi event: 1
BSSID 0: fc ec da 8a 9b 90
BSSID 1: 78 8a 20 b4 f0 87
try to connect to AP with BSSID: fc ec da 8a 9b 90
Wifi event: 4
I: connectionEvent

[WiFi] Connected
Wifi event: 7
I: connectionEvent

hilft das weiter???

lumapu, 2024-01-11:
@You69Man das habe ich in Log gesehen `(IntegerDivideByZero)`.
Muss nicht von dir kommen, ich mache man einen review des codes ab Version `0.8.46`. Evtl. hast du ja eine Idee 😊

You69Man, 2024-01-11:
> @You69Man das habe ich in Log gesehen `(IntegerDivideByZero)`. Muss nicht von dir kommen, ich mache man einen review des codes ab Version `0.8.46`. Evtl. hast du ja eine Idee 😊

Die Graph Skalierung hätte in Sonderfällen durchaus das Potential für DivByZero, aber eigentlich hätte ich geschaut, alle abzufangen. Vielleicht hab ich was übersehen. Ich kann mir morgen Vormittag mal etwas Zeit nehmen, falls du nicht schon vorher was entdeckst. Danke!

You69Man, 2024-01-11:
@lumapu: 
Weißt du wie man aus dem backtrace der guru mediation auf die zeile im c-code zurückschließen kann?
Backtrace: 0x400d5f47:0x3ffb20f0 0x400d66ff:0x3ffb2120 0x400ddbce:0x3ffb2170 0x400ddc78:0x3ffb21c0 0x40196719:0x3ffb21e0 0x400eb235:0x3ffb2200 0x400ee782:0x3ffb2270 0x40107f69:0x3ffb2290
Ich hätte es einmal mit dem xtensa-esp32-elf-objdump versucht, aber da bekomme ich eine Fehlermeldung "*** Couldn't reserve space for cygwin's heap, Win32 error 0"

You69Man, 2024-01-11:
> hilft das weiter???
Danke, so etwas kann immer nützlich sein!

lumapu, 2024-01-11:
ich hab das beim ESP32 nicht nicht gemacht, am besten ging es, wenn der Fehler direkt bei mir auftrat, da der Decoder direkt in VScode drin ist

You69Man, 2024-01-11:
> ich hab das beim ESP32 nicht nicht gemacht, am besten ging es, wenn der Fehler direkt bei mir auftrat, da der Decoder direkt in VScode drin ist

Kannst du diesen Fehler (reboot loop) denn bei dir reproduzieren? Ich hab ihn bei mir noch nie gesehen, weder auf ESP8266 noch auf ESP32, weder mit noch ohne Display...
Kann es damit zusammenhängen, dass ich die plugin config übergabe von vielen Einzelparametern auf einen Strukturpointer reduziert habe? Ich könnte es mir zwar nicht erklären, aber wer weiß...

You69Man, 2024-01-11:
> Leider nicht! Versucht habe ich jetzt noch WebInstaller mit Erase Device auf 0.8.36. Dann auf 0.8.51 gleiches Problem. Das Ganze noch einmal mit Update auf 0.8.46, läuft einwandfrei. Alles irgendwie [...]

Kannst du auch zu .47 und .48 was sagen? Ab welcher Version tritt der Fehler genau auf? Danke!

lumapu, 2024-01-11:
@You69Man ich habe einige Divisonen jetzt noch abgesichert, kommt mit der `0.8.52` (heute Nacht) 😉

```diff
diff --git a/src/plugins/Display/Display_Mono.h b/src/plugins/Display/Display_Mono.h
index 008f737..cb1ecee 100644
--- a/src/plugins/Display/Display_Mono.h
+++ b/src/plugins/Display/Display_Mono.h
@@ -166,7 +166,10 @@ class DisplayMono {
         }
 
         uint8_t sss2pgpos(uint seconds_since_start) {
-            return(seconds_since_start * (mPgWidth - 1) / (mDisplayData->pGraphEndTime - mDisplayData->pGraphStartTime));
+            uint32_t diff = (mDisplayData->pGraphEndTime - mDisplayData->pGraphStartTime);
+            if(diff)
+                return (seconds_since_start * (mPgWidth - 1) / diff);
+            return 0
         }
 
         void calcPowerGraphValues() {
@@ -175,6 +178,8 @@ class DisplayMono {
             mPgTimeOfDay = (mDisplayData->utcTs > mDisplayData->pGraphStartTime) ? mDisplayData->utcTs - mDisplayData->pGraphStartTime : 0; // current time of day with respect to current sunrise time
             if (oldTimeOfDay > mPgTimeOfDay) // new day -> reset old data
                 resetPowerGraph();
+            if(0 == mPgPeriod)
+                mPgPeriod = 1;
             mPgLastPos = std::min((uint8_t) (mPgTimeOfDay * (mPgWidth - 1) / mPgPeriod), (uint8_t) (mPgWidth - 1));  // current datapoint based on currenct time of day
         }
 
@@ -190,15 +195,13 @@ class DisplayMono {
         uint8_t getPowerGraphXpos(uint8_t p) {
             if ((p <= mPgLastPos) && (mPgLastPos > 0))
                 return((p * (mPgWidth - 1)) / mPgLastPos);  // scaling of x-axis
-            else
-                return(0);
+            return 0;
         }
 
         uint8_t getPowerGraphYpos(uint8_t p) {
-            if (p < mPgWidth)
+            if ((p < mPgWidth) && (mPgMaxPwr > 0))
                 return((mPgData[p] * (uint32_t) mPgHeight / mPgMaxPwr)); // scaling of data to graph height
-            else
-                return(0);
+            return 0;
         }
 
         void plotPowerGraph(uint8_t xoff, uint8_t yoff) {
diff --git a/src/plugins/Display/Display_Mono_128X32.h b/src/plugins/Display/Display_Mono_128X32.h
index 6ab21a6..e904769 100644
--- a/src/plugins/Display/Display_Mono_128X32.h
+++ b/src/plugins/Display/Display_Mono_128X32.h
@@ -1,5 +1,5 @@
 //-----------------------------------------------------------------------------
-// 2023 Ahoy, https://ahoydtu.de
+// 2024 Ahoy, https://ahoydtu.de
 // Creative Commons - https://creativecommons.org/licenses/by-nc-sa/4.0/deed
 //-----------------------------------------------------------------------------
 
@@ -107,7 +107,7 @@ class DisplayMono128X32 : public DisplayMono {
         void printText(const char *text, uint8_t line) {
             setFont(line);
 
-            uint8_t dispX = mLineXOffsets[line] + pixelShiftRange / 2 + mPixelshift;
+            uint8_t dispX = mLineXOffsets[line] + (pixelShiftRange / 2) + mPixelshift;
 
             if (isTwoRowLine(line)) {
                 String stringText = String(text);
diff --git a/src/plugins/Display/Display_Mono_128X64.h b/src/plugins/Display/Display_Mono_128X64.h
index 2d5c13c..34f3583 100644
--- a/src/plugins/Display/Display_Mono_128X64.h
+++ b/src/plugins/Display/Display_Mono_128X64.h
@@ -193,13 +193,13 @@ class DisplayMono128X64 : public DisplayMono {
             mDisplay->setFont(u8g2_font_ncenB10_symbols10_ahoy);
             char sym[]=" ";
             sym[0] = mDisplayData->RadioSymbol?'A':'E';                 // NRF
-            mDisplay->drawStr(widthShrink / 2 + mPixelshift, mLineYOffsets[l_RSSI], sym);
+            mDisplay->drawStr((widthShrink / 2) + mPixelshift, mLineYOffsets[l_RSSI], sym);
 
             if (mDisplayData->MQTTSymbol)
                 sym[0] = 'J'; // MQTT
             else
                 sym[0] = mDisplayData->WifiSymbol?'B':'F';              // Wifi
-            mDisplay->drawStr(mDispWidth - mDisplay->getStrWidth(sym) - widthShrink / 2 + mPixelshift, mLineYOffsets[l_RSSI], sym);
+            mDisplay->drawStr(mDispWidth - mDisplay->getStrWidth(sym) - (widthShrink / 2) + mPixelshift, mLineYOffsets[l_RSSI], sym);
             mDisplay->sendBuffer();
 
             mExtra++;
@@ -241,8 +241,8 @@ class DisplayMono128X64 : public DisplayMono {
                 mLineYOffsets[i] = yOff;
                 dsc = mDisplay->getDescent();
                 yOff -= dsc;
-                if (l_Time == i)   // prevent time and status line to touch
-                    yOff++;     // -> one pixels space
+                if (l_Time == i) // prevent time and status line to touch
+                    yOff++;      // -> one pixels space
                 i++;
             } while(l_MAX_LINES>i);
         }
diff --git a/src/plugins/Display/Display_Mono_64X48.h b/src/plugins/Display/Display_Mono_64X48.h
index 68aa3cc..a4ddc6a 100644
--- a/src/plugins/Display/Display_Mono_64X48.h
+++ b/src/plugins/Display/Display_Mono_64X48.h
@@ -96,7 +96,7 @@ class DisplayMono64X48 : public DisplayMono {
         }
 
         void printText(const char *text, uint8_t line) {
-            uint8_t dispX = mLineXOffsets[line] + pixelShiftRange/2 + mPixelshift;
+            uint8_t dispX = mLineXOffsets[line] + pixelShiftRange / 2 + mPixelshift;
 
             setFont(line);
             mDisplay->drawStr(dispX, mLineYOffsets[line], text);
diff --git a/src/plugins/Display/Display_Mono_84X48.h b/src/plugins/Display/Display_Mono_84X48.h
index d9f1d98..175fa17 100644
--- a/src/plugins/Display/Display_Mono_84X48.h
+++ b/src/plugins/Display/Display_Mono_84X48.h
@@ -1,5 +1,5 @@
 //-----------------------------------------------------------------------------
-// 2023 Ahoy, https://ahoydtu.de
+// 2024 Ahoy, https://ahoydtu.de
 // Creative Commons - https://creativecommons.org/licenses/by-nc-sa/4.0/deed
 //-----------------------------------------------------------------------------
```

lumapu, 2024-01-11:
> Kann es damit zusammenhängen, dass ich die plugin config übergabe von vielen Einzelparametern auf einen Strukturpointer reduziert habe? Ich könnte es mir zwar nicht erklären, aber wer weiß...

nein, das glaube ich nicht, ich finde das sehr gut und konsquent wie du es geändert hast - du bist auch einer der Contributoren, deren Code man (fast) ohne Änderungen übernehmen kann - sehr cool 😊

soelli22, 2024-01-12:
> Kannst du auch zu .47 und .48 was sagen? Ab welcher Version tritt der Fehler genau auf? Danke!

Ich glaube, mit der .47 ging es los. Habe nun die neue .52 drauf, und........... es läuft!!! Ihr seid meine Helden!
Ich baue nun mal alles Zusammen und lasse es bis heute Abend in Ruhe laufen, aber es wirkt alles wieder stabil!

soelli22, 2024-01-12:
```
0.8.52 - 455d29a
---------------
webSerial, buffer overflow!

08:40:01.252 I: PV-Anlage 2/ch0/F_AC: 50.020 Hz
08:40:01.253 I: PV-Anlage 2/ch0/PF_AC: 1.001 
08:40:01.254 I: PV-Anlage 2/ch0/Temp: 9.300 °C
08:40:01.255 I: PV-Anlage 2/ch0/ALARM_MES_ID: 1.000 
08:40:01.256 I: PV-Anlage 2/ch0/YieldTotal: 564.059 kWh
08:40:01.267 I: PV-Anlage 2/ch0/P_DC: 4.200 W
08:40:01.268 I: PV-Anlage 2/ch0/Efficiency: 97.619 %
08:40:01.268 I: PV-Anlage 2/ch0/MaxPower: 4.100 W
08:40:01.269 I: 
08:40:01.270 I: (#0) Radio infos: -6 -6 4 -6 -6 | t: 96, s: 91, f: 0, n: 5 | p: 2
08:40:01.282 I: (#0) TX 27 CH40 | 15  *  *  *  * 85 08 79 52 80 0b 00 65 a0 ec d1 00 00 00 00 00 00 00 00 78 7b 54 
08:40:01.787 I: (#0) RX  40ms | 27 CH75 | 95  *  *  *  *  *  *  *  * 01 00 01 01 46 00 08 00 1a 01 40 00 06 00 13 00 06 92 
08:40:01.791 I: (#0) RX  86ms | 27 CH61 | 95  *  *  *  *  *  *  *  * 02 3d f2 00 04 c1 26 00 00 00 00 08 fa 13 89 00 2a f9 
08:40:01.804 W: (#0) frame 3 missing: request retransmit (8 attempts left)
08:40:01.805 I: (#0) TX 11 CH40 | 15  *  *  *  * 85 08 79 52 83 a7 
08:40:01.908 I: (#0) RX  37ms | 23 CH75 | 95  *  *  *  *  *  *  *  * 83 00 00 00 02 03 ea 00 5c 00 01 2e 8e 00 
08:40:01.911 I: (#0) Payload (42): 00 01 01 46 00 08 00 1a 01 40 00 06 00 13 00 06 3d f2 00 04 c1 26 00 00 00 00 08 fa 13 89 00 2a 00 00 00 02 03 ea 00 5c 00 01 
08:40:02.928 -----
I_DC: 0.050 A
08:40:01.206 I: PV-Anlage 2/ch1/P_DC: 1.600 W
08:40:01.207 I: PV-Anlage 2/ch1/YieldTotal: 174.116 kWh
08:40:01.208 I: PV-Anlage 2/ch1/Irradiation: 0.405 %
08:40:01.208 I: PV-Anlage 2/ch1/MaxPower: 1.600 W
08:40:01.219 I: PV-Anlage 2/ch2/U_DC: 31.900 V
08:40:01.220 I: PV-Anlage 2/ch2/I_DC: 0.080 A
08:40:01.221 I: PV-Anlage 2/ch2/P_DC: 2.600 W
08:40:01.222 I: PV-Anlage 2/ch2/YieldTotal: 389.943 kWh
08:40:01.233 I: PV-Anlage 2/ch2/Irradiation: 0.658 %
08:40:01.233 I: PV-Anlage 2/ch2/MaxPower: 2.600 W
08:40:01.234 I: PV-Anlage 2/ch0/U_AC: 228.900 V
08:40:01.235 I: PV-Anlage 2/ch0/I_AC: 0.020 A
08:40:01.246 I: 08:40:02.427 I: (#1) Radio infos: -6 -6 4 -6 -6 | t: 96, s: 90, f: 0, n: 6 | p: 2
08:40:02.429 I: (#1) TX 27 CH40 | 15  *  *  *  * 85 08 79 52 80 0b 00 65 a0 ec d2 00 00 00 00 00 00 00 00 88 6f 0c 
08:40:03.934 I: (#1) RX 132ms | 23 CH75 | 95  *  *  *  *  *  *  *  * 83 00 00 00 02 03 ea 00 5c 00 01 10 2d 9d 
08:40:03.937 W: (#1) frame 1 missing: request retransmit (8 attempts left)
08:40:03.938 I: (#1) TX 11 CH40 | 15  *  *  *  * 85 08 79 52 81 1a 
08:40:02.082 I: (#1) RX  31ms | 27 CH75 | 95  *  *  *  *  *  *  *  * 01 00 01 01 34 00 05 00 11 01 3e 00 08 00 1a 00 02 9b 
08:40:03.091 W: (#1) frame 2 missing: request retransmit (7 attempts left)
08:40:03.092 I: (#1) TX 11 CH40 | 15  *  *  *  * 85 08 79 52 82 19 
08:40:03.225 I: (#1) RX  31ms | 27 CH75 | 95  *  *  *  *  *  *  *  * 02 a8 24 00 05 f3 37 00 00 00 00 08 ed 13 8a 00 29 8f 
08:40:03.229 I: (#1) Payload (42): 00 01 01 34 00 05 00 11 01 3e 00 08 00 1a 00 02 a8 24 00 05 f3 37 00 00 00 00 08 ed 13 8a 00 29 00 00 00 02 03 ea 00 5c 00 01 
08:40:03.245 -----
08:40:03.246 I: com loop duration: 1976ms
08:40:03.246 -----
webSerial, buffer overflow!
```
Hat der Buffer overflow noch etwas zu sagen???

You69Man, 2024-01-12:
> Ich glaube, mit der .47 ging es los. Habe nun die neue .52 drauf, und........... es läuft!!! Ihr seid meine Helden! Ich baue nun mal alles zusammen und lasse es bis heute Abend in Ruhe laufen, aber [...]

Super, wenn es jetzt läuft! @lumapu ist der Held!

Bei mir ging heute früh überhaupt nichts mehr, nur Alarme, keine Leistung am Display, und ich dachte natürlich gleich wieder an Ahoy Softwarefehler. Dabei war's ganz was anderes: In die Außensteckdose ist Feuchtigkeit geraten und der FI ist gefallen! Die Ahoy hat ganz korrekterweise Frequenz-Alarme angezeigt, und nach Entfeuchtung der Steckdose funktioniert auch wieder alles wie gewohnt :)

Gubi2023, 2024-01-12:
Kann eure Begeisterung leider nur bedingt teilen, bei mir startet sich mit 8.52 die DTU alle 10 Min. neu, somit Zugriff auf Daten nur sehr beschränkt möglich.

EinLaie, 2024-01-12:
Leider gehts mir wie Gubi2023. Die 8.46 läuft bei mir. Vielleicht stimmt auch was mit der Zeitsync nicht. Sobald ich mit NTP synce kommt "synced at: 1970-01-01, 01:00:11, difference: 1705067180ms" oder die DTU bootet neu.

mtet2, 2024-01-12:
Hier wurde vor einiger Zeit mal der Tip gegeben den NTP-Server local von z.B. der Fritzbox mit Angabe der IP zu nehmen.
Hatte bei mir eine Verbesserung gebracht. 
Die IP auf die eigenen Gegebenheiten anpassen. Im Bild ist die Fritzbox Standardeinstellung zu sehen.
[image]

soelli22, 2024-01-12:
Ich habe seit über 8 Stunden keinen Reboot mehr gehabt. Also für mich läuft es. Soll ich das Issue schließen und Gubi2023 öffnet ein neues oder lassen wir es hier laufen?

You69Man, 2024-01-12:
> Ich habe seit über 8 Stunden keinen Reboot mehr gehabt. Also für mich läuft es. Soll ich das Issue schließen und Gubi2023 öffnet ein neues oder lassen wir es hier laufen?

Ich glaub es gibt schon einen neuen, der ganz ähnlich klingt: #1345

knickohr, 2024-01-12:
Hat der Buffer overflow noch etwas zu sagen???

Nein, das sagt nur das die serielle Ausgabe wegen zu vieler Daten übergelaufen ist. Du ballerst ja auch alle Informationen raus 😉

lumapu, 2024-01-12:
hier bitte wie @Gubi2023  es schon richtig gemacht hat angeben, ob man einen ESP8266 oder ESP32 hat. Ich denke wir haben hier ein Speicherproblem beim ESP8266 - hatten wir schon lange nicht mehr