lumapu/ahoy #1346: Übersetzungsfehler in deutsch ?

Gubi2023, 2024-01-10:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.49

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Änderung in Settings-Inverter

### Error description

 in der deutschen Übersetzung wurde bei den Settings aus "Reset 'max' values at midnight" plötzlich "Maximalwerte bei Sonnenuntergang zurücksetzen". Was stimmt nun??

lumapu, 2024-01-10:
mitternachts ist richtig, Wird in der nächsten Version gefixt.

lille2000, 2024-01-10:
Nur ein Vorschlag: 
Vielleicht kann man die Abkürzung “Begr.”  mit “Limit” ersetzen
[image]
.

lumapu, 2024-01-10:
ok, es hieß halt früher "Power Limit" und heißt jetzt wie bei Hoymiles "Active Power Control" (APC). Aber dennoch finde ich "Limit" hier auch besser als das abgekürzte "Begr."

huaak, 2024-01-11:
..spricht etwas dagegen in der deutschen Version aus dem doch recht sperrigen 
"Zuletzt empfangene Daten wurden angefragt um: *Zeitstempel*" 
ein "Zuletzt empfangene Daten um: *Zeitstempel*" zu machen?!..

huaak, 2024-01-11:
..achja, in der lang.json ist in Line 94 & 104 ein kleiner Typofehler: "Verindung" statt "Verbindung"..

Gubi2023, 2024-01-11:
[image]
ich würde das Fragezeichen weglassen...

Gubi2023, 2024-01-12:
blaue Felder auch übersetzen oder so lassen?
[image]

zappoXX, 2024-01-14:
[image]

closed 2024-01-19 by Gubi2023