lumapu/ahoy #1350: 0.8.36: "Reset values and YieldDay at midnight" nur bei "Pause communication during night"?

fsck-block, 2024-01-11:
Das zurücksetzen der Werte um Mitternacht klappt bei mir nur wenn man die Kommunikation in der Nacht ausgeschaltet hat.

Wahrscheinlich in der Version 0.8.7 mit `* made night communication inverter depended` reingekommen.
Dort wurde die entsprechende Bedingung zugefügt.

```
@@ -358,6 +365,8 @@ void app:: zeroIvValues(bool checkAvail, bool skipYieldDay) {
            continue;  // skip to next inverter
        if (!iv->config->enabled)
            continue;  // skip to next inverter
+       if (iv->commEnabled)
+           continue;  // skip to next inverter

        if (checkAvail) {
            if (!iv->isAvailable())            
```

Allerdings ist mir nicht klar warum man die Werte nur zurücksetzen darf wenn keine Kommunikation mit dem Inverter geplant ist.

War das Absicht oder kann das weg?

lumapu, 2024-01-11:
wurde nicht bedacht, danke für den Hinweis

lumapu, 2024-01-11:
wobei, wenn ich es so recht überlege würde das ja stören wenn man eine Batterie hat. Dieser Wechselrichter sollte doch genau dann zählen, da ich ja über Nacht wissen will wie viel ich aus dem Akku entnommen habe. 
Hier einen weiteren Haken einzubauen wäre glaube ich zu viel des guten. Was ich mir vorstellen könnte, dass man den Reset per MqTT und API ermöglicht.

fsck-block, 2024-01-12:
Ich habe bei mir jetzt kein Problem, ich habe nun die Kommunikation über Nacht abgeschaltet. Man kommt nur vielleicht nicht sofort darauf dass es  mit `Pause communication during night` zusammenhängt wenn's nicht klappt.
Vielleicht kann man einen Hinweis bei der config seite geben.

> Hier einen weiteren Haken einzubauen wäre glaube ich zu viel des guten. 

Das glaube ich auch. Ist so oder so schon verwirrend wann was "resetted" wird.

Aber um vielleicht mehrere Fälle abdecken zu können könnte man aus dem Häckchen `Reset values and YieldDay at midnight`  ein frei wählbaren Zeitpunkt HH:MM:SS machen. Dann kann jeder sich den passenden Zeitpunkt wählen.
Der Batterie-Nutzer "resetted" am Abend, andere am Morgen und ich für meine Auswertung um Mitternacht.

fsck-block, 2024-02-17:
Hi @lumapu 
Mit der Änderung von #1429 ist auch dieser Issue erledigt und könnte geschlossen werden.

Die mit PR #1418 eingefügten Hinweise bzgl. Nachkommunikation müssten wieder raus. 
Willst du das direkt andern oder soll ich einen PR erstellen ?

lumapu, 2024-02-17:
gerne per PR 😃

closed 2024-05-30 by lumapu