lumapu/ahoy #1351: Incorrect power control value after update

haraldhotzbehofsits, 2024-01-12:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.52

### Github Hash

455d29a

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Debug Serial Log output

```shell
-
```

### Error description

Power has been at 100% before with 0.8.36, after update to 0.8.52 the live view shows 10%, manual change to 100% brings it back to 100%.

[image]

lumapu, 2024-01-12:
yes there has been a change recently. I introduced one decimal place for the power limit. Since then the limit has to be multiplied by `10` if it is set from MqTT. 

Example:
target limit: 87.9%
send limit: `879`

The documentation about that is missing, for that reason I added the "documentation" label to this issue

haraldhotzbehofsits, 2024-01-13:
I recogniced that today, that my power limiter is broken by that change, now I know where to fix it.

stefan123t, 2024-10-29:
@haraldhotzbehofsits also described or fixed in #1684 

@lumapu do we need to keep this open for the documentation ?
I have documented the [OpenDTU Swagger REST API UI](https://github.com/tbnobody/OpenDTU/issues/378#issuecomment-2217027795) (OpenAPI [`opendtu_openapi.yaml`](https://github.com/user-attachments/files/16228674/opendtu_openapi_20240714.txt)).