lumapu/ahoy #1353: Bitte neue Namensgebung erläutern für den Bereicht Actions/Developer

rmayergfx, 2024-01-12:
Nach der Version 0.8.45 wurde der Bereich umgestellt, das single File ahoydtu_dev.zip existiert nun nicht mehr, dafür sind nun dort im Bereich viele einzelne *.bin zum Download vorhanden.

Aus dem entpackten ahoydtu_dev.zip wurde von mir das File 240106_ahoy_0.8.45_1aa14eb_esp32.bin genutzt.
Nun sehe ich im Bereich viele einzelne Files mit dev-esp32-*
Finde das ziemlich verwirrend, gerade für Neueinsteiger. 
Bin mir auch hier gerade selbst nicht sicher, welches neue File dem alten 240106_ahoy_0.8.45_1aa14eb_esp32.bin entsprechen würde.
dev-esp32-wroom32 oder dev-esp32-wroom32-de ?

BastyESP, 2024-01-12:
Der alten wohl das ohne DE, wenn du es übersetzt haben willst in Deutsch, dann mit DE...

Fand es anfangs auch komisch, hab's dann einfach Versucht und hat so funktioniert.

rmayergfx, 2024-01-12:
???

lumapu, 2024-01-12:
ich habe schon Ideen zur Verbesserung, warte mal noch paar Tage. Ansonsten stimmt das natürlich was @BastyESP schreibt.
| Firmware | Sprache |
|---|---|
| dev-esp32-wroom32 | Englisch |
| dev-esp32-wroom32-de | Deutsch |

rmayergfx, 2024-01-12:
Für mich und viele Neueinsteiger verwirrt schon die Bezeichnung wroom32, die war vorher so nicht im Filenamen.

lumapu, 2024-01-12:
das hat sich im Prinzip nicht geändert, wenn du die Firmware von hier lädst:
https://fw.ahoydtu.de/fw/dev/ahoy_v0.8.53/

rmayergfx, 2024-01-13:
Doch hat es sich, denn die Namen wurden geändert.... Schau dir es noch mal genau an.
Das führt zu unnötigen Verwirrungen. Warum behält man die alten Daten nicht einfach bei oder packt eine Legende auf der Seite mit dazu?

MetaChuh, 2024-01-13:
> Warum behält man die alten Daten nicht einfach bei oder packt eine Legende auf der Seite mit dazu?

@rmayergfx  weil alles neu macht der mai 😉

word play pun: wünschte der mai wär jetzt schon da ... solar ist steigend aber immer um eine gewisse nuance gerade zu wenig 🙈

scherz bei seite: mach's evtl. ähnlich wie ich: produktiv dtus nicht verändern, auch wenn's dann wie bei mir "retro" ahoy versionen sind,
und maximal in deinen dev dtus, die dir egal sind, neue versionen testen.
dann hat keiner zusatzstress.

(disclaimer: natürlich wäre auch ich happy über eine neue rock solid version, aber erstens funzen unsere aller alten setups, und zweitens würde es dev mäßig viel langsamer voran, wenn man bei jeder code zeile dran denken muss, was es breaken könnte, statt mal einfach was zu probieren ... trial and error rulez 💪😂)

rmayergfx, 2024-01-13:
Das eine hat doch mit dem anderen nichts zu tun. Aber man muß nicht noch unnötig die Namenskonventionen ändern. Egal welcher Monat gerade auf dem Kalender steht.

lumapu, 2024-01-15:
@rmayergfx jetzt siehst du was du zuletzt installiert hast und kann in Zukunft genau das richtige Image heraussuchen. Das wird sich aber noch weiter verbessern, das erfordert aber noch etwas Geduld
[image]

rmayergfx, 2024-01-15:
Ab welcher Release ist diese Ansicht aktiv? Ich sehe es mit der 0.8.45 eben nicht und daher ist es etwas tricky.
Sag Bescheid wenn es eine neue stable release gibt, bei der es eingebaut ist.

lumapu, 2024-02-17:
jetzt ist es eingebaut @rmayergfx

closed 2024-02-17 by lumapu