lumapu/ahoy #1355: Fusionsboard ab Version 0.8.52 mit CMT Bootschleife

teletapi, 2024-01-13:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.50

### Github Hash

e4cb948

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

ohne

### Debug Serial Log output

```shell
kein Log
```

### Error description

Sobald beim Fusionsboard die 0.8.52 oder 0.8.53 (bei der 51 bin ich nicht sicher hab ich nicht probiert) geflasht wird egal ob per Ota oder VSC Flash und das CMT ist aktiviert in den System Configs beginmt eine dauer Bootschleife. Man kann mit etwas Glück in den System Configs das CMT Häckchen entfernen und speichern danach ist die Bootschleife weg.
Von der 0.8.52 konnte ich auch nicht downgraden bei der 0.8.53 konnte ich wieder downgraden auf die 0.8.50 die auch wieder mit CMT funktioniert.
Display 1309 2,42"

teletapi, 2024-01-13:
Gleiches gilt im übrigen auch für den ESP32

lumapu, 2024-01-13:
du denkst ich kann es bei mir reproduzieren?
Werde es gleich testen.

teletapi, 2024-01-13:
https://discord.com/channels/984173303147155506/1144045244166443159

Wir sind hier schon seid einer ganzen Weile am testen und kommen nicht so richtig weiter

You69Man, 2024-01-14:
See also #1354

knickohr, 2024-01-14:
Bitte mal dazu schreiben, ob ein Display dran hängt, und welches.

Dirk-Ahoy, 2024-01-14:
Fehler ist jetzt gefunden - sollte bald gefixt werden.   Bis dahin Display abstellen oder den Graphen auf >0 % stellen - dazu muss es aber Nacht sein oder Antenne abmachen das kein Empfang vom Wechselrichter ist.

You69Man, 2024-01-14:
Siehe #1360

closed 2024-01-15 by teletapi