lumapu/ahoy #1358: Angebot: Info über compilierte Module in version string

fsck-block, 2024-01-14:
Da nun auch MQTT wie andere Module auch an/abschaltbar ist nimmt damit die Anzahl der möglichen Varianten pro SW-version weiter zu. Es ist vielleicht in Zukunft bei Fehlermeldung hilfreich zu wissen, welche Module enthalten sind.

Ich habe dafür eine kleine Änderung implementiert bei welcher eine Kennung zur Versionkennung je nach compiliertem Modul zugefügt wird. Z.b. für die 0.8.54 mit Prometheus, Display und History: -> "0.8.54-PDH"
Mögliche Werte:
P - Prometheus
M - MQTT
D - Display
H - History
A - AP Only

Steht als PR  #1357  zur Verfügung

knickohr, 2024-01-14:
Noch schöner wäre es man könnte sich diese Module im Git auswählen und es baut für Dich dann diese Version.

fsck-block, 2024-01-14:
Stimmt, dann brauchst du um so mehr die Info was rein gebaut wurde.
Allerdings übersteigt das aktuell meine Fähigkeiten.

knickohr, 2024-01-14:
Kannst Du die Sprache auch noch mit kenntlich machen ?  Oder ist das eine andere Ecke. Mir fiel bei der 54 auf das sie offenbar beides kann. Mal ist sie DE, mal EN 😵

fsck-block, 2024-01-14:
Sprache geht über `#define LANG_DE`.
Kann ich mal rein machen.

fsck-block, 2024-01-14:
Das würde dann so aussehen:
[image]

lumapu, 2024-01-14:
wird in `0.8.55` gemerged =), siehe auch Kommentar dort