lumapu/ahoy #1359: Neuere Versionen (0.8.52 und 0.8.54) nicht updatebar

kiu77, 2024-01-14:
## Hardware
ESP32

### nRF24L01+ Module
nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode

### Antenna:
external antenna (SMA)

### Power Stabilization:
100uF Electrolytic Capacitor connected between +3.3V and GND (Pin 1 & 2) of the NRF Module

### Build & Flash Method:
AhoyDTU Webinstaller
OTA

Bis zu den development versions 0.8.54 klappte das Update mit der eingebauten Firmware Funktion wunderbar.
Beim Versuch auf eine der Versionen 0.8.52 oder .54 upzudaten kommt die Fehlermeldung, dass das Update schief ging
und die alte Firmware bleibt drauf. 
Ich habe es mit der englischen und der deutschen Version probiert:
240114_ahoy_0.8.54_60111d0_esp32-wroom32.bin
und
240114_ahoy_0.8.54_60111d0_esp32-wroom32-de.bin
aber kein Erfolg.
Dann habe ich das ESP32-Modul per Online-Updater neu geflachst mit Löschen des EEPROMs.
Auch von der dann installierten Version 0.8.36 aus klappte das FW-Update nicht.
Was machen ich da falsch?

knickohr, 2024-01-14:
Sollte ein success noch kommen, dann Power off, kurz warten und wieder Power on, also Stecker ziehen.

Wenn das auch nicht funktioniert, bleibt der Weg über den AP nich. Im AP booten, WLAN also kurz ausschalten, und dort das Update machen.

kiu77, 2024-01-14:
Nichts funktioniert.
Via AP-Mode funktioniert das Update zwar, aber dann bekommt die neue Firmware nicht einmal mehr den AP-Modus hin.
Ich kann nicht mehr in das WLAN AHOY-DTU einloggen. Er fragt sogar neu nach dem PW, aber nach "esp_8266" rödelt es eine Weile rum, um dann wegen Zeitüberschreitung abzubrechen, neuer Versuch einzuloggen, neu PW eingeben => Endlosschleife.
Ich gebe es auf. Es funktioniert einfach nicht, egal was ich mache. Hab jetzt bestimmt über 10 x neu geflasht mit EEPROM-Reset.

knickohr, 2024-01-14:
Dann warte auf die 55

Joh49, 2024-01-15:
Mit 0.8.56 ist wohl ein großer Wurf gelungen. Mein ESP32 ließ sich ohne Zicken updaten, die Gesamtleistung ist erstmals auch auf dem PC ausgegeben, kein Reboot, schnelle Synchronisation. Alles von .49 bis .54 wurde verweigert und nur .48 war nutzbar. Anerkennung und Danke.

kiu77, 2024-01-15:
@knickohr 
Habe gewartet...

@Joh49 
Danke für den Hinweis. Bei mir geht es auch damit ;-)

kiu77, 2024-01-15:
Alles wieder gut mit 0.8.56

closed 2024-01-15 by kiu77