lumapu/ahoy #1361: [Bug] History Charts does not contain the whole day

lille2000, 2024-01-14:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.53

### Github Hash

Unbekannt

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

—

### Error description

There is no history data on display and Webgui at the end of the day.
Some versions ago I had a graph of the whole day ( between sunrise and sunset)
Now it‘s different: It seems that that the displaced timespan is only the last some hours ( not taking into account that communication is paused during night)

[image]
[image]
[image]

lille2000, 2024-01-14:
[image]

lille2000, 2024-01-14:
[image]
[image]

knickohr, 2024-01-14:
Check the uptime of the DTU

lille2000, 2024-01-14:
Uptime: 1d10h

You69Man, 2024-01-14:
fix for display graph is in work

You69Man, 2024-01-15:
PR: #1364

lille2000, 2024-01-16:
Das externe Display wurde bisher nicht mehr zu früh zurückgesetzt. ✅👍

Die Webgui  zeigt maximal die letzten 200 Messwerte ( alle 15s Einsen ) an und stoppt auch zum Inverterstop.  200 Messwerte sind nicht viel, aber ich habe in einem anderen Issue  gesehen, dass amm der Form schon gearbeitet wird.

lille2000, 2024-01-17:
Hier #1336 steht, dass die letzten 256  Messpunkte in der Webgui angezeigt werden.  Es ist also kein Bug sondern Feature.

Ich beobachte die Displayanzeige noch einige ZZage und schließe danach das Ticket.

BTW:
256 Werte, alle 15s gemessen = 64 Minuten History.  Das ist nicht viel bzw. mir bringt diese Webgui-Anzeige in dieser Form leider keinen Mehrwert (da ich am Nachmittag eh nicht mehr viel Licht habe) 
Es ist mir aber klar, dass  die kleinen Mikroprozessoren nicht viel RAM haben…

knickohr, 2024-01-17:
Dann müßt ihr halt das Inverter Intervall höher setzen 😉

dtuuser, 2024-01-17:
Beim 5 Minuten Intervall würde man 16 Stunden darstellen können

closed 2024-02-17 by lumapu