lumapu/ahoy #1368: [Bug] 0.8.57 MQTT yield day ab dem zweiten Tag = 0

MiniOh, 2024-01-17:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.57

### Github Hash

-

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

-

### Error description

Hallo, 

in der Version 0.8.57 wird yield day über MQTT als 0 ausgegeben "ab dem zweiten Tag".
Nach einem Reboot wird yield day korrekt ausgegeben, nach einer Nacht (ohne Neustart der DTU) wird inverter/total/YieldDay
als 0 zurück gegeben. In der GUI stimmen die Werte.
Wenn man manuell "an Tag 2" einen Rebbot ausführt, wird yield day wieder richtig ausgegeben.

Werte und Gesamtertrag um Mitternacht zurücksetzen - JA
Werte bei Sonnenuntergang zurücksetzen - NEIN
Werte zurücksetzen, sobald der Wechselrichter nicht erreichbar ist - NEIN
Maximalwerte mitternachts zurücksetzen - NEIN
Kommunikation starten ohne gültige Zeit (sinnvoll im AP Modus) -NEIN
Grid-Profil auslesen - JA

Beste Grüße

tester277, 2024-01-17:
Kann ich für identische Hardware bestätigen. Nach Reboot am 2 ten Tag wird YieldDay wieder angezeigt und auch weiter per mqtt entsprechend übermittelt/aktualisiert.
Wobei folgende Settings bei mir auf nein stehen:
Reset values and YieldDay at midnight
Reset values at sunset
Reset values when inverter status is 'not available'
Reset 'max' values at midnight
Start without time sync (useful in AP-Only-Mode)

Danke und Grüße

lumapu, 2024-01-17:
ok, habt ihr `disable night communication` in den Invertersettings an?

tester277, 2024-01-17:
Bingo - Pause communication during night ist aktiv und Reboot Ahoy at midnight ist (logischerweise) nicht aktiv.

MiniOh, 2024-01-17:
> ok, habt ihr `disable night communication` in den Invertersettings an?

Bei mir ist die Option aus.

Kein Neustart nachts.
Keine Disable Night Communication

MiniOh, 2024-01-17:
[image]

lumapu, 2024-01-17:
verstanden, das kann die aktuelle Logik tatsächlich gerade nicht, bruache irgendein event um eine Variable, die den Wert zurücksetzt - bin am überlegen.

MiniOh, 2024-01-17:
Ok, also ein nächtlicher Neustart würde das "Problem" dann erst mal
umgehen?

Blöde Frage. Seit wann besteht das Verhalten und warum war es vorher nicht da?
Ich hatte die Optionen Reboot und Pause Communication nicht verändert.

lumapu, 2024-01-17:
ja würde es. Wir haben genau an dieser Stelle geschraubt, da es immer wieder zu viele MqTT Übertragungen gab in der Vergangenheit. Ich werde das hoffentlich die Tage fixen können, dann kannst du die Option wieder wie zuvor stellen.

closed 2024-02-17 by lumapu