lumapu/ahoy #1370: [Bug] Warnmeldung bei Firmwareupdate

Gubi2023, 2024-01-18:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.58

### Github Hash

????

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Firmwareupdate

### Error description

wenn ich beim Update die Sprache nur von englisch auf deutsch ändere (gleiche Firmware 8.58), erhalte ich folgende Warnmeldung 
[image]
in Gegenrichtung (deutsch auf englisch) kommt  keine Warnung

zappoXX, 2024-01-18:
Bei mir auch...

[image]

knickohr, 2024-01-18:
Kommt nur einmal. Wenn ihr aber eine ältere Firmware aufspielt als die 58 dann kann es wieder auftreten. Ist kein Bug, wurde mit der 57 eingeführt, sollte ab jetzt bei neueren Versionen nicht mehr auftreten.

lumapu, 2024-01-18:
ist leider nicht ganz trivial aus den beiden Strings das environment korrekt zu extrahieren. Ich bemühe mich hier in Zukunft noch nachzubessern

Gubi2023, 2024-01-18:
> Kommt nur einmal. Wenn ihr aber eine ältere Firmware aufspielt als die 58 dann kann es wieder auftreten. Ist kein Bug, wurde mit der 57 eingeführt, sollte ab jetzt bei neueren Versionen nicht mehr auftreten.
8.59:  Warnmeldung kommt immer noch, jetzt in beiden Richtungen. englischer Text unverändert, beim deutschen ist der Punkt jetzt vorhanden.

MetaChuh, 2024-01-18:
zum thema multilingual localization:
multilingual localization is more complicated than one thinks at first glance.

bearbeitung seit einführung kostet, auch bei diesem projekt, den devs bis jetzt mehr zeit, als genau wie vorher zu antworten, warum es ahoy dtu nur auf englisch gibt/gab.

siehe: andere projekte, open source, sowie major companies, die jahre lang mit den problemen zu kämpfen hatten und haben, dass zb. durch string lengths, plötzlich dieser oder jener button/div/zeile entweder nicht mehr lesbar ist, bzw. durch overflow zeilenumbrüche die optik nicht mehr ist wie vorher, ganz zu schweigen von online übersetzten texten, wenn grad' kein native speaker da war (letzteres nicht hier, jedoch in fast jeder von alexibexi vorgelesenen anleitung)

lumapu, 2024-01-19:
@Gubi2023 danke für deine Hinweise, der deutsche Text in der englischen Version kommt aus deinem Browser, Stelle den auch auf english dann passt's 😉
Ich glaube langsam haben wir das multi language in Griff, ist echt unglaublich wie viele eine Deutsche Version wollen

closed 2024-02-17 by lumapu