lumapu/ahoy #1372: [Bug]: Am Handy  Settings-Inverter-Edit/Delete nicht sichtbar

Gubi2023, 2024-01-19:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.58

### Github Hash

0.8.59

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Setup Inverter

### Error description

öfters mal mas neues: Am Handy (Iphone und Android)  kann seit V8.58 das Fenster Settings-Inverter nicht mehr verschoben werden, somit sind die Buttons -Edit/Delete nicht erreichbar.
[image]
Ausserdem sind in 8.59 die Texte dazu auf deutsch, obwohl englische Firmware geladen

Gubi2023, 2024-01-22:
V8.63:  Verhalten noch unverändert; über Handy keine Settings-Änderungen möglich, da die Buttons im Inverter-Feld nicht mehr sichtbar sind!

MetaChuh, 2024-01-23:
dirty workaround für jetzt: handy im querformat halten.

ps: erinnert mich grad an apple's erstes kommentar zum iphone 4 antenna problem:
it's not broken, people are just holding it wrong 😂

Gubi2023, 2024-08-16:
jetzt ok. Danke 👍

closed 2024-08-16 by Gubi2023