lumapu/ahoy #1375: [Bug] mqtt daten  "Yield Day" werden nicht mehr gesendet

Oxbow6, 2024-01-20:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.36

### Github Hash

x

### Build & Flash Method

ESP Tools (flash)

### Setup

MqTT Brocker mit 60 sec. intervall

### Debug Serial Log output

```shell
total
YieldTotal = 904.753
P_AC = 574.1
P_DC = 604
```

### Error description

Seit dem update auf die Version 0.8.36 habe ich keine total Yield Day Daten mehr auf meinem OpenhaB. 
Habe mir mal mit einem MQTT Explorer das angeschaut und die Ahoy sendet keine total yield Day daten mehr über MQTT.
Vor dem Update hat das aber noch funktioniert.

Gubi2023, 2024-01-20:
wird in "history" auch nicht angezeigt. 
[image]
gleiche Ursache??

lumapu, 2024-01-20:
@Gubi2023 nein, unten steht ja alle 86400s (also jeden Tag ein Strich).

Die Releaseversion sollte das noch ohne weiteres können. Kannst du bitte die aktuelle Dev testen?
https://fw.ahoydtu.de/fw/dev/ahoy_v0.8.60/

Gubi2023, 2024-01-20:
ok, hab ich fehlinterpretiert.
deshalb nochmals meine Frage: wäre es möglich, beim total yield per day "updated" in Std/Min. anzugeben? (86400sec ist etwas sperrig...)
https://github.com/lumapu/ahoy/pull/1336#issuecomment-1890446205

[image]

Oxbow6, 2024-01-21:
Hallo 
Auch mit der v08.60 Version habe ich das gleiche Problem, es werden keine Yield Day über MQTT gesendet. 
Wenn ich aber den Discovery Config (homeassistant) sende funktioniert die Yield Day Anzeige für einen Tag, danach habe ich weider das gleiche Problem.

lumapu, 2024-01-21:
@knickohr ist unser MqTT Experte, er hätte sich doch schon längst gemeldet, wenn das nicht mehr funktioniert - wobei er momentan jeden Tag eine neue Version testet.

knickohr, 2024-01-21:
Das wäre mir sofort aufgefallen 😉

Wenn der Topic inverter/total/YieldDay oder inverter(total/YieldTotal gemeint ist, kann ich keine Probleme feststellen. Bei den einzelnen Invertern habe ich nicht nachgeschaut und nutze die auch nicht.

Oxbow6, 2024-01-21:
Hallo 
es geht um den  inverter/total/YieldDay, der nach 24 Stunden dann plötzlich nicht mehr läuft. Auch wird dieser nicht auf Null gesetzt sondern hat immer noch den Wert vom letzten Tag. Habe die Daten extra über einen MQTT Explorer kontrolliert.

lumapu, 2024-01-21:
also das gleiche wie in #1368 ?

Oxbow6, 2024-01-21:
So in der Art, das Problem sollte ja gelöst worden sein ab der Version 0.8.57, denn ich benutze die Version 0.8.60
Bei mit steht im MQTT Explorer am 2 Tag einfach der Wert vom Vortag drin und dieser verändert sich auch nicht. 
Erst nach einem Discovery Config (homeassistant) wird der Wert dann korrekt angezeigt.

total
YieldTotal = 926.717
YieldDay = 10565.9
P_AC = 4.7
P_DC = 13.9

Oxbow6, 2024-01-21:
Nein
über Nacht schalte ich die Kommunikation der Wechselrichter aus.

lumapu, 2024-01-22:
ich kann das nicht nachvollziehen, auch heute kamen wieder die Daten korrekt rein. Anbei meine Einstellungen

[image]

Oxbow6, 2024-01-22:
Das sind aktuell meine Einstellungen:

[image]

Oxbow6, 2024-01-24:
Hallo 
Ich habe nun aktuell die Version 0.8.62 am laufen und mit dieser habe ich das Problem nicht mehr, Obwohl laut Log hier nichts am MQTT geändert worden ist. Ich werde das weiter beobachten

tester277, 2024-01-24:
Bei mir tritt der Fehler seit gestern  auf (Version 0.8.64). Was hat sich geändert? Habe einen Inverter außer Betrieb (gestern tagsüber) genommen. Bis zum Ende gestern sind Daten via mqtt in meiner Datenbank. Seit heute nich mehr. Da hilft auf kein  restart.

tester277, 2024-01-24:
Entschuldigt für das hin und her. Habe es geprüft. Die total Werte fehlen seit gestern. Es sind nur die Werte des letzen (verbliebenen) Inverters vorhanden. So gesehen ist das eher works as designed. Wozu doppelt identische Werte für einen Inverter erfassen.
[image]

lumapu, 2024-01-24:
kann nicht folgen. Du meinst warum werden die Totalwerte bei nur einem Inverter gesendet?

lumapu, 2024-01-27:
das ist absichtlich so. Der Zustand war garnicht so einfach herzustellen - @knickohr weiß ein Lied davon zu singen.
Hätten wir es nicht so implementiert, würde der Yield sehr stark hin und herspringen, wenn einzelne Inverter mal nicht erereichbar sind ...

knickohr, 2024-01-27:
Bei „nicht erreichbar“ ist nicht nur deaktiviert oder ausgeschaltet gemeint, es reicht schon wenn mal zwischendrin keine Produktionsdaten kommen. Das Ganze zappelt dann wie ein Kuhschwanz hin und her.

tester277, 2024-01-27:
Bei mir war es nur ein Übergangszustand. Ich sehe da keinen Fehler der unbedingt korrigiert werden müsste.

Oxbow6, 2024-02-04:
Hallo
Ich hätte noch eine andere Frage, besteht die Möglichkeit über MQTT auch die Maximale Leistung aller Wechselrichter abrufen?  im Ordner Total habe ich aktuell nur diese 4 Werte.
Gruss und Danke

total
YieldTotal = 1092.71
YieldDay = 1532.35
P_AC = 2297.7
P_DC = 2418.2

knickohr, 2024-02-04:
Ist doch drin, für jeden Inverter und für jeden seiner Channels 😉

[image]

Nur der gesamte Totalwert nicht.

Oxbow6, 2024-02-04:
Hallo Knickohr
Ich würde gerne nur den Totalbetrag abgefragt haben, denn sonst müsste ich alle 32 Channels einzeln abfragen und das geht dann wieder auf die Performance meiners Openhabs.

knickohr, 2024-02-04:
OK, der Issue ist noch auf, aber wenn Du möchtest, kannst Dubdas auch in die Wunschliste rein schreiben :  #1199

Oxbow6, 2024-02-07:
Hallo Knickohr
Ich verwende nun die Version 0.8.75 mit dem Mqtt MaxPower und das funktioniert super. Vielen Dank

[image]

Oxbow6, 2024-02-12:
Hallo Knickohr
Ich habe noch ein kleines Problem mit dem MQTT MaxPower, plötzlich gegen den Abend stimme die Werte nicht mehr überein.
Vermute mal dass dies immer noch Sonnenuntergang passiert
Hast Du eine Vermutung ?
zb : Ahoy 2338.9W
MQTT 1450 W

knickohr, 2024-02-12:
Ein oder mehr Inverter ?
Wenn mehr, dann werden die fehlenden nicht mehr addiert.

Oxbow6, 2024-02-13:
Hallo
Es sind mehere Inverter ( 7 Stück )

knickohr, 2024-02-13:
Ja, dann werden nur die momentan aktiven aufsummiert.

Oxbow6, 2024-02-15:
Habe nun zum testen die Version 0.8.81 aufgespielt. 
Mit der bekomme ich keine Live AC und DC Daten über MQTT.

lumapu, 2024-02-17:
das wurde mit der 0.8.82 gefixt

closed 2024-02-17 by lumapu

emmrichd, 2025-03-16:
Ich habe das Problem, mit missing Total day, in der 0.8.155 - 5feb293 immer noch. ESP32

emmrichd, 2025-03-16:
Könnte daran liegen, dass einer meiner Wechselrichter nicht aktiv ist. Die Logik, warum dann kein Total kommen sollte erschließt sich mir aber nicht.