lumapu/ahoy #1377: [Bug] 0.8.59-minimal : Zugriff History führt zum OOM-Crash

fsck-block, 2024-01-20:
Bei der 0.8.59 minimal version (also ohne history, mqtt, etc) ist der History-Butten weiterhin verfügbar.
Das an sich wäre nicht so schlimm, doch wenn man die History aufruft, führt dies zu einem OOM-Crash.

Soweit ich das sehen kann, werden die Seiten `/api/powerHistory` und `/api/YieldDayHistory` vor Browser ohne Pause abgerufen. Dies führt dann zu einem OOM-Crash.

Möglicher fix: 
Das JSON object 'refresh' bei powerHistory und YieldDayHistory auf einen Tag setzen und auch in einer Minimal Version liefern.

knickohr, 2024-01-20:
Die minimal-Versionen sind nur zum Test.

fsck-block, 2024-01-21:
> Die minimal-Versionen sind nur zum Test.

Habe getestet, ist ein Fehlerchen drin :laughing:

Spass beiseite: 
Ich finde auch diese Art der Modularisierung und damit in Konsequenz die "minimal-Version" sehr schick.
Was nicht drin ist, kann keine Seiteneffekte produzieren. 
Verstehe auch, dass es schwieriger ist in einen fertigen Code so eine Modulariserung einzubauen als von Anfang an.

knickohr, 2024-01-21:
Ja, aber auf hast es bereits korrigiert 😉

Wollte damit nur sagen das diese Version eigentlich nie den Weg in die Öffentlichkeit finden sollte.

closed 2024-02-17 by lumapu