lumapu/ahoy #1379: Wo begrenz man die Leistung?

firedp, 2024-01-20:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [x ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ x] external antenna (SMA)

### Power Stabilization:
* [x ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.60
Github Hash: _______

### Build & Flash Method:
* [x ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Kurz gefragt. Komme von 0.6.12 und sehr lange das Thema nicht mehr verfolgt.
Jetzt im Dunkeln auf 0.8.60 gegangen und finde überhaupt nicht mehr die Möglichkeit die Leistung zu begrenzen.
Wer kann helfen?

mtet2, 2024-01-20:
Unter Daten auf Leistungsbegrenzung klicken und dann erscheit das Menü.
Bilder sagen mehr als Worte:

[image]

firedp, 2024-01-20:
Vielen Dank, auf die Idee dort zu gucken und klicken wäre ich tatsächlich ohne Hilfe nicht gekommen.

lumapu, 2024-01-21:
unter https://ahoydtu.de/screenshots wird es auch beschrieben