lumapu/ahoy #1381: [Bug] 0.8.60 lässt sich nicht von 0.8.59 updates und verbindet sich nicht via WLAN

kiu77, 2024-01-21:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.60

### Github Hash

abc1234

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

nicht besonderes

### Debug Serial Log output

```shell
kein log
```

### Error description

1. 0.8.59 läuft gut und responsiv.
2. Ein Update OTA via eingebaute Update-Funktion auf 0.8.60 klappt wiederholt nicht (Fehlermeldung und bleibt auf 0.8.59)
3. Nachdem Aufspielen von 0.8.36 samt Flash Reset via Webinstaller klappt zwar ein Update OTA auf 0.8.60, aber dann keine Verbindung via WLAN mehr.

lumapu, 2024-01-21:
Punkt 3 ist normal
Kann es sein, dass auch ein Soft-Reboot bei der 0.8.59 nicht funktioniert hat? Manchmal ist das auch bei mir der Fall, daher kein Update ohne Power-Cycle möglich. Die Ursache habe ich noch nicht gefunden.

kiu77, 2024-01-21:
Den USB-Stecker hatte ich dann auch immer gezogen.
Neu: Der Weg von 0.8.36 Webinstaller mit Reset -> 0.8.59 OTA hängt auch beim WLAN, AP-Funktion tut nicht.
Aber dann: 0.8.36 Webinstaller mit Reset -> 0.8.58 OT -> 0.8.59 OT -> 0.8.60 OT geht. Erstaunlich. Sogar ohne Steckerziehen.

rmayergfx, 2024-01-21:
Hier keinerlei Probleme mit Update 0.8.59 OT -> 0.8.60.

kiu77, 2024-01-21:
Interessant. Jetzt habe ich keine Ahnung, wo es hängen könnte.

Gubi2023, 2024-01-21:
bei mir war update problemlos. Leider mit 8.60 relativ viele Neustarts der DTU, die 8.59 lief 20 Std. problemlos durch....

lumapu, 2024-01-21:
bitte immer den ESP Typ dazuschreiben, sonst kann mans sehr schwer einordnen. (@Gubi2023 )

knickohr, 2024-01-21:
Workaround :  Mal in den AP booten und dann den Update machen. Sollte success kommen und sich trotzdem die alte Version wieder melden, dann denn Update nochmal machen und auf success warten. Danach die DTU komplett ausschalten und neu starten (Power Cycle, nicht nur Reboot).

Gubi2023, 2024-01-21:
@lumapu: immer noch ESP8266  ;-)   https://github.com/lumapu/ahoy/issues/1345#issuecomment-1889218537

huaak, 2024-01-21:
..also bei mir Update problemlos und seit 16h kein reboot..
[image]

 Wenn ich bisher Probleme beim Updaten hatte (selten) ging es zumeist mit einem anderen Laptop bzw. nach Check der Stromversorgung vom ESP (incl. reboot).
 
 ..sehe gerade der Footer könnte auch mal auf 2024 aktualisiert werden..  ;-)

lille2000, 2024-01-21:
Irgendwo habe ich in einem Ticket hier gelesen, dass die ESP8266 Probleme mit den WLAN  Kanälen 12-14 hätten. 
Ich hatte die selben massive Probleme wie @kiu77 mit etlichen Vorgängerversionen.
Seit dem ich im Router diese Kanäle ausgeschlossen hatte, hatte ich nicht einen einzigen Aussetzer beim Update. Ich weiß, es klingt dämlich und nach Voo-Doo…

PS: Mit 0.8.60 ist alles stabil (esp8266 + 2x HM + Display)

lumapu, 2024-01-21:
> @lumapu: immer noch ESP8266  ;-)   https://github.com/lumapu/ahoy/issues/1345#issuecomment-1889218537

danke, ich habe überall mit verschiedenen Aufbauten zu tun, verzeihe bitte, dass ich das nicht in Kopf habe 😉

Jumperm, 2024-01-25:
> Irgendwo habe ich in einem Ticket hier gelesen, dass die ESP8266 Probleme mit den WLAN Kanälen 12-14 hätten. Ich hatte die selben massive Probleme wie @kiu77 mit etlichen Vorgängerversionen. Seit [...]

Ich war auch in dem Ticket involviert und betreibe eine kleine DTU mit einer 8266 und einem WR. Besteht das Problem des Kanal 13+ nur für den 8266 und haben die ESP32 keine Probleme damit?
Um zukünftig für Klarheit zu sorgen sollte das eventuell auf der Webseite erwähnt werden. Ich würde sogar behaupten, dass es für den 8266 genau so wichtig ist wie der 100uF 😉

lumapu, 2024-01-29:
ich habe jetzt eine Warnung auf der `index` Seite eingebaut, wenn der ESP seit dem Start mal in Kanal 12 - 14 war (nur ESP8266). Ich hoffe das sensibilisiert ein wenig.

closed 2024-05-30 by lumapu