lumapu/ahoy #1382: [Bug] Fehler in radio statistics?

Gubi2023, 2024-01-21:
### Platform

ESP8266

### Assembly

the DTU was already assembled

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.6.1

### Github Hash

???

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Radio-Statistic

### Error description

hat sich in den Radio Statistics wieder ein Fehler eingeschlichen? die Summe der Rx ergeben keine 100% mehr

Gubi2023, 2024-01-21:
[image]

lumapu, 2024-01-21:
kam durch eine Änderung rein und wurde jetzt gefixt.

closed 2024-01-24 by Gubi2023