lumapu/ahoy #1383: Hallo zusammen, wie kann ich Tageswerte, sowie Max Werte täglich zurücksetzen, ohne dass die Gesamtwerte ebenfalls zurückgesetzt werden? Bzw., kann man die Totalwerte überhaupt zurück setzen?

Brummi70, 2024-01-22:
## Hardware
  * [ x] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

knickohr, 2024-01-22:
Vielleicht findest Du hier Deine passenden Einstellungen ?

[image]

P.S. die Betreffzeile ist kein Freitextfeld 😉

Brummi70, 2024-01-22:
sorry, das mit der Betreffzeile hatte ich überlesen.....
löscht die Einstellung "Reset values and YieldDay at midnight nicht die Total- Werte....ich habe mich nicht getraut diese Einstellung zu testen!?

lille2000, 2024-01-22:
Die „Totalwerte“ kommen aus den Invertern und werden daraus ausgelesen. Die werden mit keiner Einstellung gelöscht. 
Die Einstellungen wie im Bild oben sollten für dich passen.

MetaChuh, 2024-01-22:
@Brummi70 
soferne du keinen batterie inverter hast, der über die mitternachtsgrenze an ist, passt's wie von knickohr beschrieben.
falls du einen inverter über die mitternachtsgrenze hinaus verwendest, musst du den um 23:59 softwaremäßig rebooten, sonst holt sich die dtu die zurückgesetzten tageswerte seit einschalten wieder aus dem inverter, die wie von lille2000 beschrieben ebenfalls aus dem inverter kommen.

sollte zweiteres der fall sein, gib bescheid, was du für raspis, synos, linux, windows, mac geräte nachts rennen hast, um dafür tipps zugeben, wie man einen cron oder geplanten task dort einrichtet.
(die dtu selbst hat die funktionalität "restart inverter at midnight" derzeit nicht)

Brummi70, 2024-01-22:
super, vielen Dank....diese Infos habe ich benötigt....ihr seid schon extrem fit hier.....Danke an alle ;-)

GrisuXX, 2024-01-23:
Moin,

an dieser Stelle habe ich jetzt das Problem mit dem Reset vom Tagesertrag.
Nach der Empfehlung von Knickohr schalte ich den Inverter nicht mehr über einen Schütz sondern per API ein und aus. 
Funktioniert soweit gut aber der Tagesertrag wird nicht zurückgesetzt.
Auch bei einem Software Neustart vom Inverter bleibt der Wert in der DTU erhalten.
Erst bei einem anschließenden Neustart der DTU wird der Tagesertrag zurückgesetzt.

OK, ich habe das jetzt Tagsüber ausprobiert und müsste wie beschrieben den Inverter um 23:59 neu starten.

Der fängt aber anschließend gleich wieder mit der Produktion an.
Wenn er vor dem Reset aus war soll er anschießend auch aus sein.
Ansonsten greift das BMS ein und schaltet ab. Das soll aber nur die letzte Instanz zum Schutz der Zellen sein.

Bin für jeden Tipp dankbar

MetaChuh, 2024-01-23:
@GrisuXX 

ich hab' ein ähnliches setup.
(redodo r12v300, hm-300, orion 12/24/20)
früher auch über ac shelly plug an und ausgeschaltet, jetzt über inverter via dtu.

ich verwende pro batterie shelly uni um die volt der batterien zu messen (ca €15,-)
meine steuerung schaltet dann per dtu api `{"id": $inverter_key, "cmd": "power", "val": 0}` den jeweiligen inverter wieder aus, wenn die volts unter 12v gesunken sind.
sprich wenn der inverter nach reboot wieder startet, wird er sofort wieder über den inverter abgedreht.

zusätzlich habe ich die startup drosselung auf 5% relative persistent gesetzt, da die steuerung dann bei bedarf sowieso hochregelt und nicht gleich beim einschalten 100% aus dem akku zieht.

closed 2024-02-17 by lumapu