lumapu/ahoy #1384: Ahoy 0.8.60 wird morgens nicht wach

firedp, 2024-01-22:
## Hardware
  * [x ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ x]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.60
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Guten Morgen.
Nach meiner Auszeit mit der gut laufenden 0.6.12 bin ich bei 0.8.59 und 0.8.60 gelandet und habe mich etwas in die fortgeschrittene Bedienung gearbeitet.
Aber 0.8.60 möchte morgens nicht selbstständig aufwachen und mit den Wrs kommunizieren. Ich habe die Kommunikation mit den WRs schon auf -45 Minuten am Morgen und -45 Minuten nach Sonnenuntergang eingestellt und auch die Koordinaten gemäß Google auf 5 Stellen hinter dem Komma eingegeben.
Aber wach wird Ahoy bei mir nicht ohne Reboot oder stromlos machen.
Einstellungsfehler?

knickohr, 2024-01-22:
Zeig uns mal diese Systemseite.

[image]

firedp, 2024-01-22:
[image]
Hoffe es hilft

knickohr, 2024-01-22:
Hast Du überhaupt den Inverter in den Settings aktiviert ?

firedp, 2024-01-22:
Wie meinst Du das? Du meinst meine 2 Hoymiles hm-600?
Die sind aktiviert, es wird ja 30-60 Minuten aktiv alles angezeigt bevor die DTU ohne Ankündigung abschaltet.

knickohr, 2024-01-22:
Moment langsam !

Also was jetzt ?  Du schreibst das die DTU nicht aufwacht, und nun sendet sie doch für 30 bis 60 Minuten Datem. Was denn nu ?

firedp, 2024-01-22:
Stimmt, ungenau.
Es ist aufgefallen weil sie heute morgen dunkel blieb. Reboot und sie startete wie immer. Dann musste ich aus dem Haus und am Nachmittag war sie wieder aus. Strom ab und dran und es lief wieder, war aber plötzlich gefühlt nach 30-60 Minuten wieder aus.
Sie scheint die eventuell die Verbindung zu den Wechselrichter zu verlieren und deshalb abzuschalten. weil über PC komme ich ja weiterhin dran und kann sie über Software rebooten.
Bis auf das Update habe ich nichts verändert und habe seit 1 Jahr sogar 2 Ahoy und eine openDTU parallel laufen gehabt.

knickohr, 2024-01-22:
OK,

woraus schließt Du das sie aus ist ?  Geht nur das Display aus oder kannst Du sie per Webinterface auch nicht mehr erreichen ?

has5 Du auch mal im Webserial geschaut ob da noch Daten fliegen ?

firedp, 2024-01-23:
Jetzt wird es schwieriger. Ich habe den 2 anderen DTUs gestern Abend den Stecker gezogen und die jetzt alleinige 0.8.62 ist in der Früh angesprungen und läuft bis jetzt.
Ob die anderen DTUs wirklich störend wirken? Wäre kein Drama, aber war früher kein Problem.
Ich werde nach einem fehlerfreien Tag heute nochmal die anderen DTUs einschalten und den morgigen Tag beobachten, wenn ihr denkt das es einen Zusammenhang geben könnte.

knickohr, 2024-01-23:
Du legst uns wohl gerne Eier in den Korb ?  😡

Die (bisher 10) Gebote von Ahoy- und OpenDTU :
1. Gebot :  Du sollst nicht senden ohne Elko !
2. Gebot :  Du sollst den Elko huldigen !
3. Gebot :  Ein Elko ist durch nichts zu ersetzen außer durch noch mehr Elkos.
4. Gebot :  Ohne Elko am NFR-Modul kein Support.
5. Gebot :  Du sollst die Boot-Taste beim ESP32 während des Flashens drücken !
6. Gebot :  Du sollst einen DNS in den Network Setting eintragen wenn Du statische IPs verwendest.
7. Gebot :  Du sollst ein gutes Netzteil und ein „dickes“ USB-Kabel verwenden.
8. Gebot :  Du sollst keine Adapterplatinen anschließen (Psalm 3, beim Heiligen Stefant)
9. Gebot :  Du sollst lesen die Changelogs bevor Du Fragen stellst und Issues aufmachst !
10. Gebot :  Du sollst keine anderen DTUs neben Dir haben !

firedp, 2024-01-23:
Sorry. 
Dann bin ich gut dabei und werde 9 und 10 besonders beachten. Wollte keinen verärgern.

knickohr, 2024-01-23:
Alles fein, ist niemand sauer, aber hättest Du das gleich am Anfang gesagt, dann wäre die Anrütwort prompter gekommen 😉

closed 2024-02-17 by lumapu