lumapu/ahoy #1387: [Bug] Befehle werden endlos gesendet, wenn WR nicht verfügbar ist

Telekatz, 2024-01-22:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.62

### Github Hash

e039820

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

...

### Debug Serial Log output

```shell
0.8.62 - e039820
---------------
18:36:18.834 I: MQTT got topic: inverter/ctrl/limit/0
18:36:19.005 I: (#0) sendControlPacket cmd: 0b
18:36:19.260 I: (#0) sendControlPacket cmd: 0b
18:36:19.517 -----
18:36:19.518 I: (#0) sendControlPacket cmd: 0b
18:36:19.773 I: (#0) sendControlPacket cmd: 0b
18:36:20.033 -----
18:36:20.034 I: (#0) sendControlPacket cmd: 0b
18:36:20.289 I: (#0) sendControlPacket cmd: 0b
18:36:20.544 -----
18:36:20.545 I: (#0) sendControlPacket cmd: 0b
18:36:20.800 I: (#0) sendControlPacket cmd: 0b
18:36:21.060 -----
18:36:21.767 -----
18:36:21.768 I: (#0) sendControlPacket cmd: 0b
18:36:22.027 I: (#0) sendControlPacket cmd: 0b
18:36:22.283 -----
18:36:22.284 I: (#0) sendControlPacket cmd: 0b
18:36:22.539 I: (#0) sendControlPacket cmd: 0b
18:36:22.794 -----
18:36:22.795 I: (#0) sendControlPacket cmd: 0b
18:36:23.055 I: (#0) sendControlPacket cmd: 0b
18:36:23.310 -----
18:36:23.311 I: (#0) sendControlPacket cmd: 0b
18:36:23.566 I: (#0) sendControlPacket cmd: 0b
[... 233 lines cut ...]
18:37:05.672 I: (#0) sendControlPacket cmd: 0b
18:37:05.927 I: (#0) sendControlPacket cmd: 0b
18:37:06.187 -----
18:37:06.187 I: (#0) sendControlPacket cmd: 0b
18:37:06.442 I: (#0) sendControlPacket cmd: 0b
18:37:06.697 -----
18:37:06.698 I: (#0) sendControlPacket cmd: 0b
18:37:06.953 I: (#0) sendControlPacket cmd: 0b
18:37:07.213 -----
18:37:07.213 I: (#0) sendControlPacket cmd: 0b
18:37:07.468 I: (#0) sendControlPacket cmd: 0b
18:37:07.723 -----
18:37:07.724 I: (#0) sendControlPacket cmd: 0b
18:37:07.979 I: (#0) sendControlPacket cmd: 0b
18:37:08.234 -----
18:37:08.235 I: (#0) sendControlPacket cmd: 0b
18:37:08.490 I: (#0) sendControlPacket cmd: 0b
18:37:08.745 -----
18:37:08.746 I: (#0) sendControlPacket cmd: 0b
18:37:09.006 I: (#0) sendControlPacket cmd: 0b
```

### Error description

Wenn ein Befehl (Inverter on/off/restart/limit) an einen WR gesendet wird, der nicht verfügbar ist, wird der Befehl immer wieder erneut gesendet. Hier wäre es sinnvoll, den Sendevorgang irgendwann abzubrechen, wenn der WR nicht antwortet.

lumapu, 2024-01-22:
danke für den Hinweis

khbraemer, 2024-01-22:
Hallo, bei gleichem Code 08.62  sieht Webserial folgendermassen aus, was fuer mich in Ordnung ist.

0.8.62 - e039820
---------------
21:17:02.704 I: MQTT got topic: HoymilesinverterDach/ctrl/limit/0
21:17:03.005 I: (#0) no communication to the inverter (night time)
21:17:03.007 I: (#0) no communication to the inverter (night time)
21:17:17.748 I: MQTT got topic: HoymilesinverterDach/ctrl/limit/0
21:17:18.005 I: (#0) no communication to the inverter (night time)
21:17:18.006 I: (#0) no communication to the inverter (night time)
21:17:33.000 I: (#0) no communication to the inverter (night time)
21:17:47.732 I: MQTT got topic: HoymilesinverterDach/ctrl/limit/0
21:17:48.005 I: (#0) no communication to the inverter (night time)
21:17:48.006 I: (#0) no communication to the inverter (night time)
21:18:03.000 I: (#0) no communication to the inverter (night time)
21:18:18.000 I: (#0) no communication to the inverter (night time)
21:18:33.000 I: (#0) no communication to the inverter (night time)
21:18:47.700 I: MQTT got topic: HoymilesinverterDach/ctrl/limit/0
21:18:48.006 I: (#0) no communication to the inverter (night time)
21:18:48.008 I: (#0) no communication to the inverter (night time)
21:19:03.000 I: (#0) no communication to the inverter (night time)

sunrise	23.1.2024, 08:21:57
sunset	23.1.2024, 17:09:29
Communication start	23.1.2024, 08:21:57
Communication stop	23.1.2024, 17:09:29
Night behaviour	not communicating

lumapu, 2024-01-22:
warum läuft das Script in der Nacht weiter? Naja stört ja nicht aber sieht komisch aus 😉

Telekatz, 2024-01-23:
Die Version 0.8.63 77b1f96 hat das Problem leider nicht behoben.

lumapu, 2024-01-24:
wie kann ich es nachstellen?

Telekatz, 2024-01-25:
- WR über den DC Anschluss mit Spannung versorgen
- Ahoy starten und warten bis der WR erkannt wird und Daten empfangen werden
- DC Anschluss am WR trennen und etwas warten bis die Kondensatoren leer sind
- Ein Power Limit zum WR senden (wenn noch ein ACK vom WR kommt etwas länger warten und erneut senden, dann sind die Kondensatoren noch nicht leer)

lumapu, 2024-01-25:
ist ein Akku-Szenario, das kann ich leider nicht nachstellen, oder ist das eine "Nachtsimulation"? Ich kann trotzdem mal aus Codesicht darüber nachdenken.

Telekatz, 2024-01-26:
Wenn man die Option abstellt, die nachts die Kommunikation pausiert, sollte es auch möglich sein, das Problem in der Nacht nachzustellen.

lumapu, 2024-02-04:
bitte mit `0.8.71` testen

Telekatz, 2024-02-04:
Außer dass jetzt auch noch der Bug drin ist, dass sich ein ausgeschalteter Inverter nicht wieder einschalten lässt, wird das Problem nicht vollständig verhindert. Man kann immer noch während der 5 Minuten, in denen der Inverter nicht mehr verfügbar ist und Ahoy das noch nicht festgestellt hat, Befehle an den Inverter schicken, die ständig wiederholt werden. Diese werden auch dann noch weiter gesendet, nachdem festgestellt wurde, das der Inverter nicht mehr verfügbar ist.

lumapu, 2024-02-17:
bitte nochmals um einen Test mit der aktuellen Version

Telekatz, 2024-02-17:
Hab die Version 0.8.83 getestet. Das Problem ist immer noch vorhanden. Bis zu 15 Minuten nach dem letzten Empfang vom Inverter lassen sich Befehle an den Inverter schicken. Ohne ACK vom Inverter werden die endlos wiederholt. Bis der Inverter wieder Antwortet.

```shell
0.8.83 - 5ebfe5a
14:41:57.129 I: (#0) RX  38ms | 27 CH40 | 95 01
14:41:57.131 I: (#0) RX  82ms | 23 CH23 | 95 82
14:41:57.133 -----
14:41:57.314 I: (#0) RX  39ms | 27 CH61 | 95 01
14:41:57.315 I: (#0) RX  86ms | 27 CH40 | 95 02
14:41:57.316 I: (#0) RX 134ms | 23 CH40 | 95 83
14:41:57.318 -----
14:41:57.319 I: com loop duration: 318ms
14:41:57.319 -----
14:42:09.005 -----
14:42:10.411 -----
14:42:10.412 I: com loop duration: 3411ms
14:42:10.412 -----
14:42:18.406 -----
14:42:18.407 I: com loop duration: 1406ms
14:42:18.407 -----
14:42:23.273 I: MQTT got topic: inverter/ctrl/limit/0
14:42:24.006 I: (#0) sendControlPacket cmd: 0b
14:42:24.258 I: (#0) sendControlPacket cmd: 0b
14:42:24.510 I: (#0) sendControlPacket cmd: 0b
14:42:24.762 I: (#0) sendControlPacket cmd: 0b
14:42:25.014 -----
14:42:25.015 I: (#0) sendControlPacket cmd: 0b
14:42:25.267 I: (#0) sendControlPacket cmd: 0b
14:42:25.519 I: (#0) sendControlPacket cmd: 0b
14:42:25.771 I: (#0) sendControlPacket cmd: 0b
14:42:26.023 -----
14:42:26.024 I: (#0) sendControlPacket cmd: 0b
14:42:26.276 I: (#0) sendControlPacket cmd: 0b
14:42:26.528 I: (#0) sendControlPacket cmd: 0b
14:42:26.780 I: (#0) sendControlPacket cmd: 0b
14:42:27.032 -----
14:42:28.437 -----
14:42:28.438 I: (#0) sendControlPacket cmd: 0b
14:42:28.690 I: (#0) sendControlPacket cmd: 0b
14:42:28.942 I: (#0) sendControlPacket cmd: 0b
14:42:29.194 I: (#0) sendControlPacket cmd: 0b

...

16:34:02.400 -----
16:34:02.401 I: (#0) sendControlPacket cmd: 0b
16:34:02.653 -----
16:34:02.654 I: (#0) sendControlPacket cmd: 0b
16:34:02.906 -----
16:34:02.907 I: (#0) sendControlPacket cmd: 0b
16:34:03.159 -----
16:34:03.160 I: (#0) sendControlPacket cmd: 0b
16:34:03.412 -----
16:34:03.413 I: (#0) sendControlPacket cmd: 0b
16:34:03.665 -----
16:34:03.666 I: (#0) sendControlPacket cmd: 0b
16:34:03.918 -----
16:34:03.919 I: (#0) sendControlPacket cmd: 0b
16:34:04.171 -----
16:34:04.172 I: (#0) sendControlPacket cmd: 0b
16:34:04.424 I: (#0) RX  42ms | 17 CH40 | d1 81
16:34:04.425 I: (#0) has accepted power limit set point 30.00 with PowerLimitControl 0
16:34:04.428 -----
16:34:04.429 I: com loop duration: 6700424ms
16:34:04.430 -----
16:34:07.169 I: (#0) RX  26ms | 27 CH61 | 95 01
16:34:07.171 I: (#0) RX  73ms | 27 CH40 | 95 02
16:34:07.172 I: (#0) RX 120ms | 23 CH40 | 95 83
16:34:07.173 I: alarm ID incremented to 4
16:34:07.175 -----
16:34:07.175 I: com loop duration: 175ms
16:34:07.176 -----
16:34:17.182 I: (#0) RX  42ms | 27 CH61 | 95 01
16:34:17.184 I: (#0) RX  85ms | 27 CH61 | 95 02
16:34:17.185 I: (#0) RX 134ms | 23 CH40 | 95 83
```

lumapu, 2024-08-16:
bitte den Stand der Dinge aktualisieren, ich habe das etwas aus dem Fokus verloren.

stefan123t, 2024-10-29:
@Telekatz heute ist Katertag, bitte noch einmal mit v0.8.152 testen und den aktuellen Stand zurückmelden. Danke!
Vielleicht können wir das Issue ja dann :crossed_fingers: schließen ?

khbraemer, 2024-10-30:
Hallo, von meiner Seite ist alles ok.Mit freundlichen GrüßenKarl-Heinz Brämer

Vielleicht können wir das Issue ja dann 🤞 schließen ?

stefan123t, 2024-10-30:
@Telekatz wenn es für Dich auch in Ordnung ist kannst Du Dein Issue als gelöst schließen ?

Telekatz, 2024-10-30:
Der Fehler ist leider auch in v0.8.152 noch vorhanden.

Ich habe den Inverter zum Testen an ein Netzteil angeschlossen. Getestet habe ich wie folgt:
- Der Inverter war am Anfang des Tests in Ahoy verfügbar.  
- Nach Abschalten des Netzteils wurde ab 19:19:50.429 nichts mehr empfangen. 
- Um 19:20:06.741 habe ich einen Ausschaltbefehl gesendet, der dann ständig wiederholt wurde. 
- Um 19:24:50 hat Ahoy den Status des Inverters auf nicht verfügbar geändert. Der Befehl wurde trotzdem weiter gesendet. 
- Das Netzteil wurde um 19:32:00 wieder eingeschaltet.
- Um 19:32:42.292 hat der Inverter dann wieder geantwortet. 

`0.8.152 - 78cbd77
---------------
19:19:30.328 I: (#0) RX  29ms | 27 CH23 | 95 81
19:19:30.331 -----
19:19:30.495 I: (#0) RX  21ms | 27 CH03 | 95 01
19:19:30.496 I: (#0) RX  68ms | 27 CH03 | 95 02
19:19:30.497 I: (#0) RX 116ms | 23 CH75 | 95 83
19:19:30.500 -----
19:19:30.501 I: com loop duration: 501ms
19:19:30.501 -----
19:19:33.766 I: MQTT got topic: inverter/ctrl/limit/0
19:19:33.768 I: json: {"val":50,"path":"ctrl","cmd":"limit_nonpersistent_absolute","id":0}
19:19:33.771 I: (#0) sendControlPacket cmd: 0b
19:19:34.023 I: (#0) RX  30ms | 17 CH03 | d1 81
19:19:34.024 I: (#0) has accepted power limit set point 50.00 with PowerLimitControl 0
19:19:34.028 -----
19:19:34.030 I: com loop duration: 260ms
19:19:34.030 -----
19:19:35.078 I: (#0) RX  26ms | 27 CH23 | 95 81
19:19:35.081 -----
19:19:35.247 I: (#0) RX  18ms | 27 CH23 | 95 01
19:19:35.249 I: (#0) RX  72ms | 27 CH03 | 95 02
19:19:35.250 I: (#0) RX 119ms | 23 CH03 | 95 83
19:19:35.252 -----
19:19:35.253 I: com loop duration: 248ms
19:19:35.253 -----
19:19:40.352 I: (#0) RX  28ms | 27 CH03 | 95 01
19:19:40.354 I: (#0) RX  76ms | 27 CH03 | 95 02
19:19:40.672 I: (#0) RX  18ms | 23 CH03 | 95 83
19:19:40.674 -----
19:19:40.675 I: com loop duration: 676ms
19:19:40.676 -----
19:19:45.352 I: (#0) RX  34ms | 27 CH03 | 95 01
19:19:45.354 I: (#0) RX  83ms | 27 CH03 | 95 02
19:19:45.420 I: (#0) RX  18ms | 23 CH03 | 95 83
19:19:45.423 -----
19:19:45.499 I: (#0) RX  28ms | 17 CH03 | 95 81
19:19:45.501 I: (#0) Inv loss: 2 of 20, DTU loss: 5 of 38. ACKs: 16
19:19:45.502 -----
19:19:45.502 I: com loop duration: 502ms
19:19:45.503 -----
19:19:50.361 I: (#0) RX  32ms | 27 CH03 | 95 01
19:19:50.363 I: (#0) RX  80ms | 27 CH03 | 95 02
19:19:50.426 I: (#0) RX  15ms | 23 CH23 | 95 83
19:19:50.429 -----
19:19:50.429 I: com loop duration: 421ms
19:19:50.430 -----
19:19:55.704 -----
19:19:55.705 I: com loop duration: 705ms
19:19:55.706 -----
19:20:00.703 -----
19:20:00.704 I: com loop duration: 704ms
19:20:00.705 -----
19:20:05.704 -----
19:20:05.705 I: com loop duration: 705ms
19:20:05.706 -----
19:20:06.741 I: MQTT got topic: inverter/ctrl/power/0
19:20:06.742 I: json: {"val":0,"path":"ctrl","cmd":"power","id":0}
19:20:06.745 I: (#0) sendControlPacket cmd: 01
19:20:06.997 I: (#0) sendControlPacket cmd: 01
[... 163 lines cut ...]
19:32:45.179 I: (#0) RX 123ms | 23 CH40 | 95 83
19:32:45.181 -----
19:32:45.182 I: com loop duration: 177ms
19:32:45.182 -----
19:32:50.199 I: (#0) RX  47ms | 27 CH40 | 95 01
19:32:50.201 I: (#0) RX  94ms | 27 CH40 | 95 02
19:32:50.211 I: (#0) RX 142ms | 23 CH23 | 95 83
19:32:50.215 -----
19:32:50.216 I: com loop duration: 208ms
19:32:50.217 -----
19:32:55.056 W: RX other inverter 95 83 24 05 ea f4 d7 c8 fa dd 56 ab b4 ce eb 4a 92 f6 90 ea ee 5b 26 c8 fb 68 ec 
19:32:55.179 I: (#0) RX  37ms | 27 CH40 | 95 01
19:32:55.180 I: (#0) RX  58ms | 27 CH40 | 95 01
19:32:55.181 I: (#0) RX  84ms | 27 CH23 | 95 02
19:32:55.182 I: (#0) RX 132ms | 23 CH23 | 95 83
19:32:55.185 -----
19:32:55.186 I: com loop duration: 186ms
19:32:55.186 -----
19:33:00.164 I: (#0) RX  27ms | 27 CH61 | 95 01
19:33:00.166 I: (#0) RX  67ms | 27 CH40 | 95 02
19:33:00.167 I: (#0) RX 117ms | 23 CH40 | 95 83
19:33:00.170 -----
19:33:00.170 I: com loop duration: 171ms
19:33:00.171 -----
`

stefan123t, 2024-10-30:
@Telekatz danke fürs Testen und die Fehler-Beschreibung:

> Der Fehler ist leider auch in v0.8.152 noch vorhanden.
> Ich habe den Inverter zum Testen an ein Netzteil angeschlossen. Getestet habe ich wie folgt:
> * Der Inverter war am Anfang des Tests in Ahoy [...]

Was soll an dieser Stelle z.B. nach Sonnenuntergang geschehen ?
Sollen Befehle wenn sie vom Wechselrichter nicht empfangen / quittiert werden nach einer gewissen Zeit bzw. wenn der WR als nicht erreichbar markiert wurde aus der Queue entfernt werden ?

Wie sollen wir den Umstand, daß wir einen Befehl zuerst angenommen haben, diesen aber nicht an den WR senden konnten, gegenüber dem Auftraggeber (REST API, MQTT Topic, etc.) zurückmelden ?
Der ursprüngliche REST API Call / MQTT Request an die AhoyDTU ist an dieser Stelle ja längst abgearbeitet und ad acta gelegt.
Brauchen wir dazu einen Error Status / Callback Handler und/oder ein extra Command Queue Error topic ?

> * Um 19:24:50 hat Ahoy den Status des Inverters auf nicht verfügbar geändert. Der Befehl wurde trotzdem weiter gesendet.
> * Das Netzteil wurde um 19:32:00 wieder eingeschaltet.
> * Um 19:32:42.292 hat der Inverter dann wieder geantwortet.

Telekatz, 2024-10-30:
> Sollen Befehle wenn sie vom Wechselrichter nicht empfangen / quittiert werden nach einer gewissen Zeit bzw. wenn der WR als nicht erreichbar markiert wurde aus der Queue entfernt werden ?
>
> Wie [...]

Meiner Meinung nach sollte die Anzahl der Versuche, den Befehl zu senden, begrenzt werden. Und gegenüber dem Auftraggeber sollte jeder Befehl (nicht nur Limitänderungen) mit einem ACK oder NACK bestätigt werden.

stefan123t, 2024-11-02:
> Meiner Meinung nach sollte die Anzahl der Versuche, den Befehl zu senden, begrenzt werden. 

Ja irgendwann wird der Inverter auch als nicht erreichbar markiert. Spätestens da ist es eigentlich unsinnig immer noch weiter das Kommando zu wiederholen. Andererseits, wenn der Inverter wieder erreichbar ist, müssten wir für jedes Kommando klären ob es so wichtig ist/war, daß es dann immer noch / noch einmal gesendert werden soll, z.B. ein persistent Limit ?

> Und gegenüber dem Auftraggeber sollte jeder Befehl (nicht nur Limitänderungen) mit einem ACK oder NACK bestätigt werden.

Das wird schwierig, da die AhoyDTU sowohl über die REST API als auch über die MQTT topics die Befehle zwar entgegennimmt aber dann davon asynchron abarbeitet. D.h. wir müssten entweder einen eigenen Fehlerspeicher analog zu den Events des WR führen, über den wir dann fehlerhafte Befehle mitloggen oder anderweitig einen Rückkanal für den Callback initieren.