lumapu/ahoy #1389: Falsche Anzeige auf Display

Edibo63, 2024-01-23:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.64

### Github Hash

?

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

update auf 0.8.64

### Error description

In der Anzeige im epaper  steht WIFI not connected.
Es funktioniert aber alles.

Edibo63, 2024-01-23:
Korrektur:  ist ein ESP32

knickohr, 2024-01-23:
Kann ich nicht bestätigen 😲

Edibo63, 2024-01-23:
Stand den ganzen Tag auf not connected, war aber verbunden und Zugriff möglich.
Reboot zeigte keine Änderung.
Update nochmal gemacht danach kurz Stromversorgung weg.
Jetzt zeigt es wieder die IP an.

closed 2024-01-23 by Edibo63