lumapu/ahoy #1390: Stromaufnahme nach Sunset dev 0.8.65

rmayergfx, 2024-01-25:
ESP32-wroom.de DEV 
Habe aus der Zeit vor AhoyDTU noch einen Shelly Pro 1PM verbaut. Der zeigt mir im 10 sek. Takt nach Sonnenuntergang permanent Verbrauch an, 0,032A 1.1W, wobei die 1.1W so alle 5 Sekunden an und aus gehen. Auf jeden Fall erkennt der Pro 1PM hier einen Verbrauch und das wundert mich doch etwas. Verbaut ist ein Display, das auch nach Sonnenuntergang ausgeschaltet wird, dennoch schläft der ESP32 nicht wirklich, sondern schickt da anscheinend noch fleissig Daten oder irgendwas verbraucht da noch unnötig strom.

dtuuser, 2024-01-25:
Hatte vor Ahoy auch einen Shelly zwischen WR und Netz. Mein WR zieht 1 bis 1,5 Watt im Leeelauf, also nachts

lumapu, 2024-01-25:
gehts jetzt um die Stromaufnahme von Ahoy oder von dem Wechselrichter? Ich glaube @rmayergfx meinte Ahoy

knickohr, 2024-01-26:
Naja, wir verbrauchen auch Kalorien wenn wir schlafen 😅

rmayergfx, 2024-01-26:
@knickohr 
ja damit wir schlank bleiben
was mich wundert ist das der Stromverbrauch nicht kontinuierlich erscheint sondern alle paar Sekunden ein Peek im System erscheint. Da die Settings für Sunset/Sunrise ja eigentlich super funktionieren wäre es die Frage ob man das System dann nicht noch tiefer schlafen lassen könnte, mir scheint es so als würde er alle paar Sekunden irgendwas prüfen Netzwerk, WLAN Lan, keine Ahnung, denn das könnte er sich ja bis zum Sunrise sparen.

Mein Workaround aktuell ist ein Schedule im Shelly Pro 1PM das 20 min. nach Sunset und 15 min vor Sunrise das ganze einfach Nachts komplett abschaltet und somit keinerlei Strom verbraucht. Nur ohne Shelly geht das halt beim Otto Normalverbraucher nicht und da besteht die Frage, ob man hier noch die Settings optimieren kann, das Nachts wirklich der Stromverbrauch nahe 0 ist und das Teil schläft und nur zu den gesetzten Uhrzeiten aufwacht.

Joh49, 2024-01-26:
<<wenn man nicht extern komplett ab und an schaltet muß die DTU ja immer wieder auf die Uhr sehen um zum Sonnenaufgang wieder voll aktiv zu sein. Es ist also eine geringe Hintergrundaktivität auch nachts notwendig.

rmayergfx, 2024-01-26:
Es gibt für ESP32 und ESP8266 verschiedene Sleep Modi, Deep Sleep ist davon die effizienteste und da läuft nur noch intern RTC damit er wieder aufwacht. 
https://www.electronicshub.org/esp8266-deep-sleep-mode/
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html

Da könnte man bestimmt noch optimieren.

knickohr, 2024-01-26:
Deep Sleep geht vermutlich in die Hose, weil dann auch das Webinterface und MQTT nicht mehr funktioniert. Sprich, es wird von Sonnenuntergang bus Aufgang nicht, gar nichts mehr kommen. Ergo kannste auch nicht mal mehr auf die Statistik schauen.

rmayergfx, 2024-01-26:
Meine Statistik läuft auf HomeAsssitant. Gibt auch andere Lösungen. AhoyDTU dient mir nur rein zur Datenübermittlung.
Ich brauche Nachts kein Webinterface oder gar MQTT, MQTT sendet auch nichts wenn man mit Sunset arbeitet.

dtuuser, 2024-01-26:
Falls über den Deep Sleep nachgedacht wird, dann bitte einen Schalter zum Aktivieren einbauen, da ich persönlich auch nach Sonnenuntergang auf das Webinterface zugreife. Abends auf der Couch mal schnell nachschauen, was die einzelnen Zellen so am Tag geleistet haben.

Peertje-62, 2024-01-27:
Hi, something to think about
I have measured my esp 32 (Dev 0.8.65) on the DC side varies between 0,01;0.02 & 0,05 A would be at 5,13 Volt somewhere between 0,05 and 0,25 watt, I think thats okay 

May be I am wrong but;
''Shelly pro pm measures on the AC side, at 220V and 1,1 Watt that would be 0,005A !
or 0,032 A at 220 would be 7 Watt !, which one is it?
(You cann't fix it if it's not broken)
regards

rmayergfx, 2024-01-28:
Take your calculator. 7W, 24h, 365d would be 61320Wh 61,3kWh. The shortest night midsommer is about 8 hours,
so if we only calc with 8h offline we are still at 20 kWh. Depending on an average of 42.29 cents per kilowatt we are at ~ 8€ a year you are paying if there is no batterie you are charching with solar. so it would be a good idea to get this value as low as possible.
with balcony solar you will not get money back from your energy provider, you have to pay at night if you have no batteries.

fschmeidl, 2024-12-13:
+1 for this feature.

The power consumption might be little but with more and more ahoy dtu users it sums up. Why not reduce it, if possible?

Additionally, some people want to reduce the radio radiation in their homes during night. The antenna can already be turned off but the esp is still accessing the wifi (or at least trying to (some people also turn their wifi off during the night)). No need to discuss the sense in this, but we should understand that some people concerned about this.

Here and in other threads people have pointed out issues with the user experience when you suddenly can't access the dashboard anymore. It think this can be addressed with a warning popup when either selcting the deep-sleep mode or saving it. The default should be of course no deep sleep so that the ahoy dtu can be set up at what ever time.