lumapu/ahoy #1391: [Bug] ESP8266 Sporadische Disconnects / Restarts - Exception (29)

n14ro, 2024-01-26:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.36

### Github Hash

6fb0535

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

- Reboot at midnight
- LEDs off
- added 2 Inverter with each 2 modules
- NPT: pool.ntp.org
- MQTT to Home Assistant - Topic: /Solar/Gruendach
- MQTT Interval: 5s

### Debug Serial Log output

```shell
2024-01-26 14:31:34	åÇn«rêÚn|Ïlƒl`bbrl‚nbÇnl`årlƒl‰˛I: Initializing FS ..
I:  .. done
I: Config Version: 7
I: Settings valid: true
I: Radio Config:
SPI Frequency		= 1 Mhz
Channel			= 3 (~ 2403 MHz)
Model			= nRF24L01+
RF Data Rate		= 250 KBPS
RF Power Amplifier	= PA_LOW
RF Low Noise Amplifier	= Enabled
CRC Length		= 16 bits
Address Length		= 5 bytes
Static Payload Length	= 32 bytes
Auto Retry Delay	= 1000 microseconds
Auto Retry Attempts	= 15 maximum
Packets lost on
    current channel	= 0
Retry at
2024-01-26 14:31:34	tempts made for
    last transmission	= 15
Multicast		= Allowed
Custom ACK Payload	= Disabled
Dynamic Payloads	= Enabled
Auto Acknowledgment	= Enabled
Primary Mode		= RX
[... 1623 lines cut ...]
- Die Sonne ist untergegangen
- 23:00 Uhr Das Geht läuft jetzt seit 6 ohne Neustart. Wahrscheinlich bis heute nacht 0 Uhr und dann bis ca. 9 Uhr, wenn die ersten Sonnenstrahlen auf die Module scheinen.

Auszug vom Netzwerk zum Zeitpunkt der USB-Aufzeichnung:
[image]

Netzwerkumgebung:

- Fritzbox im Keller
- Accesspoint im EG (ubiquity LR)
- Accesspoint im OG (ubiquity pro)

Aufbau des Gerätes: 

- verbunden mit kurzen dicken Kabeln - gelötet
- Antennen schauen voneinander weg
- kurzes USB-Kabel am alten Mac-mini
[image]

Zu der Exception konnte ich bisher nur adurioforum was finden: https://www.arduinoforum.de/arduino-Thread-8266-Abstürze-mit-Fehler-„Exception-29-“?page=4
Ich bin mir nicht sicher, ob hier ein heap Problem vorliegt. Der sah zumindest bei den Stichproben während der Aufzeichnung immer recht konstant aus:
[image]

Wer bis hier gekommen ist, erstmal vielen Dank und
Grüße

You69Man, 2024-01-27:
Das ist einmal eine richtig guter, und ausführliche Fehlerbeschreibung! 
Auffällig ist, dass die disconnects häufig genau zur vollen Stunde passieren:
[image]
Das schließt jedenfalls Hardwareprobleme einmal aus. 
Was genau passiert jede volle Stunde?

n14ro, 2024-01-27:
Ich wüsste nicht, was netzwerktechnisch anders sein sollte als nachts. Bisher sehe ich das Verhalten nur beim Ahoy-Gerät. Andere Geräte z.B. ein HomePod und sonstige mobile Endgeräte zeigen das Verhalten nicht. Kann man mit den vielen Daten aus dem ersten Post (z.B. dem Crashdump?) was anfangen?

Das Gerät liegt sonst einsam aufm Küchenschrank.

Grüße

lumapu, 2024-01-27:
ich weiß nicht genau, ob ich schlau werde aus der Exception, hier mal dekodiert:

<details>
<summary>Exception</summary>

```
Exception: 29 (StoreProhibited: A store referenced a page mapped with an attribute that does not permit stores)

epc1:     0x4025ccdf: ieee80211_add_xrates at ??:?
epc2:     0x00000000
epc3:     0x00000000
excvaddr: 0x00000019
depc:     0x00000000

ctx: sys

sp:       0x3fffed10
end:      0x3fffffb0
offset:   0x00000150

stack:
0x4025ce1b: ieee80211_send_probereq at ??:?
0x4026c4c6: chip_v6_rxmax_ext at ??:?
0x4025ede3: scan_remove_probe_ssid at ??:?
0x4025e9b4: scan_start at ??:?
0x4025e534: ieee80211_regdomain_chan_in_range at ??:?
0x4026aecc: chip_v6_set_chanfreq at ??:?
0x4024e541: mem_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c:211
0x4024e5a0: do_memp_malloc_pool at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:295
0x4025e9ef: scan_start at ??:?
0x4024e5d0: memp_malloc at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c:356
0x40260968: chm_start_op at ??:?
0x40252d39: sys_timeout_abs at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c:205
0x4026092c: chm_start_op at ??:?
0x4025e9a4: scan_start at ??:?
0x4025e978: scan_start at ??:?
0x4025e943: scan_start at ??:?
0x40252f28: sys_check_timeouts at /local/users/gauchard/arduino/arduino_esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/timeouts.c:396
0x4025eeb1: scan_remove_probe_ssid at ??:?
0x40277d31: ets_timer_handler_isr at ??:?
0x40277d76: ets_timer_handler_isr at ??:?
0x4023cb46: loop_task(ETSEventTag*) at core_esp8266_main.cpp:?
0x40105c91: call_user_start_local at ??:?
0x40105c97: call_user_start_local at ??:?
0x4010000d: call_user_start at ??:?
0x401000ab: app_entry_redefinable at ??:?
0x3ffe8368: ?? at /workdir/repo/newlib/newlib/libc/reent/impure.c:23
0x40249764: __ssputs_r at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf.c:232
0x401005cc: ets_post at ??:?
0x401005cc: ets_post at ??:?
0x401005cc: ets_post at ??:?
0x40105b77: lmacTxFrame at ??:?
0x40106429: ets_timer_disarm at ??:?
0x401005cc: ets_post at ??:?
0x401005a4: ets_intr_unlock at ??:?
0x401028e2: pp_post at ??:?
0x40104c8b: lmacProcessTxSuccess at ??:?
0x40106429: ets_timer_disarm at ??:?
0x40103b54: wDev_ProcessFiq at ??:?
0x40105add: lmacTxFrame at ??:?
0x401005cc: ets_post at ??:?
0x401005cc: ets_post at ??:?
0x40102d29: rcUpdateTxDone at ??:?
0x40105b77: lmacTxFrame at ??:?
0x401028f8: pp_post at ??:?
0x40105b77: lmacTxFrame at ??:?
0x401005cc: ets_post at ??:?
0x401005cc: ets_post at ??:?
0x40102d29: rcUpdateTxDone at ??:?
0x40105b8f: lmacRxDone at ??:?
[... 22 lines cut ...]
0x401005cc: ets_post at ??:?
0x401028f8: pp_post at ??:?
0x4027334b: ppTxPkt at ??:?
0x4025cfa7: ieee80211_send_probereq at ??:?
0x4025de7c: ieee80211_freedom_output at ??:?
0x4025ede3: scan_remove_probe_ssid at ??:?
0x4025e9b4: scan_start at ??:?
0x40261028: cnx_start_handoff_cb at ??:?
0x4010656d: ets_timer_arm_new at ??:?
0x402609b7: chm_start_op at ??:?
0x40261028: cnx_start_handoff_cb at ??:?
0x4026092c: chm_start_op at ??:?
0x4025e9a4: scan_start at ??:?
0x4025e978: scan_start at ??:?
0x4025e943: scan_start at ??:?
0x4027a0c4: wifi_station_ap_number_set at ??:?
0x4025eef4: scan_remove_probe_ssid at ??:?
0x40261028: cnx_start_handoff_cb at ??:?
0x4025e921: scan_start at ??:?
0x401005cc: ets_post at ??:?
```
</details>

Auffällig finde ich die `scan_start`, das würde zu dem WiFi-loss passen. Wird die API verwendet? Die Exceptions bitte nicht kürzen, die brauche ich komplett, musste jetzt einige Zeilen annehmen, weiß daher nicht, ob das Ergebnis stimmt.
Interessant wäre auch wie sich die Heap-Fragmentation verhält. Es würde dafür sprechen, dass bei zwei Invertern schneller ein Reboot kommt, wenn die Heapfragmentation hoch ist bzw. steigt.

n14ro, 2024-01-27:
Tut mir Leid. Ich musste den Code kürzen, da nur rund 65000 Zeichen erlaubt waren. Die Sachen die ich im Dump gekürzt habe sind Wiederholungen. Das heißt, überall da wo ich was rausgeschnitten habe (wie vor und nach dem … cut …) stand dieses „feefeffe“. 

Was meinst du mit API? Ich verwende eigentlich nur MQTT zu homeassistant. Selten, dass mal die Webseite übern Browser angesteuert wird.

lumapu, 2024-01-28:
ok, genau diesen Teil habe ich dann wieder erzeugt, in der Annahme, dass sich die Zeilen wiederholen.

Die API ist eine Schnittstelle, die des öfteren von externen Tools wie einer Hausautomatisierung abgefragt wird, deswegen frage ich. Wenn du das nicht verwendest und auch die Website selten aufrufst wird das nicht das Problem sein.

Evtl. hilft es wenn du die heap_fragmentation per MqTT mitloggen könntest um zu sehen, wie die sich über die Zeit entwickelt.

n14ro, 2024-01-28:
Vielen Dank für eure Hilfe soweit. Wie kann ich die heap_fragmantation per MQTT mitloggen? Geht es ggf. auch mit USB?
Heute wurde das Gerät um 12:30 wieder am USB angesteckt. Wenig später zeigte sich das gleiche Spiel. Wird zur vollen Stunde irgend eine Zeit gesynced? Timeserver vielleicht?
[image]

lumapu, 2024-01-28:
Die Zeit wird nur alle 12 Stunden synchronisiert. Heap Fragmentation wird nur per MqTT ausgegeben. Ich würde empfehlen den Haken bei `print inverter data` mal zu entfernen, dann wird das Serielle Log nicht ganz so "zugespamt".

theunknownstarman, 2024-01-29:
Ich hätte ähnliche Probleme und konnte sie in den Unifi WLAN Einstellungen für meine IoTs identifizieren.

Evtl. hilft Dir das weiter?

[image]

n14ro, 2024-01-29:
Unter welchen Topic finde ich heap_fragmentation? 
[image]

@theunknownstarman Danke für den Tipp. Bis auf BSS ist es wohl identisch. Unter Sicherheit habe ich folgendes gefunden:

[image]

Dort sind 3600s eingestellt. Also exakt 1h. Ich habe die Funktion mal deaktiviert und beobachte das Verhalten.

lumapu, 2024-01-29:
in deinem Screenshot der letzte Eintrag heißt `heap_frag`

n14ro, 2024-01-29:
> in deinem Screenshot der letzte Eintrag heißt `heap_frag`

Oh man… Danke. 
Ich lasse den MQTT Explorer mal mitlaufen.

n14ro, 2024-02-03:
Ich habe den MQTT Explorer die letzten Tage laufen lassen. Leider zeigt er aber nur die letzten paar Stunden an.

Nachfolgend die Nachrichten im Graphen, die MQTT-Explorer mitgeschnitten hat. Kann man aus denen was lesen?

[image]

Zum vergleich wieder ein Ausschnitt aus dem Netzwerk:

[image]

Auffällig ist wieder, dass das System bis 9 Uhr sauber läuft, dann ab PV-Ertrag, die Probleme beginnen. Sogar so, dass der MQTT-Teil relativ schnell abschmiert. Seit 17 Uhr war dann wieder Ruhe. 

Ich werde jetzt noch diese BSS Transition (wie von @theunknownstarman vorgeschlagen)abschalten und weiter beobachten.

lumapu, 2024-02-03:
soweit sind die Graphen schon hilfreich und sehen gut aus - bis auf die stündliche Uptime. Ein Test wäre auch die DTU für paar stunden im AP-Modus laufen zu lassen, hierzu einfach eine ungültige WLAN SSID eintragen und speichern + Neustart.

n14ro, 2024-02-06:
Moin, ich hatte das Rekeying vorvorgestern im Netzwerk abgeschaltet. Vorgestern lief der ESP inkl. MQTT bis 21:30 Uhr durch (geprüft durch Uptime). Im Homeassistant kamen alle MQTT-Nachrichten an. Ich hatte dann 21 Uhr trotzdem nochmal USB-Log angeworfen und mitlaufen lassen. 

Leider ist gestern morgen wieder die MQTT-Verbindung abgerissen "MQTT disconnected, reason: TCP disconnect". Auch über die Webseite lies sich das Gerät nicht erreichen. Es hat sich über den Tag auch nicht gefangen (wie sonst durch die Reconnects/Neustarts durch das Rekeying). Ich habe Abends einen harten Neustart durchgeführt.

```shell
I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 2343, s: 175, f: 25, n: 2143 | p: 2
I: (#1) TX 27 CH40 | 15  *  *  *  * 81 92 06 36 80 0B 00 65 C0 A0 4D 00 00 00 00 00 00 00 00 94 BE 5B 
I: (#1) request timeout: 500ms
-----
I: com loop duration: 1775ms
-----
I: (#0) Radio infos: -6 -6 -6 4 -6 | t: 2344, s: 263, f: 21, n: 2060 | p: 255
I: (#0) TX 27 CH61 | 15  *  *  *  * 81 92 06 36 80 0B 00 65 C0 A0 5B 00 00 00 00 00 00 00 00 F4 58 84 
I: (#0) RX  87ms | 27 CH03 | 95  *  *  *  *  *  *  *  * 02 6E C2 00 02 9F A7 00 03 00 03 08 E3 13 87 00 42 3C 
I: (#0) RX 147ms | 23 CH03 | 95  *  *  *  *  *  *  *  * 83 00 00 00 03 03 E8 00 69 00 01 6A 8F 73 
W: (#0) frame 1 missing: request retransmit (3 attempts left)
I: (#0) TX 11 CH61 | 15  *  *  *  * 81 92 06 36 81 7C 
I: (#0) RX  34ms | 27 CH03 | 95  *  *  *  *  *  *  *  * 01 00 01 01 1C 00 0C 00 23 01 39 00 0B 00 23 00 00 B7 
I: (#0) Payload (42): 00 01 01 1C 00 0C 00 23 01 39 00 0B 00 23 00 00 6E C2 00 02 9F A7 00 03 00 03 08 E3 13 87 00 42 00 00 00 03 03 E8 00 69 00 01 
-----
I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 2344, s: 175, f: 25, n: 2144 | p: 2
I: (#1) TX 27 CH61 | 15  *  *  *  * 81 92 06 36 80 0B 00 65 C0 A0 5C 00 00 00 00 00 00 00 00 C4 7E DA 
I: (#1) request timeout: 500ms
-----
I: com loop duration: 1673ms
-----
I: (#0) Radio infos: -6 -6 -6 4 -6 | t: 2345, s: 264, f: 21, n: 2060 | p: 255
I: (#0) TX 27 CH61 | 15  *  *  *  * 81 92 06 36 80 0B 00 65 C0 A0 73 00 00 00 00 00 00 00 00 F5 A6 53 
I: (#0) request timeout: 10572ms
-----
I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 2345, s: 175, f: 25, n: 2145 | p: 2
I: (#1) TX 27 CH75 | 15  *  *  *  * 81 92 06 36 80 0B 00 65 C0 A0 81 00 00 00 00 00 00 00 00 91 BB 97 
I: MQTT disconnected, reason: TCP disconnect
I: (#1) request timeout: 5443ms
-----
I: MQTT disconnected, reason: TCP disconnect
I: (#0) Radio infos: -6 -6 -6 3 -6 | t: 2346, s: 264, f: 21, n: 2061 | p: 255
I: (#0) TX 27 CH61 | 15  *  *  *  * 81 92 06 36 80 0B 00 65 C0 A0 87 00 00 00 00 00 00 00 00 31 90 55 
I: (#0) RX  40ms | 27 CH03 | 95  *  *  *  *  *  *  *  * 01 00 01 01 1D 00 0D 00 24 01 3A 00 0B 00 24 00 00 B4 
I: MQTT disconnected, reason: TCP disconnect
W: (#0) frame 2 missing: request retransmit (3 attempts left)
I: (#0) TX 11 CH61 | 15  *  *  *  * 81 92 06 36 82 7F 
```

Heute hingegen lief ahoy durch. Bis 18:21 Uhr, dann gab es einen Neustart:

```shell
I: com loop duration: 1522ms
-----

2024-02-06 18:21:32	I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 2227, s: 569, f: 232, n: 1426 | p: 255
I: (#0) TX 27 CH40 | 15  *  *  *  * 81 92 06 36 80 05 00 65 C2 6A 9A 00 00 00 00 00 00 00 00 0B 24 00 

2024-02-06 18:21:33	I: (#0) request timeout: 500ms
-----

2024-02-06 18:21:33	I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 2227, s: 1094, f: 274, n: 859 | p: 2
I: (#1) TX 27 CH61 | 15  *  *  *  * 81 92 06 36 80 05 00 65 C2 6A 9B 00 00 00 00 00 00 00 00 9B 29 D3 

2024-02-06 18:21:34	I: (#1) request timeout: 504ms
-----
I: com loop duration: 1521ms
-----

2024-02-06 18:21:47	I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 2228, s: 569, f: 232, n: 1427 | p: 255
I: (#0) TX 27 CH61 | 15  *  *  *  * 81 92 06 36 80 05 00 65 C2 6A A9 00 00 00 00 00 00 00 00 FA 64 82 

2024-02-06 18:21:48	I: (#0) request timeout: 500ms
-----

[... 110 lines cut ...]
2024-02-06 18:22:15	0 5C 85 E0 

2024-02-06 18:22:15	I: (#1) request timeout: 500ms
-----
I: com loop duration: 1511ms
-----

2024-02-06 18:22:25	I: (#0) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 255
I: (#0) TX 27 CH40 | 15  *  *  *  * 81 92 06 36 80 01 00 65 C2 6A D1 00 00 00 00 00 00 00 00 3C 63 3F 

2024-02-06 18:22:25	I: (#0) request timeout: 500ms
-----

2024-02-06 18:22:26	I: (#1) Radio infos: -6 -6 -6 -6 -6 | t: 1, s: 0, f: 0, n: 1 | p: 2
I: (#1) TX 27 CH40 | 15  *  *  *  * 81 92 06 36 80 01 00 65 C2 6A D2 00 00 00 00 00 00 00 0
2024-02-06 18:22:26	0 CC 77 97 

2024-02-06 18:22:26	I: (#1) request timeout: 500ms
-----
I: com loop duration: 1510ms
```

free_heap und heap_frag zeigten keine Auffälligkeiten.

Ist es weiterhin sinnvoll im AP-Modus zu testen?

CyberLine, 2024-02-24:
Moin, ich reihe mich mal ein in die geplagten von dem Thema, auch hier Probleme seit dem Update auf 0.8.36 mit Unifi Netzwerk. Sobald der WR anfängt zu produzieren scheint der Ahoy kurz danach Offline zu gehen. Gerät ist noch Online in der UDM, zeigt aber keinen Traffic mehr an und reagiert auch nicht mehr auf ping etc.

theunknownstarman, 2024-02-24:
@CyberLine Versuchs mal mit Einstellungen oben. So wie ich das einschätze, dann liegt es bei Dir an den Einstellungen der WLAN Parameter in der UDM

CyberLine, 2024-02-25:
> @CyberLine Versuchs mal mit meinen Einstellungen oben. So wie ich das einschätze, dann liegt es bei Dir an den Einstellungen der WLAN Parameter in der UDM

Nein, hat nix gebracht. Hätte mich aber auch gewundert. Die WLAN settings hab ich seit 5 Jahren nicht geändert und bis zum Firmware Update lief der ja auch über ein Jahr problemlos.

theunknownstarman, 2024-02-25:
@CyberLine geb Dir Recht mit deiner Überlegung, allerdings kamen in den letzten 5 Jahren haufenweise Updates sowohl für UDM APs/Switches als auch für die UDM selbst und da hat sich sehr sehr viel geändert zwischenzeitlich.
Wenn Du "meine" Settings oben testest, dann solltest Du ein paar Tage warten, denn - zumindest bei mir - wird jede Tag ein Remapping der APs in Verbindung mit den WLAN Clients durchgeführt. Das hat immensen Einfluß auf die Empfangsparameter dieser.

CyberLine, 2024-02-25:
> @CyberLine geb Dir Recht mit deiner Überlegung, allerdings kamen in den letzten 5 Jahren haufenweise Updates sowohl für UDM APs/Switches als auch für die UDM selbst und da hat sich sehr sehr viel [...]

Naja, gestern umgestellt, dann ahoy restarten und heute nach ner Stunde Produktion ist er wieder weg. Ich denke ich werde einfach ein downgrade auf 0.6 machen und nie wieder updaten

theunknownstarman, 2024-02-25:
Fairerweise als Ergänzung noch der Hinweis:

Ich habe zwei separate Ahoy DTUs bei mir laufen:
- einen "alten" 8266 und 
- einen ESP32

Der ESP läuft DEUTLICH stabiler als der 8266.

Das wäre evtl. auch nochmal eine Idee.

n14ro, 2024-02-25:
Guten morgen zusammen, 

ich bin ebenfalls noch fleißig am Testen. Ich hatte 3 DTUs mit 8266 zur gleichen Zeit aufgebaut:

- Schwiegereltern: 
   - 8266 Version 0.6.9
   - NRF24 Plus
   - Fritzbox (Entfernung 5m)
   - Nutzung: fast tägliches Aufrufen der Webseite
   - bisher keine Probleme berichtet
- Eltern
   - 8266 Version 0.8.83
   - NRF24 Plus
   - Fritzbox (Entfernung 0,5m)
   - Nutzung: gelegentliches Aufrufen der Webseite
   - Probleme seit Beginn (0.6.9) bis mindestens 0.8.36
- Bei mir 
   - 8266 Version 0.8.83
   - NRF24 Plus
   - ubiquity (Entfernung 8m)
   - Nutzung: MQTT + seltenes Aufrufen der Webseite
   - Probleme seit Beginn (0.6.9) (siehe Threadstart)

Die hier im Thread genannten Einstellung am Rekeying und BSS Transition nacheinander deaktiviert. Rekeying hat das fast stündliche "Abreißen" korrigiert. Trotzdem bleiben die auch von @CyberLine genannten "Abrisse". BSS hat bei mir keine merklichen Veränderungen gebracht.

Ich habe am Mittwoch ebenfalls ein ESP32 geliefert bekommen. NRF24 Plus angelötet (vom 8266 abgelötet) und läuft nun seit 3 Tagen ohne Probleme. Bisher nur ein gewollter Neustart für angepasste MQTT-Einstellungen.

@theunknownstarman Vielleicht ist das das Problem? Kommt der 8266 ggf. mit z.B. Kanalwechsel nicht zuverlässig klar? Nach einem "Abriss" scheint die Ahoy-Applikation ja normal weiterzulaufen, nur halt ohne TCP-Verbindung nach außen. Es scheint auch kein neuer Verbindungsaufbau versucht zu werden. So zumindest mein laienhafter Rückschluss aus Logs und erlebtes Verhalten des 8266.

Grüße und noch ein schönen Sonntag.

theunknownstarman, 2024-02-25:
Ja. Du liegst mit hoher Wahrscheinlichkeit richtig, dass der Kanalwechsel (und vermutlich auch ein AP Wechsel) ein Problem für den 8266 ist.

CyberLine, 2024-02-25:
Vielleicht noch zu erwähnen: meine Ahoy hängt per AP Pinning fest an einem AP, wie schon immer und auch die Kanäle sind fix vergeben, da ich mir sonst mit dem Zigbee in die quere komme. Ich hab insgesamt knapp 10 8266 im Einsatz für Klima, Gaszähler etc. und keiner hat bisher Probleme. Nur die Ahoy seit dem Update.

CyberLine, 2024-02-26:
Ich hab jetzt auch mal free_heap und heap_frag mitgeschnitten bis zum ausfall vor einer Stunde (scheint immer relativ zuverlässig um 9:15 zu passieren.) Zuletzt sagte er auch mqtt_connected: false und dann war auch web weg.
[image]
[image]

CyberLine, 2024-02-26:
Hab jetzt mal auf nen ESP32 gewechselt und hoffe das Thema hat sich damit erledigt.

lumapu, 2024-02-26:
davon gehe ich aus, der esp32 ist viel zuverlässiger

n14ro, 2024-03-01:
Kurze Rückmeldung: Mit dem ESP32 hatte ich noch keinen ungeplanten Neustart. Auch die Daten via MQTT sehen in Home Assistant gut aus.
[image]

Ich werde wohl den zweiten ESP8266 auch noch auf den ESP32 umbauen.

CyberLine, 2024-03-02:
> Kurze Rückmeldung: Mit dem ESP32 hatte ich noch keinen ungeplanten Neustart. Auch die Daten via MQTT sehen in Home Assistant gut aus.
> [image]
> 
> Ich werde wohl den zweiten ESP8266 auch noch auf den ESP32 umbauen.

Gleiches ist bei mir zu beobachten: seit dem Wechsel auf ESP32 keine issues mehr.

theunknownstarman, 2024-03-02:
Wie habt ihr es geschafft die MQTT Werte beim Gerätewechsel beizubehalten?
Reicht die gleiche Benamsung des Gerätes aus um die gleichen MQTT Werte zu behalten?

lumapu, 2024-03-10:
nach meinem Verständnis schon, aber es gibt da auch noch die Client-ID. Die muss mWn. auch übereinstimmen.

bub4, 2024-03-16:
Ich bin am Wochenende auch auf UniFi umgestiegen. Die Migration von Fritz verlief soweit gut, bis auf die Ahoy-DTU. Die wurde nach dem Wifi und IP-Umzug zwar im UniFi Netwerk angezeigt, sie war aber nicht mehr erreichbar. Ich habe daraufhin dann die neueste Version installiert. Nach umverkabeln der Antenne hat es letztendlich wieder alles funktioniert. Sie stieg aber regelmäßig aus und die GUI war nicht mehr erreichbar.

Ich habe daraufhin alle Versionen ausprobiert und nichts half - sporadische Ausfälle worauf sie aus- und wieder eingesteckt werden musste.

Dann hab ich an den Netzwerkeinstellungen rumgespielt und "IGMP Snooping" deaktiviert.
...keine Ahnung ob es damit zusammenhängt, aber sie läuft nun seit über zwei Tagen durch mit Version 0.6.9

Meine DTU basiert auch auf dem Wemos D1 Mini - also kein ESP32. 
Einen Kondensator habe ich nicht eingelötet.

n14ro, 2024-03-16:
Gerade nachgeschaut: IGMP snooping ist bei mir noch nie aktiv gewesen.

theunknownstarman, 2024-03-16:
Bei mir auch so. 
Vermutlich ist die Multicast Integration im 8266 nicht korrekt entwickelt. Ob es beim ESP32 besser ist, kann ich nicht sagen.

n14ro, 2024-03-16:
Der ESP32 performt bei mir im gleichen ubiquity-Netzwerk an der gleichen Stelle und den gleichen Einstellungen unvergleichlich besser. Ab und zu ist zwar die GUI nicht erreichbar oder sehr träge. Fängt sich dann wenig später wieder. MQTT läuft dabei immer stabil durch. 

[image]

closed 2024-05-30 by lumapu