lumapu/ahoy #1395: [Bug]export settings different format

hrolofs, 2024-01-31:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.68

### Github Hash

163a9e4

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

MqTT,Display,NRF

### Error description

Ich habe zwei DTUs (ESP32/NRF, blinkyparts) und opendtufusion v2.1 mit der 0.8.68 laufen.

Die DTUs speichern die Config in unterschiedlichen Versionen. Bei der opendtufusion fehlt z.B. region und timezon in den exportierten settings file.

1. ESP32/NRF, blinkyparts
`	"version": 10,
	"wifi": {
		"ssid": "XXXXXXXX",
		"pwd": "",
		"ap_pwd": "XXXXXXXX",
		"hidd": false,
		"dev": "AhoyDTU2",
		"adm": "XXXXXXXX",
		"prot_mask": 60,
		"dark": false,
		"reb": true,
		"region": 0,
		"timezone": 1,
		"ip": "",
		"mask": "",
		"dns1": "",
		"dns2": "",
		"gtwy": ""
	},`

2. opendtufusion v2.1
`	"wifi": {
		"ssid": "XXXXXXXX",
		"pwd": "",
		"ap_pwd": "XXXXXXXX",
		"hidd": false,
		"dev": "AhoyDTU1",
		"adm": "XXXXXXXX",
		"prot_mask": 60,
		"dark": false,
		"reb": true,
		"ip": "",
		"mask": "",
		"dns1": "",
		"dns2": "",
		"gtwy": ""
	},`

lille2000, 2024-01-31:
Ich habe zwei mal exportiert:
Der erste Export hatte Version=8.  (zb ohne Region)
Danach habe ich nichts in den Einstellungen verändert und nochmal gespeichert.
Ein weiterer Export hatte die Version=10 (mit Region)

Der Export erzeugt scheinbar keinen vollständiges Daten-Set  mit der aktuellen Konfiguration sondern exportiert nur, was beim letzten Speichern gesetzt/vorhanden war. Neue Parameter die in neueren Ständen dazukommen sind werden nicht exportiert.

hrolofs, 2024-01-31:
`Ein weiterer Export hatte die Version=10 (mit Region)`
gleich mal ausprobieren. 

Jetzt wird auch bei der opendtufusion ein Version 10 exportiert.

lumapu, 2024-01-31:
würde das aus eurer Sicht heißen, man muss die Settings erst auf den neusten Stand bringen, damit sie auch identisch sind? Die fehlenden Felder werden automatisch ausgefüllt, wenn man ein älteres Backup einspielt, aus meiner Sicht ist diese inkosistenz hier nicht tragisch, da es keine Funktoionsänderung / -einschränkung gibt.

hrolofs, 2024-01-31:
`da es keine Funktoionsänderung / -einschränkung gibt`
Wenn die Funktion Settings schreiben/laden in sich konsistent ist besteht auch kein Problem.
Dachte halt, gleiche Firmware Version hat gleichen Output.

War halt etwas verwirrt.

Neue DTU upgedatet, Settings in der Weboberfläche von DTU1 auf DTU2 übernommen, Settings abgespeichert. 

Wollte dann im Texteditor die Settings der 1. und 2. DTU vergleichen und schauen ob ich was vergessen habe.

Wenn die Settings nicht im lesbaren json abgespeichert wären, wäre ich da nie drüber gestolpert.

Also: kein Bug :-)

lille2000, 2024-02-01:
Ich hatte nur eine andere Erwartung:
Ich dachte, ein ExportFile mit dem SW Stand im Dateinamen enthält alle für DIESE SW möglichen Einstellungsmöglichkeiten.

Es ist völlig ok, ein altes Setting beim Start aus dem Speicher zu laden und mit neuen Einstellmöglichkeit zu ergänzen. Das ist gut und sinnvoll beim Hoch- und runter-Update.

Nur beim Export hätte ich nicht den Export der gespeicherten Daten erwartet sondern das vollständige Set.

lumapu, 2024-02-01:
ja man könnte ja vor dem Export noch ein "save" ausführen, was dazu führt, dass man ein vollständiges Set bekommt.

closed 2024-02-17 by lumapu

Dirk-Ahoy, 2024-02-25:
Okay - dann Feature :)   - hab ich gerade hier nachgestellt.