lumapu/ahoy #1396: ESP32C3 SuperMini

UserRS232, 2024-02-01:
## Hardware
  * [ ] ESP8266
  * [X] ESP32  Version C3 Super Mini von Ali  https://de.aliexpress.com/item/1005006225621279.html
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [X ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [X ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [X ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ X] ESP Tools

### Debugging:
* [ X] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Hi, ich bekomm die AhoyDTU einfach nicht auf dem ESP32C3 Supermini zum laufen. Was mache ich falsch? Webinstaller von AHOY funktioniert gar nicht. Es kommt erst nur die Meldung "Preparing installation" für ca. 60sek dann "Failed to initialize." Egal ob mit Download Mode oder nicht. Chip an sich funktioniert wunderbar, wenn ich z.B. ESPHome flashe. Hardware defekt kann ich also ausschließen. 

Flash ich mit ESP Tools die passende ESP32C3 File, sehe ich im Log (siehe unten) das AHOYDTU läuft, aber er macht keinen AP auf somit lassen sich keine WLAN Daten einpflegen. Geflasht habe ich die latest bin auf 0x10000, passt das so?

[image]

Hoffe mir kann da jemand weiter helfen. Wäre klasse wenn dieser sehr kleine und kostengünstige Chip funktioniert. 

USB LOG
---------
AhoyDTU Info:
Version: 0.8.36
Github Hash: 6fb0535

---------
AP MODE
SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

Wifi event: 0
Wifi event: 2
Wifi event: 10
W: WARNING! your NRF24 module can't be reached, check the wiring
scanning APs with SSID YOUR_WIFI_SSID
Wifi event: 1
scanning APs with SSID YOUR_WIFI_SSID
Wifi event: 1
scanning APs with SSID YOUR_WIFI_SSID

Peertje-62, 2024-02-01:
Hi,
Erase first , next see picture
kind regards
[image]

Peertje-62, 2024-02-01:
https://github.com/lumapu/ahoy/blob/main/Getting_Started.md

download the flash-tool [nodemcu-pyflasher](https://github.com/marcelstoer/nodemcu-pyflasher)

UserRS232, 2024-02-02:
> Hi, Erase first , next see picture kind regards ![IMG_0071](https://private-user-images.githubusercontent.com/144957788/301600264-4f37cf22-23f3-44af-988d-f6524bfe8a71.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY4Njc2MTYsIm5iZiI6MTcwNjg2NzMxNiwicGF0aCI6Ii8xNDQ5NTc3ODgvMzAxNjAwMjY0LTRmMzdjZjIyLTIzZjMtNDRhZi05ODhkLWY2NTI0YmZlOGE3MS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwMlQwOTQ4MzZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wMDRlMGQ5YTI2OGNjZDljYjM2ZDk0ZGM4MWQwZTViMzQ4ODMyZTIzZjI2NWFmMmI0NDJjYWM1MGQ2MTQ0Y2ExJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.RCroADMW88CbPASfQ_ewiF6a2d7ihVt74KQ1y8eJ9Us) is for esp 32 I think

Thank you for your quick response, unfortunately it still doesn't work.
I have tried...
- Ahoy Webinstaller ->  "Failed to initialize." 
- ESP Flash Download 3.9.5 only firmware.bin -> Boot loop
- ESP Flash Download 3.9.5 all 4 .bin Files (latest) with Offset from Peertje-62 -> Boot loop
- NodeMCU PyFlasher 5.0.0 -> "Detecting Chip type..."   does not flash

I tried to rule out a hardware defect
- Tested with ESPHome Webinstaller -> Works perfect...
- Tested another Board (same type) -> same behave...

are there different ESP32C3 chips apart the memory?

Peertje-62, 2024-02-02:
You could try Esp-flasher (based on PyFlasher,
I don't know anymore where I've got it

Peertje-62, 2024-02-02:
https://github.com/esphome/esphome-flasher/releases/tag/1.4.0
ESPHome-Flasher-1.4.0-Windows-x64.exe

tqma1, 2024-02-04:
I think the adresss for the bootloader.bin is wrong
[image]

UserRS232, 2024-02-04:
tryed ESPHome-Flasher-1.4.0-Windows-x64.exe  -> Bootloop "invalid header"
tryed with the adress tqma1 mentioned -> Worked!!!  (Thank you!)

Now Ahoy DTU is running at least on the ESP32C3 but the log output shows a strange message and it doesn't open a WiFi AP either.  I tried the 0.8.36 release and the 0.8.71 dev with different chips. Same error every time... no wlan AP...

LOG output:
```
[12:36:21]E (3767) esp_littlefs: ./components/esp_littlefs/src/littlefs/lfs.c:1347:error: Corrupted dir pair at {0x0, 0x1}
[12:36:21]
[12:36:21]E (3768) esp_littlefs: mount failed,  (-84)
[12:36:21]E (3770) esp_littlefs: Failed to initialize LittleFS
[12:36:21][  2028][E][LittleFS.cpp:95] begin(): Mounting LittleFS failed! Error: -1
[12:36:21]I: .. format ..
[12:36:21][  2035][E][esp32-hal-misc.c:128] disableCore0WDT(): Failed to remove Core 0 IDLE task from WDT
[12:36:21]I: .. success
[12:36:21][  2134][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
[12:36:21]W: failed to load json, using default config
[12:36:21]I: Settings valid: false
[12:36:21]W: WARNING! your NRF24 module can't be reached, check the wiring
[12:36:21]
[12:36:21]---------
[12:36:21]AhoyDTU Info:
[12:36:21]Version: 0.8.71
[12:36:21]Github Hash: 883aefb
[12:36:21]
[12:36:21]---------
[12:36:21]AP MODE
[12:36:21]SSID: AHOY-DTU
[12:36:21]PWD: esp_8266
[12:36:21]IP Address: http://192.168.4.1
[12:36:21]---------
[12:36:21]
[12:36:21]Wifi event: 0
[12:36:21]Wifi event: 2
[12:36:21]Wifi event: 10
[12:36:22]Wifi event: 11
[12:36:22]Wifi event: 10
[12:36:23]W: WARNING! your NRF24 module can't be reached, check the wiring
[12:36:23]scanning APs with SSID YOUR_WIFI_SSID
[12:36:30]Wifi event: 1
[12:36:31]scanning APs with SSID YOUR_WIFI_SSID
[12:36:38]Wifi event: 1
[12:36:39]scanning APs with SSID YOUR_WIFI_SSID
[12:36:46]Wifi event: 1
[12:36:47]scanning APs with SSID YOUR_WIFI_SSID
[12:36:54]Wifi event: 1
[12:36:55]scanning APs with SSID YOUR_WIFI_SSID
```

tqma1, 2024-02-04:
Try it with this online tool
[https://espressif.github.io/esptool-js/](url)

Peertje-62, 2024-02-04:
Hi, tqma1 & userRS232
Got she screenshot from discord,
tried it my self.....
You are right and I am sorry
Regards

UserRS232, 2024-02-06:
> Try it with this online tool [https://espressif.github.io/esptool-js/](url) settings: Baudrate: 460800 adresses like above but beginning with „0x“ example: 0xe000

Try with the mentioned tool
bootloader.bin at 0x0000
partitions.bin at 0x8000
ota.bin at 0xe000
ahyostu_esp32-c3mini.bin at 0x10000

Connected to device: ESP32-C3 (revision 4)
```
**Warning: Image file at 0x0 doesn't look like an image file, so not changing any flash settings.** ???
Compressed 13200 bytes to 9522...
Writing at 0x0... (100%)
Wrote 13200 bytes (9522 compressed) at 0x0 in 0.256 seconds.
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x8000... (100%)
Wrote 3072 bytes (146 compressed) at 0x8000 in 0.061 seconds.
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0xe000... (100%)
Wrote 8192 bytes (47 compressed) at 0xe000 in 0.098 seconds.
Hash of data verified.
Compressed 1298112 bytes to 775585...
Writing at 0x10000... (2%)
.......
Writing at 0x14ab5a... (100%)
Wrote 1298112 bytes (775585 compressed) at 0x10000 in 10.723 seconds.
Hash of data verified.
```

#################################################################
log after flashed:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038bfd0
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5810,len:0x438
load:0x403cc710,len:0x90c
load:0x403ce710,len:0x25f4
entry 0x403cc710
I: Initializing FS ..
I:  .. done
**[   225][E][vfs_api.cpp:105] open(): /littlefs/settings.json does not exist, no permits for creation
W: failed to load json, using default config
I: Settings valid: false**
W: WARNING! your NRF24 module can't be reached, check the wiring

---------
AhoyDTU Info:
Version: 0.8.71
Github Hash: 883aefb

---------
AP MODE
SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

Wifi event: 0
Wifi event: 2
Wifi event: 10
W: WARNING! your NRF24 module can't be reached, check the wiring
scanning APs with SSID YOUR_WIFI_SSID
Wifi event: 1
scanning APs with SSID YOUR_WIFI_SSID
Wifi event: 1
scanning APs with SSID YOUR_WIFI_SSID

```

Unfortunately unsuccessful --> will not open an Wlan AP
what about the highlighted areas? Could that be the error?

tqma1, 2024-02-06:
Wlan AP is opened.

---------
AP MODE
SSID: AHOY-DTU
PWD: esp_8266
IP Address: http://192.168.4.1
---------

Peertje-62, 2024-02-09:
Hi, 
problem solved?
regards

Peertje-62, 2024-02-13:
If so,
you should close this item
kind regards

UserRS232, 2024-02-13:
Yesterday 6 more chips arrived.
Of the two I already had, none of them work. Both can be flashed with the last tool, they also show in the console that the AP must be open, but no AP is open.
Tried further with the 6 more ESP32-C3 Super mini from yesterday's delivery.
Result: 2 work! 4 not.
In the 2 that work, the AP is visible 20-30 seconds after boot and AHOY can be configured. The others don't work.
BUT! all chips work wonderfully with ESP-Home. I will now continue testing one of the 2 that work. But so far everything looks good so far. I hope I find the error why the others don't work.

Closed

closed 2024-02-13 by UserRS232

Peertje-62, 2024-02-13:
Thx for the update.
I did buy my first  esp8266, € 2,90
(Now I am using  ESP 32, more powerful
https://www.komputer.de/zen/index.php?main_page=product_info&cPath=38_39&products_id=481&zenid=4f75gusr54p564j2frgscdk3f0
Regards

grericht, 2024-02-20:
had the same issue. every time i had to start the board by reseting in the console.
so i could reach ap and now it is reachable by wlan. but everytime i had to reset it in the console.

grericht, 2024-03-04:
Does no one have a tip for this board? It is this one: https://forum.arduino.cc/t/esp32-c3-supermini-pinout/1189850

I can flash it with ESPHome without any problems. From Home Assistant and from the WebInstaller.
When I try to flash AhoyDTU I can access the console. When I click on install, only the message "Preparing installation" appears for about 60 seconds, then "Failed to initialize."

With ESP Home I don't have to press anything. it simply starts the installation. Pressing the buttons doesn't help here either.

When I got AhoyDTU installed (I think via ota from esp home web server) the board doesn't start properly. It only starts when I connect it to the laptop via USB and press console in the web installer. Sometimes I have to press reset in the console