lumapu/ahoy #1397: [Bug] Inverter lassen sich nicht mehr per DTU über API/MQTT/Live wieder einschalten oder neu starten

blueline13, 2024-02-02:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.70

### Github Hash

*

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

*

### Error description

1) Abschalten des Inverter per API/MQTT/Live
2) Inverter kann nun nicht mehr per DTU neu gestartet oder eingeschaltet werden. Bei der Live Ansicht kommt der Hinweis das der Inverter keine Kommandos mehr annimt. Auch bei MQTT und API reagiert er nicht mehr.
3) Zurück auf die 0.8.68, schon läuft es wieder.

Kann es sein das es eine Folge von dem hier ist : "prevent sending commands to inverter which isn't active #1387" was in der 0.8.70 eingebaut wurde. Das würde ja bedeuten, dass alle Inverter die mit Batterie betrieben werden keine Chance mehr haben den Inverter Abends einzuschalten.

lumapu, 2024-02-02:
ok, da habe ich wohl einen Denkfehler drin, da auf den Status produzierend geprüft wird, was ja im abgeschalteten Zustand nie zutreffen kann.
Der Status muss auf available geprüft werden, dann sollte es mAn. wieder funktionieren. Fix kommt bald 😉

lumapu, 2024-02-04:
bitte mit `0.8.71` nochmal testen

blueline13, 2024-02-04:
Ich habe das eben probiert mit der 0.8.71. Leider keine Änderung. Wenn ich den Inverter per LIVE ausschalte, dann versuche ihn wieder einzuschalten, kommt die Meldung "Fehler: Leistungsbegrenzung / Ansteuerung aktuell nicht möglich" Also keine Veränderung zur 0.8.70. Auch API und MQTT gehen dann nicht um ihn wieder einzuschalten.

lumapu, 2024-02-04:
Fehler kann ich jetzt nachvollziehen, (auch weil es gerade Tag ist 😇). Behebung kommt in der nächsten Version.

blueline13, 2024-02-04:
Ja die 0.8.72 läuft wieder .... DANKE !!!!

closed 2024-02-04 by blueline13