lumapu/ahoy #1398: Tageserträge und Gesamtertrag des Vortags werden nicht gelöscht trotz Einstellung

elisenjens, 2024-02-03:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

circuit board

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.7.40

### Github Hash

91e816e

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Defaults

### Error description

hab einen Haken bei "Werte und Gesamtertrag Mitternacht zurücksetzen" drin; nix geändert an den Einstellungen und vorher hats gefunzt; allerdings hatte ich auch schon manchmal devs, bei denen es nicht ging...
ich fahr nachts die Kommunikation runter; aber das mache ich schon immer; merkt Ahoy in dieser Version nicht, dass nachts abgeschaltet ist?

lumapu, 2024-02-03:
an der Stelle wurde nichts geändert seit vielen Versionen, wäre komisch, wenn es nicht funktioniert. Zur Analyse würde es helfen, wenn zwei Browsertabs geöffnet werden. Im ersten die Webserial-Konsole öffnen, im zweiten auf `[IP]/debug` gehen. Dann wieder in den ersten wechseln und die Liste der Timer analysieren, es müsste einer mit `midNi` dabei sein. Die große Zahl davor ist der Timestamp und kann hier (https://www.unixtimestamp.com/) in eine lesbare Zeit umgerechnet werden.

elisenjens, 2024-02-04:
Danke für die Anleitung; da kommt heute Mittag 

 

1707087600 = Mon Feb 05 2024 00:00:00 GMT+0100 raus.

 

Versteh aber nicht, was das im Zusammenhang mit dem Problem bedeutet

elisenjens, 2024-02-05:
Auch heute sind die Werte mit der 0.8.71 vom Vortag

elisenjens, 2024-02-05:
[image]
[image]
[image]

10:31:36.227 I: (#0) RX  30ms | 27 CH61 | 95 01
10:31:36.229 I: (#0) RX  79ms | 27 CH61 | 95 02
10:31:36.229 I: (#0) RX 129ms | 27 CH40 | 95 03
10:31:36.230 I: (#0) RX 179ms | 27 CH40 | 95 84
10:31:36.231 I: (#0) Payload (62)
10:31:36.233 -----
10:31:36.233 I: com loop duration: 234ms
10:31:36.234 -----
10:31:40.762 I: webSc, tmt: 1, rel: 1
10:31:40.763 I: prot, tmt: 1, rel: 1
10:31:40.763 I: uart, tmt: 1, rel: 5
10:31:40.764 I: hist, tmt: 1, rel: 1
10:31:40.764 I: tSend, tmt: 11, rel: 15
10:31:40.764 I: midNi, tmt: 1707174000, rel: 0
10:31:40.766 I: wifiL, tmt: 1, rel: 1
10:31:40.768 I: midRe, tmt: 1707174010, rel: 0
10:31:40.771 I: ivCom, tmt: 1707149315, rel: 0
10:31:40.774 I: Sunri, tmt: 1707149375, rel: 0
10:31:40.777 I: ntp, tmt: 34123, rel: 0
10:31:51.224 I: (#0) RX  81ms | 27 CH61 | 95 02
10:31:51.226 I: (#0) RX 129ms | 27 CH61 | 95 03
10:31:51.226 I: (#0) RX 176ms | 27 CH40 | 95 84
10:31:51.308 I: (#0) RX  35ms | 27 CH61 | 95 01
10:31:51.309 I: (#0) Payload (62)
10:31:51.311 -----
10:31:51.312 I: com loop duration: 312ms
10:31:51.312 -----

elisenjens, 2024-02-05:
1707174000 = Der 06.02.2024 um 00:00:00 Uhr

elisenjens, 2024-02-05:
> Kleine Frage: Hängt dein Inverter an einem Akku?
>
> BTW: Zeig' bitte mal deine Invertereinstellung mit Nachtkommunikation, hatte hier eher das umgekehrte Problem.

Kein Akku; Nachtkomm ist abgeschaltet

fsck-block, 2024-02-07:
Hallo @elisenjens 

Ich habe das heute Nacht nochmals mit der 0.8.42 und 0.8.74 geprüft und konnte das beschriebene Verhalten nicht nachvollziehen. Heute Morgen war YieldDay und Max AC Power zurückgesetzt.
[image]

Meine Einstellungen:
Inverter Communication:
[image]

Lat/Long: Bei mir allerdings mit `.` Punkt und nicht mit `,` Komma angegeben. Nutze aber auch nicht die DE Variante
[image]

Flags:
[image]

Kannst ja mal die EN-Variante probieren. Vielleicht liegts an den Koordinaten?

lumapu, 2024-02-07:
midnight wird aus der Zeitzone berechnet - da haben die Koordinaten keinen Einfluss. Der Trigger ist ja auch richtig gesetzt wie weiter oben schon festgestellt wurde.

closed 2024-02-08 by elisenjens

fsck-block, 2024-02-08:
@lumapu 
Es ging mir weniger um midnight sondern um die Bedingung dass `disable night communication` Koordinaten braucht.

@elisenjens 
Und so wie ich deine Screenshot lese, sind deine Koordinaten mit ',' Komma angegeben. Bei mir, auch in der DE Version, werden Koordinaten mit  Dezimalpunkt dargestellt.
Und falls die Koordinaten intern nicht in float gewandelt werden, dann gibt's auch kein `disable night communication` und damit auch kein Reset von YieldDay und MaxPower.

Probier doch mal bitte deine Koordinaten mit Dezimalpunkt einzutragen.

lumapu, 2024-02-08:
ok, dann habe ich dich falsch verstanden, deine Aussage stimmt natürlich.
Die Koordinaten werden intern als float gespeichert, da das HTML Feld den Typ number hat, wird sich der Dezimaltrenner der Browsersprache anpassen