lumapu/ahoy #1399: kann ESP32 nicht flashen

nom56, 2024-02-03:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

Peertje-62, 2024-02-03:
Hi there, is not much info to go on!
try : 
https://github.com/esphome/esphome-flasher/releases/tag/1.4.0
ore ;
https://3printd-solution.eu/Ahoy-openDTU-Fusion-fuer-HM-HMS-HMT-mit-Display
regards

nom56, 2024-02-13:
HM-1500 ist noch nicht verfügbar

closed 2024-02-13 by nom56