lumapu/ahoy #1400: [Bug] Redirects to webpage /live with dev 0.8.70

dknubben, 2024-02-03:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.70

### Github Hash

unknown, self compiled

### Build & Flash Method

ESP Tools (flash)

### Setup

nothing important for this issue

### Error description

After update to 0.8.70 most links in navigation ex. /system?v=0.8.70 are redirected to the main page /live?v=0.8.70.
Exept
/history?v=0.8.70
/about?v=0.8.70
and
/api
Browser caches were cleared.
Back to 0.8.69 - all fine.
I am not sure whats going wrong.
Some changes in the HTML (page refresh calls)? Disabling JS still redirects the page.

Peertje-62, 2024-02-03:
I had the same issue
Ahoy V0.7.80, Fusion 2.1
Done a hard reboot since it's running fine
regards

lumapu, 2024-02-03:
I tried to reproduce on ESP32 - but without success
1. I installed 0.8.69
2. save settings, reboot
3. OTA Update to 0.8.70
4. everything is fine and accessible

Peertje-62, 2024-02-03:
Okay,
if it helps?
It happend after about 14.45 hours running time, reboot at midnight!
regards

Joh49, 2024-02-03:
hatte auch diesen Effekt bei einem 8266 Wemos mini unter Firefox.
Mehrere Seiten und auch die Updateseite brachten nur die Maske "Daten".
Nach Wechsel auf den Chrome Browser kam die Updateseite wieder richtig.
Machte einen Downgrade auf 8.55 und weg war der Spuk, hab nicht weiter analysiert woran es lag.
Gruß

dknubben, 2024-02-03:
Ich konnte das mit FireFox, Chrome und Vivaldi feststellen. Zwischendurch habe ich auch einen Flash mir Erase gemacht, Fehler blieb. Erst der Rücksprung auf 0.8.69 beseitigte das Problem.

lumapu, 2024-02-03:
es wurde eine Änderung mit 0.8.70 diesbezüglich gemacht, also ganz falsch liegt ihr nicht - die Laufzeit ist noch ein Indiz, dass ich untersuchen muss, danke für den Hinweis. Normal sperrt sich Ahoy, wenn 20 Minuten kein Webzugriff war und ein Passwort gesetzt ist - genau hier vermute ich aktuell den Fehler (wenn kein Passwort gesetzt ist, sich Ahoy aber trotzdem sperrt).

lumapu, 2024-02-03:
ich glaube ich habe es gefunden:
https://github.com/lumapu/ahoy/blob/91e816e9413f5c3750d2ae0dbfea6f425e57d45f/src/web/Protection.h#L38-L46

in Zeile 42 müsste es `!=` heißen

dknubben, 2024-02-04:
Vielen Dank, das scheint das Problem tatsächlich gelöst zu haben.

closed 2024-02-04 by dknubben