lumapu/ahoy #1401: [Bug] Guru Meditation nach "plausibility check failed"

You69Man, 2024-02-04:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

circuit board

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.71

### Github Hash

883aefbe643ce04c6fe45882479f560ecd33db16

### Build & Flash Method

VSCode - Platform IO (build & flash)

### Setup

-

### Debug Serial Log output

```shell
I: (#0) Radio infos: -4 -4 -5 4 -4 | t: 39, s: 28, f: 1, n: 10 | p: 1
I: (#0) TX 27 CH61, 9 ret., rx offset: 3 | 15 0B 80
I: (#0) RX  30ms | 27 CH03 | 95 01
I: (#0) RX 148ms | 23 CH75 | 95 83
W: (#0) frame 2 missing: request retransmit (3 attempts left)
I: (#0) TX 11 CH61, 9 ret., rx offset: 3 | 15 CF 82
W: (#0) frame 2 missing: request retransmit (2 attempts left)
I: (#0) TX 11 CH61, 9 ret., rx offset: 3 | 15 CF 82
I: (#0) RX  25ms | 17 CH03 | 95 81
I: (#0) Payload (4)
E: plausibility check failed, expected 42 bytes
-----
-----
I: (#Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d8932  PS      : 0x00060e30  A0      : 0x800dcffe  A1      : 0x3ffb2110
A2      : 0xe69cbe73  A3      : 0x00000000  A4      : 0x3ffca664  A5      : 0x00060c23
A6      : 0x00060c23  A7      : 0x00000001  A8      : 0x800d8932  A9      : 0x3ffb20f0
A10     : 0x3ffb211c  A11     : 0xe69cbe73  A12     : 0x3ffca664  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x4008ab1d  LEND    : 0x4008ab2e  LCOUNT  : 0xfffffffe

Backtrace: 0x400d892f:0x3ffb2110 0x400dcffb:0x3ffb2150 0x400ddc59:0x3ffb21a0 0x400ddc8c:0x3ffb21e0 0x400dceea:0x3ffb2200 0x400ebe35:0x3ffb2230 0x400ef462:0x3ffb2270 0x40108be9:0x3ffb2290

  #0  0x400d892f:0x3ffb2110 in Heuristic::printStatus(Inverter<float>*) at hm/Heuristic.h:136 (discriminator 19)
  #1  0x400dcffb:0x3ffb2150 in Communication::innerLoop(CommQueue<(unsigned char)100>::queue_s const*) at hm/Communication.h:80
  #2  0x400ddc59:0x3ffb21a0 in Communication::loop()::{lambda(bool, CommQueue<(unsigned char)100>::queue_s const*)#1}::operator()(bool, CommQueue<(unsigned char)100>::queue_s const*) const at hm/Communication.h:63
      (inlined by) Communication::loop()::{lambda(bool, CommQueue<(unsigned char)100>::queue_s const*)#1}::operator()(bool, CommQueue<(unsigned char)100>::queue_s const*) const at hm/Communication.h:48
  #3  0x400ddc8c:0x3ffb21e0 in std::_Function_handler<void (bool, CommQueue<(unsigned char)100>::queue_s const*), Communication::loop()::{lambda(bool, CommQueue<(unsigned char)100>::queue_s const*)#1}>::_M_invoke(std::_Any_data const&, bool&&, CommQueue<(unsigned char)100>::queue_s const*&&) at .platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #4  0x400dceea:0x3ffb2200 in std::function<void (bool, CommQueue<(unsigned char)100>::queue_s const*)>::operator()(bool, CommQueue<(unsigned char)100>::queue_s const*) const at .platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
  #5  0x400ebe35:0x3ffb2230 in CommQueue<(unsigned char)100>::get(std::function<void (bool, CommQueue<(unsigned char)100>::queue_s const*)>) at hm/CommQueue.h:82
      (inlined by) Communication::loop() at hm/Communication.h:48
      (inlined by) app::loop() at app.cpp:153
  #6  0x400ef462:0x3ffb2270 in loop() at main.cpp:42
  #7  0x40108be9:0x3ffb2290 in loopTask(void*) at .platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
```

### Error description

Ist heute morgen kurz nach update auf 0.8.71 passiert. 
Momentan läuft's wieder.

lumapu, 2024-02-04:
danke für den ausführlichen Report, damit kann man was anfangen und hoffentlich in Zukunft auch abfangen

You69Man, 2024-02-04:
Ich habe übrigens gerade noch ein paar Fault-Injection-Tests dazu gemacht. 
Das passiert deterministisch, nach jedem closeRequest aufgrund von "plausibility check failed" oder auch "record is NULL". 
Danach ist q->iv immer ein nullpointer, was zur Exception führt.

lumapu, 2024-02-04:
kann mir nicht erklären wie `q->iv` ein nullpointer sein kann, da das doch davor in der `app.cpp` `tickSend` geprüft wird.

You69Man, 2024-02-04:
> kann mir nicht erklären wie `q->iv` ein nullpointer sein kann, da das doch davor in der `app.cpp` `tickSend` geprüft wird.

So kann ich ihn deterministisch auslösen:
https://github.com/You69Man/ahoy/tree/debug_q_iv_exception

You69Man, 2024-02-04:
Der mRdPtr überholt den mWrPtr ...

**mRdPtr=9
mWrPtr=11
cmd = 17
attempts = 5
attemptsMax = 5
ts = 0
isDevControl = 0
*** printStatus ***I: (#0) Radio infos: 0 4 0 0 0 | t: 9, s: 9, f: 0, n: 0 | p: 1
I: (#0) TX 27 CH23, 12 ret., rx offset: 3 | 15 11 80
I: (#0) RX  34ms | 27 CH75 | 95 81
I: (#0) Payload (14)
-----
mRdPtr=10
mWrPtr=11
cmd = 11
attempts = 5
attemptsMax = 5
ts = 0
isDevControl = 0
*** printStatus ***I: (#0) Radio infos: 0 4 0 0 0 | t: 10, s: 10, f: 0, n: 0 | p: 1
I: (#0) TX 27 CH23, 12 ret., rx offset: 3 | 15 0B 80
I: (#0) RX  33ms | 27 CH75 | 95 01
I: (#0) RX  81ms | 27 CH61 | 95 02
I: (#0) RX 127ms | 23 CH61 | 95 83
I: (#0) Payload (42)
E: plausibility check failed, expected 42 bytes
-----
-----
mRdPtr=12
mWrPtr=11
cmd = 0
attempts = 0
attemptsMax = 0
ts = 0
isDevControl = 0
*** printStatus ***Null-Pointer!!!Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.**

You69Man, 2024-02-04:
in Communication.h wird closeRequest mehrfach hintereinander aufgerufen:

ab Zeile 300
`                    compilePayload(q);  // ruft im Fehlerfall closeRequest auf

                    if((NULL != mCbPayload) && (GridOnProFilePara != q->cmd) && (GetLossRate != q->cmd))
                        (mCbPayload)(q->cmd, q->iv);

                    closeRequest(q, true); // hier das zweite mal
                    break;`

closeRequest ruft cmdDone auf
cmdDone inkrementiert den mRdPtr der queue

lumapu, 2024-02-04:
danke für's debuggen 😊

lumapu, 2024-02-04:
teste gerade einen fix, kommt später

closed 2024-02-17 by lumapu