lumapu/ahoy #1404: ESP8266 keine Anzeige der Leistung

kiu77, 2024-02-05:
## Hardware
  * [ ] ESP8266
  * [ ] ESP32
  * [ ] Raspberry Pi

Modelname: ______
Retailer URL: ______

### nRF24L01+ Module
* [ ] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [ ] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [ ]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: _._.__
Github Hash: _______

### Build & Flash Method:
* [ ] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

kiu77, 2024-02-05:
Ich hatte eine alte Platine mit 8266 ausgekramt, die ich jemand schenken wollte. Also direkt upgedated.
Sie funktionierte nicht mehr. Da merkte ich, dass da noch die alter Belegung mit CE = D4 und IRQ = D3
der NRF-Moduls drauf war. Dies habe ich korrigiert und der Empfang von Daten via Weboberfläche geht gut.
Btw.: Die neuen Versionen sind da richtig fix geworden, Kompliment!

Drauf ist jetzt die alleraktuelleste FW 0.8.73
Das 0,96"-Display ist so angeschlossen: SCL = D1 und SDA = D2.
Aktiviere ich das Display in den Einstellungen, dann wird beim harten Reboot folgendes angezeigt:

agoydtu.de
Ahoy!
0.8.73

Soweit erst mal ein Lebenszeichen.
Doch auch wenn Daten empfangen und auf der Webseite angezeigt werden, bleibt das Display so, wie direkt nach
dem Booten, zeigt also keine Werte an.

Also habe ich manuell Clock auf D1 und Data auf D2 gesetzt. Dann aber bleibt das Display nach hartem Reboot
einfach Dunkel. Was mache ich falsch? Ganz früher mit FW < 0.6 ging es mal.

[image]

You69Man, 2024-02-05:
Gerade ausprobiert. Frischer ESP8266 mit 0.8.73 + OLED 0.96", und exakt die von dir gezeigten Einstellungen  SCL = D1 und SDA = D2.  (Graph ist 0/aus).
Display zeigt zuerst Ahoy Startseite, und ca. 1 Sek später den normalen Betriebs-Screen (momentan "offline", weil Nacht).
Wenn bei dir auch schon einmal die Startseite angezeigt wird, sollte das Display korrekt angeschlossen sein.
Dass das Display anschließend "hängen bleibt", kann nur heißen, dass hinterher der I2C Bus unterbrochen wurde ("Wackler"), oder dass sich der ganze ESP aufgehängt hat (dann würde aber auch die Webseite nicht mehr aktualisiert werden).

kiu77, 2024-02-06:
Der Fehler zeigte sich auch bei den Release 0.8.36, 0.7.36 und 0.6.9.
Überraschenderweise zeigte das Display heute morgen, nach längerer Ablagerung (stromlos sein) etwas an.
Gleich von 0.8.73 auf 0.8.74 upgedated. Selbes Bild: Es gab ne Anzeige.
Allerdings gab es beim Refresh Anzeigefehler:
[image]

Also habe ich nun auch noch das Display gewechselt auf ein 1,3"-Modell (hatte ich noch da): Jetzt tut es!
Also hatte auch das Display einen Schaden, zwar einen merkwürdigen, aber doch einen Defekt.
In den Einstellungen habe ich dann auf das 1,3"-Display umgestellt und jetzt ist nahezu alles gut.
Interessanterweise wird links jetzt die Feldstärke des Wechselrichters dargestellt (3 Balken), aber
links bleibt die Feldstärke des WLAN-Signals aus. Damit kann ich leben.

closed 2024-02-17 by lumapu