lumapu/ahoy #1405: com loop duration

sommer05, 2024-02-05:
### Platform

ESP32

### Assembly

I did the assebly by myself

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.3.6

### Github Hash

??wo finde ich den

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

kein mqtt

### Debug Serial Log output

```shell
18:29:40.577 -----
18:29:40.577 I: com loop duration: 503ms
18:29:40.578 -----
18:29:50.576 -----
18:29:50.576 I: com loop duration: 502ms
18:29:50.577 -----
```

### Error description

ich habe jetzt 3 verschiedene boards mit ver 0.8.36 probiert (esp32, esp8266 D1 mini, esp8266 lolin) bei allen dreien läuft 0.7.36 problemlos , sobald ich mit denselben settings auf 0.8.xx update bekomme ich bei allen dreien diese meldung im log und nichts wird in der live ansicht mehr angezeigt. hab mich schon durchgegoogelt aber nicht wirklich was dazu gefunden, vielleicht kann mir ja jemand helfen, wahrscheinlich nur eine kleinigkeit für euch für mich aber....
danke im voraus

knickohr, 2024-02-05:
Den Inverter enabled ?

[image]

sommer05, 2024-02-05:
ja danke ist natürlich enabled, habe echt alles schon 100* überprüft und finde keine lösung ...

blueline13, 2024-02-05:
Wie ist der Status vom NRF Modul unter System. ist da alles grün?

sommer05, 2024-02-05:
> schon bei den Reitern Input und Radio nachgesehen, ob es was zu ergänzen gibt? Die Serialnummer neu eingeben vielleicht klemmt da was.

überprüft alles richtig

sommer05, 2024-02-05:
> Wie ist der Status vom NRF Modul unter System. ist da alles grün?

ist grün und schreibt connected

closed 2024-02-05 by sommer05

UVILG, 2024-03-06:
ich habe gestern die aktuellste Version aufgespielt. Nach der Verbindung bekomme ich die Meldung  I : com loop duration: 403ms
was war die Ursache bei dem auch hier beschriebenen  Problem ?