lumapu/ahoy #1408: Sortierung der passwortgeschützten Begriife in "Settings"

Gubi2023, 2024-02-06:
es wäre schön, wenn die Begriffe in den Settings links und rechts die gleiche Reihenfolge hätte...
[image]

knickohr, 2024-02-06:
Habt ihr Sorgen !  😲🤪😉

Gubi2023, 2024-02-06:
sind halt so die Kleinigkeiten fürs perfekte Design . Ansonsten seid ihr ja wirklich super !!!
[image]

knickohr, 2024-02-06:
Ja, ist doch OK. Ist halt historisch so gewachsen,

lumapu, 2024-02-06:
ist nicht ohne weiteres möglich, da die Sortierung auf der Makse (Code im Hintergrund) basiert. Ich denke das werde ich nicht fixen, da es in unnötig größerem Code resultiert.

Gubi2023, 2024-02-06:
ok, war nur Schönheit, wenn der Aufwand zu gross ist, kein Problem.

closed 2024-02-06 by Gubi2023