lumapu/ahoy #1421: MQTT :  total-Werte kommen zu oft

knickohr, 2024-02-10:
Ja, wieder mal ein Spam-Issue von mir 🤪

Seit geraumer Zeit werden die Topics mit den total-Werten am Ende der Datenpayloads alle hintereinander wiederholt. Als das Gap noch drin war kamen sie einzeln nach den Daten.

Da wir jetzt in sehr kurzer Zeit ohne den Gap, ich hatte schon Loop durations unter 4s bei 16 Invertern, brauchen diese total-Werte nicht mehr für jeden Inverter einzeln rausgeballert werden, zumal sie ja auch noch identische Werte enthalten. Einmal am Ende des Zyklus reicht.

[image]

knickohr, 2024-02-11:
Ja, sie kommen wieder zu jedem Inverter 👍

Es würde aber reichen einmal am Ende pro Intervall, da das ja jetzt durch die sehr schnelle Abfrage flott geht.

knickohr, 2024-02-18:
Da wir ja jetzt mal eine stabile Release haben bei der auch die Kommunikation super funktioniert, könnten wir das Thema „Verschlanken von MQTT“ mal aktiv angehen 😇

Erst mal würde ich ein bißchen aufräumen und unnötige mehrfach kurz hintereinander ausgegebene Topics eliminieren :   Die Schnelle kurz hintereinander folgenden Yield-Werte wären da ein heißer Kandidat.

knickohr, 2024-02-23:
@lumapu 
Netter Versuch, es kommen weniger Total, aber leider auch neuen Bug eingebaut 😉

[image]

knickohr, 2024-02-23:
Nachtrag :
Der Bug tritt nur in den Zero-Version von heute auf. In der gestrigen Version funktioniert das Total. Es kommen auch nicht mehr 16x die Werte, nur noch 5x 😅

Ich würde sagen, Teilerfolg 😉

lumapu, 2024-02-23:
warum sollte sich hier die Zero Variante anders verhalten?
Gäbe es die nicht wäre es dann ein voller Erfolg?

knickohr, 2024-06-02:
Problem immer noch vorhanden, fällt aber mit dem neuen JSON-Payload nicht mehr so ins Gewicht. Ich bekomme jetzt pro Abfrageintervall bei 16 Invertern 3x einen total-JSON-Payload.

stefan123t, 2024-10-29:
@lumapu 5x statt 16x Wiederholungen klingt wie wenn er das zwar in der letzten Runde aber dafür mit fünf Retries macht ?

knickohr, 2024-10-30:
Ohhh, das muß ich gleich nochmal,testen. Es gab einige Veränderungen an der Datenabfrage und in der Queue. Kann gut sein das es jetzt weg ist, oder sogar noch schlimmer 🫣

stefan123t, 2024-10-30:
@knickohr ich hatte es bei mir auf der Console auch gesehen, bei (m)einem Inverter kamen die MQTT total Werte alle 5 Sekunden.
Je nachdem was die RF Routine für Kommandos an den WR geschickt hat, lagen da dann zwischen 1-3 total Werte zwischen den ebenfalls relativ regelmäßigen Frage/Antwort Paketen.
Die Daten waren halt 1-3x die tupfengleichen Werte, da es ja keine neuen Werte vom WR in demselben Zeitraum gegeben hatte.