lumapu/ahoy #1422: 0.8.78 - opendtufusion startet nicht [Bug]

hrolofs, 2024-02-11:
### Platform

ESP32

### Assembly

the DTU was already assembled

### nRF24L01+ Module

nRF24L01+ plus

### Antenna

external antenna

### Power Stabilization

Elko (~100uF)

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.78

### Github Hash

c290273

### Build & Flash Method

ESP Tools (flash)

### Setup

- Mqtt
- NRF

### Error description

0.8.78 frisch per Web Installer auf meine opendtufusion installiert. Power off/on und Display bleibt dunkel. DTU frisch mit Flash Tool aufgesetzt gleiches Ergebnis. Die DTU startet nicht (Display bleibt dunkel)

1. DTU am Laptop angeschlossen, Display bleibt dunkel
2. mit Putty auf den COM Port connected, Display geht an und DTU startet.

Zurück auf die 0.8.77 geflasht, DTU startet ganz normal.

Meine ESP32 (blinkyparts) DTU scheint dagegen mit der 0.8.78 keine Probleme zu haben.

Dirk-Ahoy, 2024-02-11:
bitte die deutsche Version nehmen die englische startet nur wenn man die DTU per USB Kabel am Rechner hat - kleiner Debug-Bug reingekommen - so kann man es aber fixen - Kabel an den Rechner dann startet sie und dann umflashen.

lumapu, 2024-02-12:
in `0.8.79` wieder behoben, danke @Dirk-Ahoy für die zwischenzeitliche Abhilfe

closed 2024-02-17 by lumapu