lumapu/ahoy #1423: 0.8.78 hängt sich im laufenden Betrieb auf

elisenjens, 2024-02-11:
### Platform

ESP8266

### Assembly

I did the assebly by myself

### Antenna

external antenna

### Power Stabilization

nothing

### Connection picture

- [ ] I will attach/upload an Image of my wiring

### Version

0.8.78

### Github Hash

c290273

### Build & Flash Method

AhoyDTU Webinstaller

### Setup

Standard ohne MQTT

### Error description

von Zeit zu Zeit (täglich) kann ich Ahoy per LAN nicht erreichen; nach Stecker raus/rein läuft sie wieder

Gubi2023, 2024-02-13:
dito

lumapu, 2024-02-16:
Leider kann man ohne Log an diesem Fehler wenig ausrichten, da ich ihn bis dato noch nicht gesehen habe. Gibt es ein Indiz wann es besonders häufig passiert?

elisenjens, 2024-02-17:
Bei mir war das mehrmals täglich spontan; seit der .81 hab ich das nicht mehr (vlt. liegt es daran, dass ich jetzt komplett neu installiert habe und das ein Update-Problem war).

closed 2024-02-17 by elisenjens