lumapu/ahoy #1424: ESP32 und OLED - (Immer noch?) Reboots

HorstyS, 2024-02-12:
## Hardware
  * [] ESP8266
  * [X] ESP32
  * [ ] Raspberry Pi

Modelname: WROOM
Retailer URL: ______

### nRF24L01+ Module
* [X] nRF24L01+ you verified this is a **Plus model** capable of the required 256kBit/s mode
* [ ] **square dot** indicates original Nordic Semicon chip 
* [ ] **round dot** indicates copy-cat / counterfeit SI labs chip

### Antenna:
* [X] circuit board
* [ ] external antenna (SMA)

### Power Stabilization:
* [X]  100uF Electrolytic Capacitor 
connected between +3.3V and GND (Pin 1 & 2) of the NRF Module
* [ ] Voltage stabilizing motherboard

### Version / Git SHA: 
Version: 0.8.79
Github Hash: _______

### Build & Flash Method:
* [X] AhoyDTU Webinstaller
* [ ] VSCode - Platform IO
* [ ] Arduino
* [ ] ESP Tools

### Debugging:
* [ ] USB Serial Log (attached)
* [ ] Setup settings (use our templates ... to be added)

hi,

ich bin leider etwas "raus" aus den ganzen issues, aber ich kann mich noch erinnern, dass es mal probleme gab mit ständigen reboots. ich habe das verhalten bei einer meiner beiden ESP32 DTUs auch, allerdings erst seit einer 0.8.7x version. wenn daten vom inverster ankommen ist alles tutti, so wie ich das sehe, und sie läuft stabil. nachdem dann aber der inverter abends keine daten mehr sendet fängt sie an ständig zu rebooten. mal alle paar sekunden, mal hält sie ein paar stunden in der nacht durch und fängt wieder an. das OLED display scheint sich wohl auch nicht mehr auszuschalten. ich habe aktuell keine entsprechenden logs dazu, kann diese aber vermutlich morgen nachreichen, wenn notwendig.
ist das thema (noch) bekannt und offen? liegt das am display?
ich bin mal zurück auf die 0.8.56 und dort tritt das problem nicht auf.
die zweite DTU läuft übrigens auch mit der 0.8.56 und einem ePaper. die lasse ich auch erst mal so, damit ich über MQTT die daten weiterhin bekomme ;)

HorstyS, 2024-02-12:
Klaro :)

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

HorstyS, 2024-02-12:
[image]

You69Man, 2024-02-12:
Ich habe jetzt deine Einstellungen mit meinen verglichen.

Hier aktuelle Einstellungen bei mir, die anders sind, als bei dir:
* Display: "Turn off while inverters are offline" : Aus
* Display / Screensaver: Motion
* Inverter / Advanced: "Pause communication during night (lat. and lon. need to be set)" :  Aus
* Inverter : "Reset values and YieldDay at midnight." : Aus
* Inverter / Interval: 5s
* MQTT : An

Bei allen anderen Einstellungen die auch noch unterschiedlich sind, kann ich mir beim besten Willen nicht vorstellen, dass sie hier einen Einfluss haben könnten.

Ich stelle jetzt einmal für einen Tag lang meine Einstellungen auf deine um, und schaue, ob es irgendwelche Reboots oder Display Effekte gibt!

You69Man, 2024-02-12:
PS: Übrigens, ich verwende doch kein Wroom32 Board, wie oben von mir behauptet, sondern ein Wemos D1 Mini ESP32 Board, allerdings mit dem Wroom32 firmware target.
Außerdem habe das Display auf den GPIO Pins 13 und 14 angeschlossen, nicht auf 21 und 22 wie bei dir. Das sollte hier aber auch keinen Unterschied machen (der ESP32 multiplext die I2C intern auf 'beliebige' Pins).

HorstyS, 2024-02-13:
ich hatte zwischen ca. 17:15 uhr gestern und heute ca 7:30 uhr nur zwei reboots. jeweils um diese uhrzeit. das display war (zumindest bis ich ins bett gegangen bin und aufgestanden bin) aus.
mal schauen wie es sich nun unter tags verhält.
mal eine frage: wenn ich ein update über das UI anstoße, wird vermutlich nur ein in place update gemacht? gibt es denn eine möglichkeit über das UI ein full update inkl. dem vorherigen kompletten löschen des geräts so wie über die DTU website im internet? vielleicht würde das einige solcher probleme beheben? oder ist hierfür generell die seite im netz zu nutzen?

You69Man, 2024-02-13:
Meinw DTU läuft nun seit >24 Std mit deinen Einstellungen ohne einen einzigen Reboot durch. Auch das Display hat sich über Nacht brav ausgeschaltet und morgens wieder ein. Daran liegt es also wohl nicht.

Zur Info noch: Unabhängig von den Einstellungen schaltet sich Display nach Power on oder nach einem Reboot immer für 60 Sekunden ein. Wenn du also in einer schnellen Reboot Schleife hängst, kann es sein, dass es deshalb nie ausschaltet. Das müsstest du dann aber daran sehen, dass immer wieder einmal der Startscreen erscheint.

Was du jetzt nich probieren könntest wäre einmal deine Settings komplett zu löschen und alles neu zu konfigurieren. Vielleicht versteckt sich irgendwo noch ein verwaister Parameter aus einer alten Firmwareversion und macht irgendeinen Unfug.

HorstyS, 2024-02-14:
meine DTU hat sich seit gestern Abend wieder sehr oft gebootet. heute morgen eigentlich immer im minutentakt. seit der inverter wieder nun seit ein paar minuten wieder liefert ist die DTU stabil. und ja, der startscreen kommt dann immer wieder ;)

ich habe auch die vermutung, dass es an irgend einem (obsoleten, falschen, ...) flag o.ä. liegt der durch ein update das ganze verursacht. vermutlich würde sich das verhalten durch einen wipe und einer neuinstallation einer aktuellen DEV version beheben lassen, aber die frage ist ob es für die entwicklung wichtig ist herauszufinden was/warum das verhalten so auftritt.

You69Man, 2024-02-14:
Nützlich  zur Fehlersuche wäre noch ein logfile das während des rebootens aufgezeichnet wurde. Wenn du ein haufiges rebooten hast , sollte das einfach einzufangen sein. Entweder mit dem Weblog oder mit USB Serieller und Putty.

dtuuser, 2024-02-14:
Bei mir, ESP32 mit ssd1306 Display, keine automatischen Reboots. Update seit Version 7 immer über Download der englischen Datei aus fw.ahoy.....  und Einspielen über die Updateseite in Ahoy. Falls das mal nicht einggespielt wird, dann muss ich vorher manuell per Web rebooten,  aber zu 90% läuft das Update durch.

Nochmals an die Programmierer..... Klasse Arbeit

HorstyS, 2024-02-15:
> Nützlich zur Fehlersuche wäre noch ein logfile das während des rebootens aufgezeichnet wurde. Wenn du ein häufiges Rebooten hast , sollte das einfach einzufangen sein. Entweder mit dem Weblog oder mit USB Serieller und Putty.

sie läuft nun seit gut 6,5 stunden ohne reboot, und ich denke, dass sie bald anfangen wird die kommunikation mit dem inverter wieder aufzunehmen.
mit weblog meinst du die webconsole im UI? und mit USB Serieller was? die DTU am rechner anstecken und dann per putty darauf zugreifen? oder kann man den SSH zugang irgendwie aktivieren und dann per WLAN darauf zugreifen?

You69Man, 2024-02-15:
> mit weblog meinst du die webconsole im UI? 
Ja genau. Dort dann einfach mit dem Copy Button rauskopieren.

>und mit USB Serieller was? 
die DTU am rechner anstecken und dann per putty darauf zugreifen? 
Ja das wäre die zweite Möglichkeit

>oder kann man den SSH zugang irgendwie aktivieren und dann per WLAN darauf zugreifen? 
SSH Zugang zur DTU ist mir nicht bekannt

HorstyS, 2024-02-16:
da steht leider nichts spannendes drin. wenn die DTU neu bootet sieht man das nur am zeitversatz des abfragerhythmus (bei mir 15 sek.):

`17:08:41.000 I: (#0) no communication to the inverter (night time)
17:08:56.000 I: (#0) no communication to the inverter (night time)
17:09:11.000 I: (#0) no communication to the inverter (night time)
17:09:26.000 I: (#0) no communication to the inverter (night time)
17:09:41.040 I: (#0) no communication to the inverter (night time)
17:09:56.000 I: (#0) no communication to the inverter (night time)
17:10:48.000 I: (#0) no communication to the inverter (night time)
17:11:03.000 I: (#0) no communication to the inverter (night time)
17:11:18.040 I: (#0) no communication to the inverter (night time)
17:11:33.000 I: (#0) no communication to the inverter (night time)
17:11:48.000 I: (#0) no communication to the inverter (night time)`

You69Man, 2024-02-16:
> da steht leider nichts spannendes drin. wenn die DTU neu bootet sieht man das nur am zeitversatz des abfragerhythmus (bei mir 15 sek.):

Der Log ist vermutlich vom Webserial. Kannst du bitte auch ein Log von der USB seriellen liefert? Dort würde man vermutlich wesentlich mehr sehen.

HorstyS, 2024-02-16:
> Der Log ist vermutlich vom Webserial. Kannst du bitte auch ein Log von der USB seriellen liefert? Dort würde man sicherlich wesentlich mehr sehen.

korrekt.
wie ist hier die vorgehensweise? am pc anstecken, laufen lassen und wenn er abends neu bootet das log ziehen? oder bleibt das log auch erhalten wenn die DTU stromlos war und neu gestartet wird?

You69Man, 2024-02-16:
> korrekt.
> wie ist hier die vorgehensweise? am pc anstecken, laufen lassen und wenn er abends neu bootet das log ziehen? oder bleibt das log auch erhalten wenn die DTU stromlos war und neu gestartet [...]

Am besten dann, wenn du die häufigen reboots siehst (Abends?) einmal mit einem seriellen Terminal (z B. PuTTY) auf die COM der Ahoy verbinden (115200 Baud). Wenn du dann den Reboot siehst, den Log mit Copy Paste rauskopieren.

HorstyS, 2024-02-19:
also, folgendes habe ich beobachtet:
wenn ich die DTU an ihrem üblichen ort mit ihrer aktuellen stromversorgung habe fängt sie am abend irgendwann an immer wieder selbstständig zu booten.
wenn ich sie an meinen PC hänge passiert das nicht. im putty steht dem entsprechend natürlich nichts spannendes (no communication...night time). es könnte evtl. an der spannungsversorgung liegen....aber wieso passiert das dann nicht mit einer 0.8.56?!....
ich habe jetzt mal auf die aktuelle DEV version upgedated und beobachte das heute abend mal. wenn es damit wieder auftritt würde ich mir mal die stromversorgung vornehmen.

HorstyS, 2024-02-20:
latest news zu dem thema:
ich habe gestern auf die 0.8.83 via UI upgedated und die DTU läuft nun seit fast 24 Stunden wieder stabil ohne neustart. die stromversorgung und der aufstellort ist der selbe wo ich zuvor die probleme hatte.
keine ahnung was das nun gewesen sein kann.
wenn ihr wollt kann ich diesen issue schließen...oder jemand anderes :)

HorstyS, 2024-02-20:
ahhhhhh.......zu früh gefreut. nachdem ich den letzten post abgesetzt habe hat die DTU wieder angefangen neu zu booten. :(
ich denke ich werde sie mal komplett platt machen und ein frisches bin-file hochladen...

HorstyS, 2024-02-21:
so, ich habe dann gestern mittag den wipe und die neuinstallation durchgeführt. die einstellungen habe ich per hand vorgenommen. ich bin auch bei der aktuellen stable version geblieben. seitdem kein neustart über nacht. aktuell produziert der inverter und die daten kommen an.
was auch immer das problem verursacht hat scheint nun wieder verschwunden zu sein.
ich schließe damit meinen post.

closed 2024-02-21 by HorstyS